CameronBodine/PINGMapper #43: V1.1.0 Own Data sonObj IndexError from first beam

AdrianPinchbeck, 2023-01-02:
Successfully loaded latest version, keeping old version elsewhere.  Test data runs successfully on new, but unable to run own data from the main.py as above.  Note: the same data is successfully processed through previous version.  Hope screenshot helps!
[image]

CameronBodine, 2023-01-02:
Thanks for trying @AdrianPinchbeck! Can you confirm a few things for me:

1. Do both .DAT and subfolder share the same name, as:
```
Rec00001.DAT
├── Rec00001
│   ├── B001.IDX
│   ├── B001.SON
│   ├── B002.IDX
│   ├── B002.SON
│   ├── B003.IDX
│   ├── B003.SON
│   ├── B004.IDX
│   └── B004.IDX
```
2. The paths are correctly specified. For example, if you are on a Windows machine, do you have an `r` in front of the path, as:
```
humFile = r'./exampleData/Test-Small-DS.DAT'
sonPath = r'./exampleData/Test-Small-DS'
projDir = r'./procData/PINGMapper-Test-Small-DS'
```

If none of that solves your issue, could you please post the contents of main.py here?

AdrianPinchbeck, 2023-01-02:
My error in path structure, following correction it almost ran, I do apologise.  However, an error was then produced almost through locating missing pings and adding nodata, with a time-out.  Changed fixNoDat to equal False in attempt to bypass.  Ran again, but another error after completion of B004 under Summary of Ping Metadata. Getting much closer, but it may have to be tomorrow now. Thanks for help so far!
[image]
[image]

CameronBodine, 2023-01-02:
Ok, give the new one a try by running `git pull`. This will likely revert your changes in `main.py` to PING-Mapper's defaults, so you will need to edit that with your parameters after the pull. LMK!

AdrianPinchbeck, 2023-01-03:
Okay, new to Git & GitHub, but crash-course done and PING-Mapper code pulled (my parameters in main.py were unchanged after pull).
Success, PING-Mapper is running.
Steadily going through parameter alterations, may have found an issue, but will report once through.
Well done!

CameronBodine, 2023-01-03:
Great! Go ahead and report any further incidents parameter issues here and I will take care of them

AdrianPinchbeck, 2023-01-03:
Noticed earlier that when exportUnknown = True and fixNoDat = True, all processes well until just after the end of Auto estimating depth.  Thus:

[image]

AdrianPinchbeck, 2023-01-03:
That example was the only error I've managed to raise since today's update.  Hopefully my recording file should be with you any time now.

AdrianPinchbeck, 2023-01-03:
One more for you to get your teeth into before I call it a day.  When changing USE_GPU = True, towards the end of Automatically estimating depth for 88 chunks, the following error is produced (processing seems much quicker though):

[image]

CameronBodine, 2023-01-03:
> One more for you to get your teeth into before I call it a day. When changing USE_GPU = True, towards the end of Automatically estimating depth for 88 chunks, the following error is produced (processing seems much quicker though):
> 
> [image]

Ok I will take care of this one too. In this case, the script is failing before doing any prediction, so it appears to be faster. This will be an easy fix. Looking at the other issue now.

CameronBodine, 2023-01-03:
> Noticed earlier that when exportUnknown = True and fixNoDat = True, all processes well until just after the end of Auto estimating depth. Thus:
> 
> [image]

Above issue caused by chunks with all NoData where no depths are estimated, resulting in differing dataframe lengths. 5606e3d fixes this issue.

CameronBodine, 2023-01-03:
@AdrianPinchbeck I removed the `USE_GPU` option for now, or you can just set it to `False`. It was getting an out of memory (OOM) error after fixing your issue you referenced earlier. 

I also fixed the issue caused by setting fixNoDat=True. Each sonar recording has it's nuances, so it is helpful to have you testing with your data! I didn't try all combinations of parameters, but I was able to export all of your data with these fixes. Thanks for testing and let me know if you come across any additional errors.

AdrianPinchbeck, 2023-01-04:
I'll run another git pull to update, and go through parameter changes.
No problem with further testing, but local rivers are currently running too high, so it will be a while before I can obtain any new recordings.  I do have some old files from an ancient 981c that I had linked to a Lowrance plotter on a smaller river, so if I'm feeling brave I'll dig them out and run one through!

AdrianPinchbeck, 2023-01-04:
Uploaded with a fresh copy, and testing going well.
Auto depth detection seemed to struggle a bit in some areas, occasionally following a clay fissure or an outcrop, but easily changed to instrument if needed. Bed structure here is mainly a mix of coarse and fine sand, clay and silt (my recording headed upstream towards the first weir in a tidal section).
Followed your test geotiff on GEP, nice office you have!  Almost got to canyon bottom myself many moons ago, when travelling the US on Greyhound as a student.  Borrowed a tent from a Flagstaff hostel, and camped on the rim - summer '91.

AdrianPinchbeck, 2023-01-04:
Tested old recordings from the Humminbird 981c, and all worked well.

CameronBodine, 2023-01-04:
This is all great news! Glad it is working for you. Yes, I did notice that the depth detection didn't work great for your recording. A lot of the data I trained the depth model on had low contrast between the water column and the bed. I think having the clay bottom with little noise in the water column could be causing the model to not do great. I think I just need to add more samples with conditions similar to your recording to achieve a better model.

That test dataset is actually from my advisor on Horseshoe bend just below Lake Powell. He used to work down there quite a lot. I did have an opportunity last spring to run a multibeam system as part of an expedition to map bathymetry in the canyon below Phantom Ranch. What an experience!!!

Also glad it is working on older recordings for you. I may go ahead and close this issue as complete. If anything else comes up, feel free to open a new issue!

closed 2023-01-04 by CameronBodine