CameronBodine/PINGMapper #46: main_batchDirectory.py typos

bthetford, 2023-01-05:
main_batchDirectory.py wasn't working (I would just get the exception from the try/catch). I removed the try/except construct and found that there was an unused argument 'detectDepth' in the arg list for `read_master_func() `. Removing this argument fixed that issue. I then got the OSerror "cannot save file to non-existent directory". I added:
`
from pathlib import Path
Path(outDir).mkdir(parents = True, exist_ok = True)
`
The script worked after this.

CameronBodine, 2023-01-05:
Thanks for the feedback @bthetford. I'm actually not able to reproduce your error. Since removing `detectDep` from the arg list worked for you, I suspect that you may not be using the latest version of the software scripts, but I'm not entirely certain. Does updating PING-Mapper by running `git pull` resolve the issue?

bthetford, 2023-01-06:
Thanks Cam, yes the install I was using was from a few months ago. I re-cloned the repository and am currently trying to create a new conda environment, but it get's stuck on "Solving environment", even after I let it go all night. Have you seen that before? I see the only difference is psutil. (Unrelated to this issue, let me know if I should post another).

CameronBodine, 2023-01-06:
Great. I will close this issue and opened a new issue with conda install #47 . I will investigate and post updates there.

closed 2023-01-06 by CameronBodine