CameronBodine/PINGMapper #47: Conda won't solve env

CameronBodine, 2023-01-06:
Thanks Cam, yes the install I was using was from a few months ago. I re-cloned the repository and am currently trying to create a new conda environment, but it get's stuck on "Solving environment", even after I let it go all night. Have you seen that before? I see the only difference is psutil. (Unrelated to this issue, let me know if I should post another).

_Originally posted by @bthetford in https://github.com/CameronBodine/PINGMapper/issues/46#issuecomment-1373881207_

CameronBodine, 2023-01-06:
I will take a look at this this morning. It may take me a bit, as install of the conda environment takes a while even when it does succeed. I will let you know!

CameronBodine, 2023-01-06:
@AdrianPinchbeck I noticed on Twitter that you also had a similar problem updating to the latest `ping` conda env. What was you workaround?

AdrianPinchbeck, 2023-01-06:
@CameronBodine I'm very new to this, but think this is what you're looking for.  I couldn't see module psutil from within 'ping', however in 'base' it was clearly installed (using pip3 show psutil).  I activated 'ping' and updated psutil using pip3 install psutil at the Anaconda prompt.

CameronBodine, 2023-01-06:
Thanks @AdrianPinchbeck . I will keep investigating. 

@bthetford as a workaround, if you still have your old `ping` env installed, go ahead and activate it, then run `pip install psutil` in the conda prompt, and you should be good to go.

For my own notes, I believe there are other issues with updated packages, even outside of `psutil` or `gdal`. Still investigating.

AdrianPinchbeck, 2023-01-06:
Thought the issue was just my end, good luck!

CameronBodine, 2023-01-06:
Ok @bthetford, trying running a `git pull` then running the installation again. It took a little over an hour for my computer to install. Not ideal, but hopefully you get it up and running. Please report back!

CameronBodine, 2023-01-06:
In addition I ran 'conda clean --all' and 'conda update conda' before install.

AdrianPinchbeck, 2023-01-06:
I think you're on the right track there Cameron, my investigations seemed to hint towards intolerance between versions.  As I say though, I'm very green here.

bthetford, 2023-01-09:
Thanks @CameronBodine, it must be an issue on my end, I can't even get through `conda update conda`, again it sticks on Solving environment indefinitely. I'll update here once I get it working.
Beau

bthetford, 2023-01-09:
Alright! Finally got this working:
I uninstalled Anaconda3 and installed miniconda3 instead, to see if that would fix the solving environment issue, but the issue persisted. It finally worked after I installed [libmamba](https://www.anaconda.com/blog/a-faster-conda-for-a-growing-community) and used that as the solver. With libmamba it finished solving in about 10 seconds! Woo!

NOTE: The first time I attempted to create the ping environment, I got the following error:
`InvalidArchiveError("Error with archive C:\\Users\\b.thetford\\AppData\\Local\\miniconda3\\pkgs\\tensorflow-base-2.3.0-eigen_py38h75a453f_0.conda.  You probably need to delete and re-download or re-create this file.  Message was:\n\nfailed with error: [Errno 2] No such file or directory: 'C:\\\\Users\\\\b.thetford\\\\AppData\\\\Local\\\\miniconda3\\\\pkgs\\\\tensorflow-base-2.3.0-eigen_py38h75a453f_0\\\\Lib\\\\site-packages\\\\tensorflow\\\\include\\\\external\\\\llvm-project\\\\mlir\\\\_virtual_includes\\\\DerivedAttributeOpInterfaceIncGen\\\\mlir\\\\Interfaces\\\\DerivedAttributeOpInterface.cpp.inc'")`

So I ran `conda clean -a` and it worked after that.

CameronBodine, 2023-01-09:
Glad you got it working! I will make a note in the installation instructions that Mamba works much better. I have also switched over to miniconda and find it to function a bit better. Thanks for posting this info!

CameronBodine, 2023-01-10:
Thanks so much for sharing the libmamba info @bthetford! It is much faster and now the recommended way of [installing PING-Mapper](https://cameronbodine.github.io/PINGMapper/docs/gettingstarted/Installation.html). Thanks for the recommendation!

bthetford, 2023-01-11:
Great, glad I could help! I found libmamba through a "conda never solves" stack exchange rabbit hole.