CameronBodine/PINGMapper #49: pingDat array out of bounds error

bthetford, 2023-01-09:
This error seems to be related to the no data fix that was recently added- It only seems to occur when I try and process a sonar recording in which I know there are bad pings (when the transducer came out of the water while going over a riffle). 
The error:
```
Exporting sonogram tiles:

        Exporting 96 sonograms for ss_port
joblib.externals.loky.process_executor._RemoteTraceback:
"""
Traceback (most recent call last):
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\joblib\externals\loky\process_executor.py", line 436, in _process_worker
    r = call_item()
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\joblib\externals\loky\process_executor.py", line 288, in __call__
    return self.fn(*self.args, **self.kwargs)
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\joblib\_parallel_backends.py", line 595, in __call__
    return self.func(*args, **kwargs)
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\joblib\parallel.py", line 264, in __call__
    for func, args, kwargs in self.items]
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\joblib\parallel.py", line 264, in <listcomp>
    for func, args, kwargs in self.items]
  File "src\class_sonObj.py", line 1392, in _exportTiles
    self._WCR_SRC(sonMeta) # Remove water column and redistribute ping returns based on FlatBottom assumption
  File "src\class_sonObj.py", line 1502, in _WCR_SRC
    pingDat[srcIndex] = intensity # Store intensity at appropriate horizontal range
IndexError: index -2147483648 is out of bounds for axis 0 with size 17351
"""
```
I get the error whether `fixNoDat` is set to  `True` or `False`. (Unfortunately my affected recording is too big to upload here).

CameronBodine, 2023-01-09:
Bummer, thought I had this one fixed!! (#43) Can you throw the recording here: https://drive.google.com/drive/folders/1-0uBe4E2C5O3ME6gLuQ1DG-w05Egcb33

I'll take a look. As a note, this workflow will locate portions of the recording where ping's weren't recorded to file. If the unit was still pinging when you pulled it out of water, those _should_ still be valid and present in the recording.

bthetford, 2023-01-09:
Uploaded the recording. The sonar was still pinging when it raised out of the water, I think it wouldn't process in a previous PINGMapper version and threw an error about an invalid Pythagorean theorem? I can try and dig up an older version and run it again. I thought maybe lifting the transducer out might have produced infinite/NA depths or something.

CameronBodine, 2023-01-09:
Ah, yes you may be right on that. I'll take a look.

bthetford, 2023-01-09:
Thanks Cameron. I'm remembering now it may have been negative depths that were produced by air-pings. Not sure why that would hurt a pythag theorem (wouldn't it get squared)? Anyhow maybe it was producing a square root of a negative.

CameronBodine, 2023-01-09:
That's helpful to know. That was also the problem with #43, but resulted from the auto depth detection. I did add a step to zero out negative depth values, so not sure why it didn't work in this case. I must have only added that to the auto depth workflow. And yes, not sure why the square wouldn't fix that on the pythagorean theorem. I'll look at that too.

CameronBodine, 2023-01-10:
Ok, I just committed a fix. Give it a try and lemme know. I believe the issue is that if there are any zero depths, then the smoothing algorithm can create negative depths. So I made a check after that step to set negative depths to zero. For the pythagorean theorem issue, I use the depth value as an index to pull values from an array. Since an index can't be negative, it throws an error.

One other note on this particular recording. Pulling the transducer out of the water set the range to the maximum value (probably somewhere around 800 ft). This makes for a very large image so it takes a while to export that particular chunk. It will appear like PING-Mapper is stuck but just let it process and it will eventually finish.

bthetford, 2023-01-11:
Thanks I'll give this a try today. Makes sense that air-pings would produce 0 depths, I would never expect to see negatives on the humminbird screen. 
And yes that recording is pretty long also. Smooth sailing until we get to the salmon spawning riffle and then we have to jam up it so we don't slam into the gravel (and of course the riffle was the main thing we wanted to image).

bthetford, 2023-01-11:
Your fix worked! Did you just have any negative depths get set to 0? 
The air-ping section looks like this:
[image]

Of course I can clip this section out, but maybe you can handle 0 depth pings using your `fixNoData` routine also? That would be slick.

CameronBodine, 2023-01-11:
Yes, just needed to fix negative depths produced after smoothing the sensor depth. I did it for the auto depth portion, but not for the sensor depth (unintentionally). Glad it works!

You could try setting `remShadow=2` and it should (hopefully) mask that section out. 

A zero depth doesn't always indicate that there are no returns across the swath. The sensor depth or auto depth method could be inaccurate.  In this case, a better routine to add would be to check the range, and if it suddenly blows up (gets suddenly very large), then flag that data and potentially remove it. 

A much easier approach would be to go ahead and stop the recording when you pull the transducer out and restart when you are ready to scan again ;-)

I'll go ahead and close this issue. Feel free to post any new issues, as always!

closed 2023-01-11 by CameronBodine

bthetford, 2023-01-11:
Sounds good, haha our goal was to scan the riffle with ~2ft deep water over it, while going upstream, which proved pretty much impossible. Hoping to return at higher flows and rescan with the riffle submerged instead! 
I'll give the shadow removal a shot.