CameronBodine/PINGMapper #54: Shadow removal - `weights` argument not defined in tensorflow/keras model call

bthetford, 2023-01-11:
When I try to process a recording (my own, or the example data using test_PINGMapper.py), using `remShadow = 1` or 2, I get the following error: 

```
Automatically detecting shadows for 7 chunks:
MODE: 1 | Remove all shadows...
joblib.externals.loky.process_executor._RemoteTraceback:
"""
Traceback (most recent call last):
  File "src\class_portstarObj.py", line 483, in _initModel
    model = tf.keras.models.load_model(weights)
NameError: name 'weights' is not defined

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\joblib\externals\loky\process_executor.py", line 436, in _process_worker
    r = call_item()
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\joblib\externals\loky\process_executor.py", line 288, in __call__
    return self.fn(*self.args, **self.kwargs)
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\joblib\_parallel_backends.py", line 595, in __call__
    return self.func(*args, **kwargs)
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\joblib\parallel.py", line 264, in __call__
    for func, args, kwargs in self.items]
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\joblib\parallel.py", line 264, in <listcomp>
    for func, args, kwargs in self.items]
  File "src\class_portstarObj.py", line 1545, in _detectShadow
    self.shadowModel = self._initModel(USE_GPU)
  File "src\class_portstarObj.py", line 486, in _initModel
    model.load_weights(self.weights)
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\tensorflow_core\python\keras\engine\training.py", line 234, in load_weights
    return super(Model, self).load_weights(filepath, by_name, skip_mismatch)
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\tensorflow_core\python\keras\engine\network.py", line 1222, in load_weights
    hdf5_format.load_weights_from_hdf5_group(f, self.layers)
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\tensorflow_core\python\keras\saving\hdf5_format.py", line 651, in load_weights_from_hdf5_group
    original_keras_version = f.attrs['keras_version'].decode('utf8')
AttributeError: 'str' object has no attribute 'decode'
"""
``` 

The error is identical whether I try to remove all or only bank shadows.

bthetford, 2023-01-12:
That fixed changed the resulting error a little bit, but it still ends with the same problem with the string `'utf8'` 
```
Automatically detecting shadows for 48 chunks:
MODE: 1 | Remove all shadows...
joblib.externals.loky.process_executor._RemoteTraceback:
"""
Traceback (most recent call last):
  File "src\class_portstarObj.py", line 483, in _initModel
    model = tf.keras.models.load_model(self.weights)
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\tensorflow_core\p
    return hdf5_format.load_model_from_hdf5(filepath, custom_objects, compile)
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\tensorflow_core\p
    model_config = json.loads(model_config.decode('utf-8'))
AttributeError: 'str' object has no attribute 'decode'
```

CameronBodine, 2023-01-12:
Can you confirm that you have a model folder in the ping mapper directory and that there are hdf and json files in there?

bthetford, 2023-01-12:
I have folders bankpick, bedpick and shadow in the folder PINGMapper/models, inside PINGMapper/models/shadow I have 1 .json file and 1.h5 file

CameronBodine, 2023-01-12:
With `ping` environment activated, can you run `conda list` and post the output?

bthetford, 2023-01-12:
```
$ conda list
# packages in environment at C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping:
#
# Name                    Version                   Build  Channel
_tflow_select             2.1.0                       gpu
absl-py                   0.15.0             pyhd3eb1b0_0
affine                    2.3.0              pyhd3eb1b0_0
aiohttp                   3.8.3            py37h2bbff1b_0
aiosignal                 1.2.0              pyhd3eb1b0_0
astor                     0.8.1            py37haa95532_0
async-timeout             4.0.2            py37haa95532_0
asynctest                 0.13.0                     py_0
attrs                     22.1.0           py37haa95532_0
blas                      1.0                         mkl
blinker                   1.4              py37haa95532_0
blosc                     1.21.0               h19a0ad4_1
bottleneck                1.3.5            py37h080aedc_0
brotli                    1.0.9                h2bbff1b_7
brotli-bin                1.0.9                h2bbff1b_7
brotlipy                  0.7.0           py37h2bbff1b_1003
bzip2                     1.0.8                he774522_0
ca-certificates           2022.10.11           haa95532_0
cachetools                4.2.2              pyhd3eb1b0_0
certifi                   2022.12.7        py37haa95532_0
cffi                      1.15.1           py37h2bbff1b_3
cfitsio                   3.470                h2bbff1b_7
charls                    2.2.0                h6c2663c_0
charset-normalizer        2.0.4              pyhd3eb1b0_0
click                     8.0.4            py37haa95532_0
click-plugins             1.1.1              pyhd3eb1b0_0
cligj                     0.7.2              pyhd3eb1b0_0
cloudpickle               2.0.0              pyhd3eb1b0_0
colorama                  0.4.6            py37haa95532_0
cryptography              38.0.1           py37h21b164f_0
cudatoolkit               10.1.243             h74a9793_0
cudnn                     7.6.5                cuda10.1_0
curl                      7.86.0               h2bbff1b_0
cycler                    0.11.0             pyhd3eb1b0_0
cytoolz                   0.12.0           py37h2bbff1b_0
dask-core                 2021.10.0          pyhd3eb1b0_0
expat                     2.4.9                h6c2663c_0
fftw                      3.3.9                h2bbff1b_1
flit-core                 3.6.0              pyhd3eb1b0_0
fonttools                 4.25.0             pyhd3eb1b0_0
freetype                  2.12.1               ha860e81_0
freexl                    1.0.6                h2bbff1b_0
frozenlist                1.3.3            py37h2bbff1b_0
fsspec                    2022.11.0        py37haa95532_0
gast                      0.2.2                    py37_0
gdal                      3.0.2            py37hb978731_4
geos                      3.8.0                h33f27b4_0
geotiff                   1.7.0                h4545760_0
giflib                    5.2.1                h62dcd97_0
glib                      2.69.1               h5dc1a3c_2
google-auth               2.6.0              pyhd3eb1b0_0
google-auth-oauthlib      0.4.4              pyhd3eb1b0_0
google-pasta              0.2.0              pyhd3eb1b0_0
grpcio                    1.42.0           py37hc60d5dd_0
gst-plugins-base          1.18.5               h9e645db_0
gstreamer                 1.18.5               hd78058f_0
h5py                      3.7.0            py37h3de5c98_0
hdf4                      4.2.13               h712560f_2
hdf5                      1.10.6               h1756f20_1
icc_rt                    2022.1.0             h6049295_2
icu                       58.2                 ha925a31_3
idna                      3.4              py37haa95532_0
imagecodecs               2021.8.26        py37hc0a7faf_1
imageio                   2.19.3           py37haa95532_0
importlib-metadata        4.11.3           py37haa95532_0
[... 113 lines cut ...]
tiledb                    2.3.3                h3649cd2_2
tk                        8.6.12               h2bbff1b_0
toml                      0.10.2             pyhd3eb1b0_0
toolz                     0.12.0           py37haa95532_0
tornado                   6.2              py37h2bbff1b_0
typing-extensions         4.4.0            py37haa95532_0
typing_extensions         4.4.0            py37haa95532_0
urllib3                   1.26.13          py37haa95532_0
vc                        14.2                 h21ff451_1
vs2015_runtime            14.27.29016          h5e58377_2
werkzeug                  0.16.1                     py_0
wheel                     0.37.1             pyhd3eb1b0_0
win_inet_pton             1.1.0            py37haa95532_0
wincertstore              0.2              py37haa95532_2
wrapt                     1.14.1           py37h2bbff1b_0
xerces-c                  3.2.4                hd77b12b_0
xz                        5.2.8                h8cc25b3_0
yaml                      0.2.5                he774522_0
yarl                      1.8.1            py37h2bbff1b_0
zfp                       0.5.5                hd77b12b_6
zipp                      3.11.0           py37haa95532_0
zlib                      1.2.13               h8cc25b3_0
zstd                      1.5.2                h19a0ad4_0
(ping)
```

CameronBodine, 2023-01-12:
Alrighty, can you post the contents of `PINGMapper.yml` inside ./PINGMapper/conda for me? Your env looks quite a bit different then my windows and ubuntu pc's.

CameronBodine, 2023-01-12:
For example, on windows:
```
(ping) PS C:\Users\bodin> conda list
# packages in environment at C:\Users\bodin\miniconda3\envs\ping:
#
# Name                    Version                   Build  Channel
_tflow_select             2.1.0                       gpu
abseil-cpp                20210324.2           hd77b12b_0
absl-py                   1.3.0            py39haa95532_0
affine                    2.3.0              pyhd3eb1b0_0
aiohttp                   3.8.3            py39h2bbff1b_0
aiosignal                 1.2.0              pyhd3eb1b0_0
astor                     0.8.1            py39haa95532_0
astunparse                1.6.3                      py_0
async-timeout             4.0.2            py39haa95532_0
attrs                     22.1.0           py39haa95532_0
blas                      1.0                         mkl
blinker                   1.4              py39haa95532_0
blosc                     1.21.0               h19a0ad4_1
bottleneck                1.3.5            py39h080aedc_0
brotli                    1.0.9                h2bbff1b_7
brotli-bin                1.0.9                h2bbff1b_7
brotlipy                  0.7.0           py39h2bbff1b_1003
bzip2                     1.0.8                he774522_0
ca-certificates           2022.10.11           haa95532_0
cachetools                4.2.2              pyhd3eb1b0_0
certifi                   2022.12.7        py39haa95532_0
cffi                      1.15.1           py39h2bbff1b_3
cfitsio                   3.470                h2bbff1b_7
charls                    2.2.0                h6c2663c_0
charset-normalizer        2.0.4              pyhd3eb1b0_0
click                     8.0.4            py39haa95532_0
click-plugins             1.1.1              pyhd3eb1b0_0
cligj                     0.7.2              pyhd3eb1b0_0
cloudpickle               2.0.0              pyhd3eb1b0_0
colorama                  0.4.6            py39haa95532_0
contourpy                 1.0.5            py39h59b6b97_0
cryptography              38.0.4           py39h21b164f_0
cudatoolkit               11.3.1               h59b6b97_2
cudnn                     8.2.1                cuda11.3_0
curl                      7.86.0               h2bbff1b_0
cycler                    0.11.0             pyhd3eb1b0_0
cytoolz                   0.12.0           py39h2bbff1b_0
dask-core                 2022.7.0         py39haa95532_0
expat                     2.4.9                h6c2663c_0
fftw                      3.3.9                h2bbff1b_1
flatbuffers               2.0.0                h6c2663c_0
flit-core                 3.6.0              pyhd3eb1b0_0
fonttools                 4.25.0             pyhd3eb1b0_0
freetype                  2.12.1               ha860e81_0
freexl                    1.0.6                h2bbff1b_0
frozenlist                1.3.3            py39h2bbff1b_0
fsspec                    2022.11.0        py39haa95532_0
gast                      0.4.0              pyhd3eb1b0_0
gdal                      3.0.2            py39hb978731_4
geos                      3.8.0                h33f27b4_0
geotiff                   1.7.0                h4545760_0
giflib                    5.2.1                h62dcd97_0
google-auth               2.6.0              pyhd3eb1b0_0
google-auth-oauthlib      0.4.1                      py_2
google-pasta              0.2.0              pyhd3eb1b0_0
grpcio                    1.42.0           py39hc60d5dd_0
h5py                      3.7.0            py39h3de5c98_0
hdf4                      4.2.13               h712560f_2
hdf5                      1.10.6               h1756f20_1
icc_rt                    2022.1.0             h6049295_2
icu                       68.1                 h6c2663c_0
idna                      3.4              py39haa95532_0
imagecodecs               2021.8.26        py39hc0a7faf_1
imageio                   2.19.3           py39haa95532_0
[... 109 lines cut ...]
tifffile                  2021.7.2           pyhd3eb1b0_2
tiledb                    2.3.3                h3649cd2_2
tk                        8.6.12               h2bbff1b_0
toolz                     0.12.0           py39haa95532_0
tornado                   6.2              py39h2bbff1b_0
typing_extensions         4.4.0            py39haa95532_0
tzdata                    2022g                h04d1e81_0
urllib3                   1.26.13          py39haa95532_0
vc                        14.2                 h21ff451_1
versioneer                0.28                     pypi_0    pypi
vs2015_runtime            14.27.29016          h5e58377_2
werkzeug                  2.2.2            py39haa95532_0
wheel                     0.35.1             pyhd3eb1b0_0
win_inet_pton             1.1.0            py39haa95532_0
wincertstore              0.2              py39haa95532_2
wrapt                     1.14.1           py39h2bbff1b_0
xerces-c                  3.2.4                hd77b12b_0
xz                        5.2.8                h8cc25b3_0
yaml                      0.2.5                he774522_0
yarl                      1.8.1            py39h2bbff1b_0
zfp                       0.5.5                hd77b12b_6
zipp                      3.11.0           py39haa95532_0
zlib                      1.2.13               h8cc25b3_0
zstd                      1.5.2                h19a0ad4_0
```

CameronBodine, 2023-01-12:
Here is something to try (since you figured out how to install the environment _way_ faster):

Remove current ping mapper env:
```
conda deactivate
conda env remove -n ping
```

Clean up:
```
conda clean -a
```

Update base conda environment:
```
conda update -n base conda
```

Then try reinstalling:
```
conda env create --file conda/PINGMapper.yml
```

bthetford, 2023-01-12:
```
$ cat PINGMapper.yml
name: ping
channels:
  - defaults
  - conda-forge
dependencies:
  - python
  - pandas
  - rasterio
  - tensorflow-gpu
  - pyproj
  - scikit-image
  - joblib
  - gdal
  - matplotlib
  - pip
  - pip:
      - psutil
```

bthetford, 2023-01-12:
I'll give those steps a shot.

bthetford, 2023-01-13:
I ran those steps you suggested and unfortunately got exactly the same error ending in `AttributeError: 'str' object has no attribute 'decode'`. Here's the package list since I cleaned and reinstalled the ping environment:
```
$ conda list
# packages in environment at C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping:
#
# Name                    Version                   Build  Channel
_tflow_select             2.1.0                       gpu
absl-py                   0.15.0             pyhd3eb1b0_0
affine                    2.3.0              pyhd3eb1b0_0
aiohttp                   3.8.3            py37h2bbff1b_0
aiosignal                 1.2.0              pyhd3eb1b0_0
astor                     0.8.1            py37haa95532_0
async-timeout             4.0.2            py37haa95532_0
asynctest                 0.13.0                     py_0
attrs                     22.1.0           py37haa95532_0
blas                      1.0                         mkl
blinker                   1.4              py37haa95532_0
blosc                     1.21.0               h19a0ad4_1
bottleneck                1.3.5            py37h080aedc_0
brotli                    1.0.9                h2bbff1b_7
brotli-bin                1.0.9                h2bbff1b_7
brotlipy                  0.7.0           py37h2bbff1b_1003
bzip2                     1.0.8                he774522_0
ca-certificates           2022.10.11           haa95532_0
cachetools                4.2.2              pyhd3eb1b0_0
certifi                   2022.12.7        py37haa95532_0
cffi                      1.15.1           py37h2bbff1b_3
cfitsio                   3.470                h2bbff1b_7
charls                    2.2.0                h6c2663c_0
charset-normalizer        2.0.4              pyhd3eb1b0_0
click                     8.0.4            py37haa95532_0
click-plugins             1.1.1              pyhd3eb1b0_0
cligj                     0.7.2              pyhd3eb1b0_0
cloudpickle               2.0.0              pyhd3eb1b0_0
colorama                  0.4.6            py37haa95532_0
cryptography              38.0.4           py37h21b164f_0
cudatoolkit               10.1.243             h74a9793_0
cudnn                     7.6.5                cuda10.1_0
curl                      7.86.0               h2bbff1b_0
cycler                    0.11.0             pyhd3eb1b0_0
cytoolz                   0.12.0           py37h2bbff1b_0
dask-core                 2021.10.0          pyhd3eb1b0_0
expat                     2.4.9                h6c2663c_0
fftw                      3.3.9                h2bbff1b_1
flit-core                 3.6.0              pyhd3eb1b0_0
fonttools                 4.25.0             pyhd3eb1b0_0
freetype                  2.12.1               ha860e81_0
freexl                    1.0.6                h2bbff1b_0
frozenlist                1.3.3            py37h2bbff1b_0
fsspec                    2022.11.0        py37haa95532_0
gast                      0.2.2                    py37_0
gdal                      3.0.2            py37hb978731_4
geos                      3.8.0                h33f27b4_0
geotiff                   1.7.0                h4545760_0
giflib                    5.2.1                h62dcd97_0
glib                      2.69.1               h5dc1a3c_2
google-auth               2.6.0              pyhd3eb1b0_0
google-auth-oauthlib      0.4.4              pyhd3eb1b0_0
google-pasta              0.2.0              pyhd3eb1b0_0
grpcio                    1.42.0           py37hc60d5dd_0
gst-plugins-base          1.18.5               h9e645db_0
gstreamer                 1.18.5               hd78058f_0
h5py                      3.7.0            py37h3de5c98_0
hdf4                      4.2.13               h712560f_2
hdf5                      1.10.6               h1756f20_1
icc_rt                    2022.1.0             h6049295_2
icu                       58.2                 ha925a31_3
idna                      3.4              py37haa95532_0
imagecodecs               2021.8.26        py37hc0a7faf_1
imageio                   2.19.3           py37haa95532_0
[... 114 lines cut ...]
tiledb                    2.3.3                h3649cd2_2
tk                        8.6.12               h2bbff1b_0
toml                      0.10.2             pyhd3eb1b0_0
toolz                     0.12.0           py37haa95532_0
tornado                   6.2              py37h2bbff1b_0
typing-extensions         4.4.0            py37haa95532_0
typing_extensions         4.4.0            py37haa95532_0
urllib3                   1.26.13          py37haa95532_0
vc                        14.2                 h21ff451_1
vs2015_runtime            14.27.29016          h5e58377_2
werkzeug                  0.16.1                     py_0
wheel                     0.37.1             pyhd3eb1b0_0
win_inet_pton             1.1.0            py37haa95532_0
wincertstore              0.2              py37haa95532_2
wrapt                     1.14.1           py37h2bbff1b_0
xerces-c                  3.2.4                hd77b12b_0
xz                        5.2.8                h8cc25b3_0
yaml                      0.2.5                he774522_0
yarl                      1.8.1            py37h2bbff1b_0
zfp                       0.5.5                hd77b12b_6
zipp                      3.11.0           py37haa95532_0
zlib                      1.2.13               h8cc25b3_0
zstd                      1.5.2                h19a0ad4_0
(ping)
```

bthetford, 2023-01-13:
PS here's every exception that comes out:
```
Automatically detecting shadows for 48 chunks:
MODE: 1 | Remove all shadows...
joblib.externals.loky.process_executor._RemoteTraceback:
"""
Traceback (most recent call last):
  File "src\class_portstarObj.py", line 483, in _initModel
    model = tf.keras.models.load_model(self.weights)
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\tensorflow_core\python\keras\saving\save.py", line 146, in load_model
    return hdf5_format.load_model_from_hdf5(filepath, custom_objects, compile)
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\tensorflow_core\python\keras\saving\hdf5_format.py", line 166, in load_model_from_hdf5
    model_config = json.loads(model_config.decode('utf-8'))
AttributeError: 'str' object has no attribute 'decode'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\joblib\externals\loky\process_executor.py", line 436, in _process_worker
    r = call_item()
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\joblib\externals\loky\process_executor.py", line 288, in __call__
    return self.fn(*self.args, **self.kwargs)
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\joblib\_parallel_backends.py", line 595, in __call__
    return self.func(*args, **kwargs)
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\joblib\parallel.py", line 264, in __call__
    for func, args, kwargs in self.items]
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\joblib\parallel.py", line 264, in <listcomp>
    for func, args, kwargs in self.items]
  File "src\class_portstarObj.py", line 1545, in _detectShadow
    self.shadowModel = self._initModel(USE_GPU)
  File "src\class_portstarObj.py", line 486, in _initModel
    model.load_weights(self.weights)
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\tensorflow_core\python\keras\engine\training.py", line 234, in load_weights
    return super(Model, self).load_weights(filepath, by_name, skip_mismatch)
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\tensorflow_core\python\keras\engine\network.py", line 1222, in load_weights
    hdf5_format.load_weights_from_hdf5_group(f, self.layers)
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\tensorflow_core\python\keras\saving\hdf5_format.py", line 651, in load_weights_from_hdf5_group
    original_keras_version = f.attrs['keras_version'].decode('utf8')
AttributeError: 'str' object has no attribute 'decode'
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "main.py", line 123, in <module>
    read_master_func(**params)
  File "src\main_readFiles.py", line 829, in read_master_func
    r = Parallel(n_jobs=np.min([len(chunks), threadCnt]), verbose=10)(delayed(psObj._detectShadow)(remShadow, int(chunk), USE_GPU, False, tileFile) for chunk in chunks)
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\joblib\parallel.py", line 1061, in __call__
    self.retrieve()
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\joblib\parallel.py", line 938, in retrieve
    self._output.extend(job.get(timeout=self.timeout))
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\site-packages\joblib\_parallel_backends.py", line 542, in wrap_future_result
    return future.result(timeout=timeout)
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\concurrent\futures\_base.py", line 435, in result
    return self.__get_result()
  File "C:\Users\b.thetford\AppData\Local\miniconda3\envs\ping\lib\concurrent\futures\_base.py", line 384, in __get_result
    raise self._exception
AttributeError: 'str' object has no attribute 'decode'
(ping)
```

CameronBodine, 2023-01-13:
Your tensorflow package still isn't updating from 2.1 to 2.6 for some reason. With the env activated, can you try `conda update tensorflow-gpu=2.6`?

CameronBodine, 2023-01-13:
I'd also be interested to see what package versions you have for your base env.

CameronBodine, 2023-01-13:
I was able to successfully recreate your error! 

# TLDR:
Running `conda install tensorflow-gpu=2.6 --force-reinstall` fixed the issue on my end.

# **Don't do this part! Just for documentation**

I changed the PINGMapper.yml to be:
```
name: ping-tf-2.1
channels:
  - defaults
  - conda-forge
dependencies:
  - python
  - pandas
  - rasterio
  - tensorflow-gpu=2.1
  - pyproj
  - scikit-image
  - joblib
  - gdal
  - matplotlib
  - pip
  - pip:
      - psutil
```

I then install the new environment:
```
conda env create -f .\PINGMapper_21.yml
```

This gives me the following package version, close (I think the exact same) to your env:
```
(ping-tf-2.1) PS C:\Users\bodin\Python\PINGMapper> conda list
# packages in environment at C:\Users\bodin\miniconda3\envs\ping-tf-2.1:
#
# Name                    Version                   Build  Channel
_tflow_select             2.1.0                       gpu
absl-py                   0.15.0             pyhd3eb1b0_0
affine                    2.3.0              pyhd3eb1b0_0
aiohttp                   3.8.3            py37h2bbff1b_0
aiosignal                 1.2.0              pyhd3eb1b0_0
astor                     0.8.1            py37haa95532_0
async-timeout             4.0.2            py37haa95532_0
asynctest                 0.13.0                     py_0
attrs                     22.1.0           py37haa95532_0
blas                      1.0                         mkl
blinker                   1.4              py37haa95532_0
blosc                     1.21.0               h19a0ad4_1
bottleneck                1.3.5            py37h080aedc_0
brotli                    1.0.9                h2bbff1b_7
brotli-bin                1.0.9                h2bbff1b_7
brotlipy                  0.7.0           py37h2bbff1b_1003
bzip2                     1.0.8                he774522_0
ca-certificates           2022.10.11           haa95532_0
cachetools                4.2.2              pyhd3eb1b0_0
certifi                   2022.12.7        py37haa95532_0
cffi                      1.15.1           py37h2bbff1b_3
cfitsio                   3.470                h2bbff1b_7
charls                    2.2.0                h6c2663c_0
charset-normalizer        2.0.4              pyhd3eb1b0_0
click                     8.0.4            py37haa95532_0
click-plugins             1.1.1              pyhd3eb1b0_0
cligj                     0.7.2              pyhd3eb1b0_0
cloudpickle               2.0.0              pyhd3eb1b0_0
colorama                  0.4.6            py37haa95532_0
cryptography              38.0.4           py37h21b164f_0
cudatoolkit               10.1.243             h74a9793_0
[... 431 lines cut ...]
typing_extensions         4.4.0            py37haa95532_0
ucrt                      10.0.20348.0         haa95532_0
urllib3                   1.26.13          py37haa95532_0
vc                        14.2                 h21ff451_1
vs2015_runtime            14.34.31931         h4c5c07a_10    conda-forge
werkzeug                  2.2.2            py37haa95532_0
wheel                     0.35.1             pyhd3eb1b0_0
win_inet_pton             1.1.0            py37haa95532_0
wincertstore              0.2              py37haa95532_2
wrapt                     1.14.1           py37h2bbff1b_0
xerces-c                  3.2.4                hd77b12b_0
xz                        5.2.8                h8cc25b3_0
yaml                      0.2.5                he774522_0
yarl                      1.8.1            py37h2bbff1b_0
zfp                       0.5.5                hd77b12b_6
zipp                      3.11.0           py37haa95532_0
zlib                      1.2.13               hcfcfb64_4    conda-forge
zstd                      1.5.2                h19a0ad4_0
(ping-tf-2.1) PS C:\Users\bodin\Python\PINGMapper>
```

Everything ran as expected after re-running the test script!!!

# Note
You may want to consider updating your python version as well (python=3.9) to try and get your environment as close to mine as possible. I'm still stumped why PING-Mapper didn't install to the same versions as mine. Perhaps you installed an older version of miniconda3? Dunno. But hopefully updating tensorflow will work for you and you can keep on mapping! Please let me know how it goes.

bthetford, 2023-01-13:
Hurray thanks Cam! I'm going to try this now. That is weird, I got the latest miniconda3 version on the site, and it says my installed conda version is 22.11.1. 
I'll try the tensorflow update alone before the python update to see if there's any difference. 
Note: After running those steps you posted (delete ping env, update base env, `conda clean -a`, and remake ping env), I got this error:
```
InvalidArchiveError("Error with archive C:\\Users\\b.thetford\\AppData\\Local\\miniconda3\\pkgs\\tensorflow-base-2.3.0-eigen_py38h75a453f_0.conda.  You probably need to delete and re-download or re-create this file.  Message was:\n\nfailed with error: [Errno 2] No such file or directory: 'C:\\\\Users\\\\b.thetford\\\\AppData\\\\Local\\\\miniconda3\\\\pkgs\\\\tensorflow-base-2.3.0-eigen_py38h75a453f_0\\\\Lib\\\\site-packages\\\\tensorflow\\\\include\\\\external\\\\llvm-project\\\\mlir\\\\_virtual_includes\\\\DerivedAttributeOpInterfaceIncGen\\\\mlir\\\\Interfaces\\\\DerivedAttributeOpInterface.cpp.inc'")
```
I tried to run `conda env create --file conda/PINGMapper.yml` again and nothing changed. Then, I ran `conda clean -a` again, retired `conda env create --file conda/PINGMapper.yml`, and it worked. Super strange! Especially since I ran `conda clean -a` right after updating the base env also.

CameronBodine, 2023-01-13:
In my experience, uninstalling Anaconda/miniconda doesn't always remove everything. So, in your case, I bet there were old files that were causing you to revert to older package versions. When you told me about `libmamba`, I did uninstall miniconda, and found that there was still a miniconda3 folder in my user directory. In addition, there were still Anaconda/miniconda start menu items with broken links, as I noted on [this issue](https://github.com/ContinuumIO/anaconda-issues/issues/352#issuecomment-1377986494). After I removed the start menu folder as well as the miniconda3 folder in my user directory, miniconda installation went smoothly.

I think the `conda clean -a` step is critical and should be run anytime you are updating or installing a new env.

Welcome to the world of open-source computing! You truly get what you pay for ;-)

Let me know once you successfully run the test script, and we can close this issue as a success! 🤜🪵🤛

bthetford, 2023-01-17:
Thanks for all your help Cam, I just got loaded with a ton of multibeam work but I'll get back to you on this soon!
Beau

CameronBodine, 2023-02-01:
Hi @bthetford, I was wondering if you had a chance to check the new changes? If they work for you, I would like to close this issue. Thanks.

bthetford, 2023-02-01:
Hi Cameron, I did not get it working but haven't had any time to test it! I can try to look at it again maybe Friday.
Thanks,
Beau

bthetford, 2023-02-10:
I got back to this yesterday and tried to force reinstall tensorflow GPU, and as you said my python version 3.10 is an issue. Going to use 3.9 and retry.

```
$ conda install tensorflow-gpu=2.6 --force-reinstall
Retrieving notices: ...working... done
Collecting package metadata (current_repodata.json): ...working... done
Solving environment: ...working... failed with initial frozen solve. Retrying with flexible solve.
Solving environment: ...working... failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): ...working... done
Solving environment: ...working... failed with initial frozen solve. Retrying with flexible solve.
Solving environment: ...working...
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed

UnsatisfiableError: The following specifications were found
to be incompatible with the existing python installation in your environment:

Specifications:

  - tensorflow-gpu=2.6 -> python[version='3.7.*|3.8.*|3.9.*']

Your python: python=3.10

If python is on the left-most side of the chain, that's the version you've asked for.
When python appears to the right, that indicates that the thing on the left is somehow
not available for the python version you are constrained to. Note that conda will not
change your python version to a different minor version unless you explicitly specify
that.

Note that strict channel priority may have removed packages required for satisfiability.

```

CameronBodine, 2023-02-20:
Hi @bthetford, I have updated `PINGMapper.yml` installation procedure with 65046d9 today. This may help with your installation. This update uses `tensorflow` instead of `tensorflow-gpu`, which others were having issues with (see #60). If you could let me know if you are able to install and run PING-Mapper, I'd appreciate it.

bthetford, 2023-02-21:
Hurray! It's working now. Tried both the test file and `main.py` on my own recording.

closed 2023-02-21 by CameronBodine