CameronBodine/PINGMapper #56: Multiple processing of sonar data - Issues

AngelDelacroix, 2023-01-31:
33 areas of interest have been surveyed.
Overlapping transects have been attempted.

Of the problems encountered:
- There is parallel black lines in part of images;
- Overlapping transects do not seem to be recognized and seem to be treated as two distinct entity;
- Some projects did not complete (IndexError: index _A_ is out of bounds for axis _B_ with size _C_, UnboundLocalError: local variable 'noDat' referenced before assignment, AttributeError: 'portstarObj' object has no attribute 'port')

In your article (PING-Mapper: Open-Source Software for Automated Benthic Imaging and Mapping Using Recreation-Grade Sonar), if I remember correctly, you wrote that files can be corrupted or missing. I don't know if it was the case here.

CameronBodine, 2023-01-31:
Thanks for trying out PING-Mapper @AngelDelacroix!

- Can you provide a sample recording file for the parallel black lines issue? This is likely a problem with the recording itself, but I will take a look.
- PING-Mapper does not currently support mosaicing multiple sonar transects into a single file, and is outside the scope of this software. This is primarily due to the fact that **a)** co-locating pixels which should be coincident is near impossible with coordinates from a recreation-grade gps in addition to **b)** lack of an IMU which would allow for correction of coordinates. If you do need to mosaic multiple transects, you could investigate [georefrencing](https://pro.arcgis.com/en/pro-app/latest/help/data/imagery/overview-of-georeferencing.htm) routines assuming you have identifiable targets in both transects, then mosaic the corrected images in a [mosaic dataset](https://pro.arcgis.com/en/pro-app/latest/help/data/imagery/mosaic-datasets.htm) or similar in QGIS.
- If you post a sample recording from the set which would not complete processing, I can try it on my end. In the future, if you post screen shots of the consol output, that is helpful!

You can upload your bad recordings [here](https://drive.google.com/drive/folders/1-0uBe4E2C5O3ME6gLuQ1DG-w05Egcb33?usp=share_link).

Files certainly become corrupt due to power issues during the survey. I will take a look at the recordings and post what I find out.

AngelDelacroix, 2023-02-02:
Files have been uploaded.
The parallel black lines come from the R00007.
Logs have also been provided.

CameronBodine, 2023-02-02:
Reporting error outputs here for reference:

# R00040
```
Exporting sonogram tiles:

        Exporting 102 sonograms for ss_port
[Parallel(n_jobs=8)]: Using backend LokyBackend with 8 concurrent workers.
[Parallel(n_jobs=8)]: Done   2 tasks      | elapsed:    8.7s
[Parallel(n_jobs=8)]: Done   9 tasks      | elapsed:   12.4s
joblib.externals.loky.process_executor._RemoteTraceback:
"""
Traceback (most recent call last):
  File "C:\Users\Utilisateur\miniconda3\envs\ping\lib\site-packages\joblib\externals\loky\process_executor.py", line 436, in _process_worker
    r = call_item()
  File "C:\Users\Utilisateur\miniconda3\envs\ping\lib\site-packages\joblib\externals\loky\process_executor.py", line 288, in __call__
    return self.fn(*self.args, **self.kwargs)
  File "C:\Users\Utilisateur\miniconda3\envs\ping\lib\site-packages\joblib\_parallel_backends.py", line 595, in __call__
    return self.func(*args, **kwargs)
  File "C:\Users\Utilisateur\miniconda3\envs\ping\lib\site-packages\joblib\parallel.py", line 263, in __call__
    return [func(*args, **kwargs)
  File "C:\Users\Utilisateur\miniconda3\envs\ping\lib\site-packages\joblib\parallel.py", line 263, in <listcomp>
    return [func(*args, **kwargs)
  File "C:\Users\Utilisateur\PINGMapper\src\class_sonObj.py", line 1392, in _exportTiles
    self._WCR_SRC(sonMeta) # Remove water column and redistribute ping returns based on FlatBottom assumption
  File "C:\Users\Utilisateur\PINGMapper\src\class_sonObj.py", line 1502, in _WCR_SRC
    pingDat[srcIndex] = intensity # Store intensity at appropriate horizontal range
IndexError: index -2147483648 is out of bounds for axis 0 with size 2404
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\Utilisateur\PINGMapper\main.py", line 123, in <module>
    read_master_func(**params)
  File "C:\Users\Utilisateur\PINGMapper\src\main_readFiles.py", line 871, in read_master_func
    Parallel(n_jobs= np.min([len(chunks), threadCnt]), verbose=10)(delayed(son._exportTiles)(i, tileFile) for i in chunks)
  File "C:\Users\Utilisateur\miniconda3\envs\ping\lib\site-packages\joblib\parallel.py", line 1061, in __call__
    self.retrieve()
  File "C:\Users\Utilisateur\miniconda3\envs\ping\lib\site-packages\joblib\parallel.py", line 938, in retrieve
    self._output.extend(job.get(timeout=self.timeout))
  File "C:\Users\Utilisateur\miniconda3\envs\ping\lib\site-packages\joblib\_parallel_backends.py", line 542, in wrap_future_result
    return future.result(timeout=timeout)
  File "C:\Users\Utilisateur\miniconda3\envs\ping\lib\concurrent\futures\_base.py", line 439, in result
    return self.__get_result()
  File "C:\Users\Utilisateur\miniconda3\envs\ping\lib\concurrent\futures\_base.py", line 391, in __get_result
    raise self._exception
IndexError: index -2147483648 is out of bounds for axis 0 with size 2404
```

Above is same error as #49.

# R00042
```
Locating missing pings and adding NoData...
[Parallel(n_jobs=8)]: Using backend LokyBackend with 8 concurrent workers.
[Parallel(n_jobs=8)]: Done   2 out of   9 | elapsed:    2.9s remaining:   10.5s
[Parallel(n_jobs=8)]: Done   3 out of   9 | elapsed:    2.9s remaining:    5.9s
[Parallel(n_jobs=8)]: Done   4 out of   9 | elapsed:    2.9s remaining:    3.7s
[Parallel(n_jobs=8)]: Done   5 out of   9 | elapsed:    2.9s remaining:    2.3s
[Parallel(n_jobs=8)]: Done   6 out of   9 | elapsed:    2.9s remaining:    1.4s
[Parallel(n_jobs=8)]: Done   7 out of   9 | elapsed:    3.0s remaining:    0.8s
[Parallel(n_jobs=8)]: Done   9 out of   9 | elapsed:    3.0s remaining:    0.0s
joblib.externals.loky.process_executor._RemoteTraceback:
"""
Traceback (most recent call last):
  File "C:\Users\Utilisateur\miniconda3\envs\ping\lib\site-packages\joblib\externals\loky\process_executor.py", line 436, in _process_worker
    r = call_item()
  File "C:\Users\Utilisateur\miniconda3\envs\ping\lib\site-packages\joblib\externals\loky\process_executor.py", line 288, in __call__
    return self.fn(*self.args, **self.kwargs)
[... 45 lines cut ...]

Traceback (most recent call last):
  File "C:\Users\Utilisateur\PINGMapper\main.py", line 132, in <module>
    rectify_master_func(**params)
  File "C:\Users\Utilisateur\PINGMapper\src\main_rectify.py", line 244, in rectify_master_func
    sDF.at[curRow, "lons"] = lastRow["lons"]
  File "C:\Users\Utilisateur\miniconda3\envs\ping\lib\site-packages\pandas\core\frame.py", line 3805, in __getitem__
    indexer = self.columns.get_loc(key)
  File "C:\Users\Utilisateur\miniconda3\envs\ping\lib\site-packages\pandas\core\indexes\base.py", line 3805, in get_loc
    raise KeyError(key) from err
KeyError: 'lons'
```
This is a new problem.

# R00044
```
Getting DAT Metadata...
Traceback (most recent call last):
  File "C:\Users\Utilisateur\PINGMapper\main.py", line 123, in <module>
    read_master_func(**params)
  File "C:\Users\Utilisateur\PINGMapper\src\main_readFiles.py", line 246, in read_master_func
    son = sonObj(sonFiles[0], humFile, projDir, tempC, nchunk) # Initialize sonObj instance from first sonar beam
IndexError: list index out of range
```
DAT may be corrupt.

CameronBodine, 2023-02-02:
[image]

Shadows are being removed from the recording by specifying `remShadow=1` or `remShadow=2`, so there is NoData on the lee of the sand ripple. If this is a problem, you can specify `remShadow=0` to leave the shadows.

---
[image]
The NoData horizontal stripes are from the sonar unit not pinging during that period for unknown reasons. Specifying `fixNoDat = True` has PING-Mapper automatically detect portions of the sonar recording which have missing data and fill those regions with NoData. You can instead set `fixNoDat = False` and the sonar returns will be stretched over that region. You can try both and see which you prefer. I personally prefer filling with NoData because the instrument wasn't collecting data over these areas.

Looking into other issues now.

CameronBodine, 2023-02-02:
R00042 is a very short file (~4 seconds) so that's why the software cannot process it. R00043 doesn't appear to have any sonar data available, as the tiles are mostly black except along the trackline. R00044 doesn't have any `SON` or `IDX` files in the folder, which would explain the error you are seeing. I was able to process R00040.

I did make some changes to the code as your files did help me spot some issues. If you navigate to the `PINGMapper` directory in your terminal and run the command below, you will be up to date with the latest changes.

```
git fetch
```

If you can let me know if you are able to process R00040, then I will close this issue as the remaining problems processing the recordings is due to issues with the recording itself.

Thanks for all the info!

AngelDelacroix, 2023-02-08:
I get the following error:

(base) PS C:\Users\Utilisateur\PINGMapper> python main.py
Traceback (most recent call last):
  File "C:\Users\Utilisateur\PINGMapper\main.py", line 33, in <module>
    from funcs_common import *
  File "C:\Users\Utilisateur\PINGMapper\src\funcs_common.py", line 30, in <module>
    from joblib import Parallel, delayed, cpu_count
ModuleNotFoundError: No module named 'joblib'

CameronBodine, 2023-02-08:
Make sure you have the conda environment activated:
'''
conda activate ping
'''
Then try running the script again. 

If that doesn't work, reinstall the environment according to the instructions on the website. 

https://cameronbodine.github.io/PINGMapper/docs/gettingstarted/Installation.html

CameronBodine, 2023-02-20:
Hi @AngelDelacroix, can you provide an update? Were you able to run PING-Mapper as expected?

CameronBodine, 2023-03-23:
I'm closing this issue as stale. If the issue persists, feel free to re-open.