CameronBodine/PINGMapper #58: Tensorflow install on windows

CameronBodine, 2023-02-15:
Currently Tensorflow 2.11 does not support GPU on windows. Unsure if this is an issue yet. But when I implement `segformer` model, I need a Tensorflow version >2.3 as I had errors with predicting using the CPU. It's possible to install Tensorflow=2.10 following the [documentation](https://www.tensorflow.org/install/pip#windows-native) with:

```
conda create --name ping python=3.9
conda activate ping
conda install -c conda-forge cudatoolkit=11.2 cudnn=8.1.0
pip install "tensorflow<2.11"
```

Run the following to see if a tensor is returned:
```
python -c "import tensorflow as tf; print(tf.reduce_sum(tf.random.normal([1000, 1000])))"
```

Run the following to see if Tensorflow can find system GPU's:
```
python -c "import tensorflow as tf; print(tf.config.list_physical_devices('GPU'))"
```

This will become relevant with #39 is implemented.

CameronBodine, 2023-02-20:
I changed installation of `tensorflow-gpu` to `tensorflow`. I haven't been able to predict in parallel using the GPU, so there is no need to have that package. I'm now also installing tensorflow using `pip` instead of `conda`. On Windows, this allows the `ping` environment to using `python=3.11` and `tensorflow=2.11`. This may be modified once I implement `SegFormer` model architecture for substrate mapping (#39).

CameronBodine, 2023-03-28:
Seems to work. Closing the issue.

closed 2023-03-28 by CameronBodine

dbuscombe-usgs, 2023-03-28:
famous last words ...