CameronBodine/PINGMapper #60: MacOS - unable to install

mjdikun, 2023-02-19:
I'm trying to install this package on MacOS 13.

But can't install tensor flow-gpu, the tensor flow documentation says there is no GPU support on MacOs.

would it be possible to release a version for Mac that doesn't depend on GPU support?

Or is there another workaround to get it working on Mac?

CameronBodine, 2023-02-19:
Good to know. I haven't tested on Mac's yet. If you go into the
PINGMapper.yml inside the conda folder, you can change the ‘tensorflow-gpu’
to ‘tensorflow’ then retry the install. Let me know how it goes!--

*Cameron S. Bodine*
PhD Candidate
Ecological and Environmental Informatics - T3 Program
<https://ecoinfo.nau.edu/index.php/nrt/>
The School of Informatics, Computing, and Cyber Systems (SICCS)
Northern Arizona University
Orcid <https://orcid.org/0000-0002-1623-3920> | GitHub
<https://github.com/CameronBodine> | LinkedIn
<https://www.linkedin.com/in/cameronbodine/> | Twitter
<https://twitter.com/bodine_cs>

mjdikun, 2023-02-19:
> Good to know. I haven't tested on Mac's yet. If you go into the PINGMapper.yml inside the conda folder, you can change the ‘tensorflow-gpu’ to ‘tensorflow’ then retry the install. Let me know how it [...]

That's exactly what I have been trying, create env just finished and the first test ran successfully.

CameronBodine, 2023-02-19:
Cool! I will change tensorflow-gpu to tensorflow as the gpu version isn't necessary.

mjdikun, 2023-02-19:
> Cool! I will change tensorflow-gpu to tensorflow as the gpu version isn't necessary.

test 2 still running, off to do treadmill, let you know later.  thanks.

mjdikun, 2023-02-19:
Running fine on the test samples.

Tried running it on one of my sonar recordings. 

I get a Maximum Supported image dimension of 65500 pixels.

Looks like it's a JPG limit.

Any way to change to a different format from the config?

mjdikun, 2023-02-19:
Changed the config file from .jpg to .tiff, don't know if that should work or not... 

but now I get a: OverflowError: cannot convert float infinity to integer

at the same place, exporting substrate files for labeling

Exporting 35 label-ready sonograms for ss_port
[Parallel(n_jobs=12)]: Using backend LokyBackend with 12 concurrent workers.
[Parallel(n_jobs=12)]: Done   1 tasks      | elapsed:    3.5s
[Parallel(n_jobs=12)]: Done   8 tasks      | elapsed:    7.3s
[Parallel(n_jobs=12)]: Done  16 out of  35 | elapsed:   18.2s remaining:   21.6s
[Parallel(n_jobs=12)]: Done  20 out of  35 | elapsed:   18.8s remaining:   14.1s
[Parallel(n_jobs=12)]: Done  24 out of  35 | elapsed:   30.5s remaining:   14.0s
[Parallel(n_jobs=12)]: Done  28 out of  35 | elapsed:   31.3s remaining:    7.8s
[Parallel(n_jobs=12)]: Done  32 out of  35 | elapsed:  1.3min remaining:    7.1s
joblib.externals.loky.process_executor._RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/site-packages/joblib/externals/loky/process_executor.py", line 436, in _process_worker
    r = call_item()
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/site-packages/joblib/externals/loky/process_executor.py", line 288, in __call__
    return self.fn(*self.args, **self.kwargs)
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/site-packages/joblib/_parallel_backends.py", line 595, in __call__
    return self.func(*args, **kwargs)
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/site-packages/joblib/parallel.py", line 263, in __call__
    return [func(*args, **kwargs)
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/site-packages/joblib/parallel.py", line 263, in <listcomp>
    return [func(*args, **kwargs)
  File "/Users/mdikun/wickedwaypoints.com/PINGMapper/src/class_sonObj.py", line 1752, in _exportLblTiles
    sonDat = resize(sonDat,
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/site-packages/skimage/transform/_warps.py", line 181, in resize
    image = ndi.gaussian_filter(image, anti_aliasing_sigma,
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/site-packages/scipy/ndimage/_filters.py", line 368, in gaussian_filter
    gaussian_filter1d(input, sigma, axis, order, output,
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/site-packages/scipy/ndimage/_filters.py", line 269, in gaussian_filter1d
    lw = int(truncate * sd + 0.5)
OverflowError: cannot convert float infinity to integer
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/Users/mdikun/wickedwaypoints.com/PINGMapper/test_PINGMapper.py", line 159, in <module>
    read_master_func(**params)
  File "/Users/mdikun/wickedwaypoints.com/PINGMapper/src/main_readFiles.py", line 908, in read_master_func
    Parallel(n_jobs= np.min([len(chunks), threadCnt]), verbose=10)(delayed(son._exportLblTiles)(i, lbl_set, spdCor, maxCrop, tileFile) for i in chunks)
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/site-packages/joblib/parallel.py", line 1061, in __call__
    self.retrieve()
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/site-packages/joblib/parallel.py", line 938, in retrieve
    self._output.extend(job.get(timeout=self.timeout))
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/site-packages/joblib/_parallel_backends.py", line 542, in wrap_future_result
    return future.result(timeout=timeout)
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/concurrent/futures/_base.py", line 439, in result
    return self.__get_result()
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/concurrent/futures/_base.py", line 391, in __get_result
    raise self._exception
OverflowError: cannot convert float infinity to integer

mjdikun, 2023-02-19:
The recording is uploading to your dropbox

The recording only has 1 short section in it where you can see something.

Came from a Gen 2 Helix 

Was testing the engine rebuild yesterday and took a small recording, but was recording the whole time even while on plane going 60...

CameronBodine, 2023-02-20:
I'm closing this issue as @mjdikun was able to install `tensorflow` instead of `tensorflow-gpu` on MacOS. This is related to #58. The solution will be to replace `tensorflow-gpu` with `tensorflow` in the `PINGMapper.yml` file.

I'm addressing the subsequent issue with exporting sonar imagery for labeling in #61.

closed 2023-02-20 by CameronBodine