CameronBodine/PINGMapper #61: Substrate label export error

CameronBodine, 2023-02-19:
Changed the config file from .jpg to .tiff, don't know if that should work or not... 

but now I get a: OverflowError: cannot convert float infinity to integer

at the same place, exporting substrate files for labeling

Exporting 35 label-ready sonograms for ss_port
[Parallel(n_jobs=12)]: Using backend LokyBackend with 12 concurrent workers.
[Parallel(n_jobs=12)]: Done   1 tasks      | elapsed:    3.5s
[Parallel(n_jobs=12)]: Done   8 tasks      | elapsed:    7.3s
[Parallel(n_jobs=12)]: Done  16 out of  35 | elapsed:   18.2s remaining:   21.6s
[Parallel(n_jobs=12)]: Done  20 out of  35 | elapsed:   18.8s remaining:   14.1s
[Parallel(n_jobs=12)]: Done  24 out of  35 | elapsed:   30.5s remaining:   14.0s
[Parallel(n_jobs=12)]: Done  28 out of  35 | elapsed:   31.3s remaining:    7.8s
[Parallel(n_jobs=12)]: Done  32 out of  35 | elapsed:  1.3min remaining:    7.1s
joblib.externals.loky.process_executor._RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/site-packages/joblib/externals/loky/process_executor.py", line 436, in _process_worker
    r = call_item()
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/site-packages/joblib/externals/loky/process_executor.py", line 288, in __call__
    return self.fn(*self.args, **self.kwargs)
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/site-packages/joblib/_parallel_backends.py", line 595, in __call__
    return self.func(*args, **kwargs)
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/site-packages/joblib/parallel.py", line 263, in __call__
    return [func(*args, **kwargs)
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/site-packages/joblib/parallel.py", line 263, in <listcomp>
    return [func(*args, **kwargs)
  File "/Users/mdikun/wickedwaypoints.com/PINGMapper/src/class_sonObj.py", line 1752, in _exportLblTiles
    sonDat = resize(sonDat,
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/site-packages/skimage/transform/_warps.py", line 181, in resize
    image = ndi.gaussian_filter(image, anti_aliasing_sigma,
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/site-packages/scipy/ndimage/_filters.py", line 368, in gaussian_filter
    gaussian_filter1d(input, sigma, axis, order, output,
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/site-packages/scipy/ndimage/_filters.py", line 269, in gaussian_filter1d
    lw = int(truncate * sd + 0.5)
OverflowError: cannot convert float infinity to integer
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/Users/mdikun/wickedwaypoints.com/PINGMapper/test_PINGMapper.py", line 159, in <module>
    read_master_func(**params)
  File "/Users/mdikun/wickedwaypoints.com/PINGMapper/src/main_readFiles.py", line 908, in read_master_func
    Parallel(n_jobs= np.min([len(chunks), threadCnt]), verbose=10)(delayed(son._exportLblTiles)(i, lbl_set, spdCor, maxCrop, tileFile) for i in chunks)
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/site-packages/joblib/parallel.py", line 1061, in __call__
    self.retrieve()
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/site-packages/joblib/parallel.py", line 938, in retrieve
    self._output.extend(job.get(timeout=self.timeout))
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/site-packages/joblib/_parallel_backends.py", line 542, in wrap_future_result
    return future.result(timeout=timeout)
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/concurrent/futures/_base.py", line 439, in result
    return self.__get_result()
  File "/Users/mdikun/miniconda3/envs/ping/lib/python3.9/concurrent/futures/_base.py", line 391, in __get_result
    raise self._exception
OverflowError: cannot convert float infinity to integer

_Originally posted by @mjdikun in https://github.com/CameronBodine/PINGMapper/issues/60#issuecomment-1435872770_

CameronBodine, 2023-02-19:
@mjdikun Thanks for the console output. That helps a lot. There could be an issue with the recording. Would you be able to upload the recording so I can have a look?

https://drive.google.com/drive/folders/1-0uBe4E2C5O3ME6gLuQ1DG-w05Egcb33

CameronBodine, 2023-02-20:
I'll take a look tomorrow. Thanks.

CameronBodine, 2023-02-20:
Hi @mjdikun, I have looked at your data and have a better understanding of what the issue is. It does come down to collecting the data in a way that PING-Mapper expects. I noticed that you had highly variable speeds while collecting the data (0-25.6 m/s). I think you mentioned that you did go on plane during the recording. The error is occurring while exporting speed-corrected imagery. This procedure uses the vessel speed to stretch the sonar tiles to true scale. Since you went on plane, the software is trying to stretch the tile across a very large distance, resulting in an out of memory error. For this particular recording, I recommend setting the `lbl_set` to 0 (`lbl_set=0`) in the `main.py` script. There will probably also be errors when you try to export georectified imagery as the range will increase dramatically when you go on plane as the sonar cannot find the bed at fast speeds and will increase the range automatically to search for the bed.

For future recordings, I have a few recommendations to help collect good quality sonar recordings:

- Start and stop recordings along intentional transects
- Maintain speeds of 4-6 mph
- Limit vessel pitch, yaw, and roll during data collection with smooth navigation
- If you are doing multiple overlapping transects, I recommend making a recording for each transect

For more information, I recommend taking a look at [this guide](https://www.researchgate.net/publication/259938373_An_Illustrated_Guide_to_Low-Cost_Sonar_Habitat_Mapping_v10) by Adam Kaeser and Thom Litts. Much of this information is outdated, but take a look at the sections about planning and acquiring good quality sonar data.

I plan on adding documentation to PING-Mapper's website on collecting high-quality sonar recordings, but that will be coming at a later date. I will close this issue for now, but feel free to ask any more questions here.

closed 2023-02-20 by CameronBodine

mjdikun, 2023-02-20:
Ah, makes sense.  I already had the Mosaics turned off..

I think it was still failing/running out of memory after turning lbl_set to 0.

But I'll try creating a recording at only low speed.

Thanks!

CameronBodine, 2023-02-20:
> I think it was still failing/running out of memory after turning lbl_set to 0.

If that is the case, go ahead and open a new issue and provide the console output.

The `lbl_set` parameter also accepts values not equal to 1. You could try a value of say 4 and see if that works. It will stretch the image along the transect track to be 4x larger.