CameronBodine/PINGMapper #77: PINGMapper for processing Humminbird Helix 7 data

henzb72, 2023-06-06:
Hi all,

I tried to run the PINGMapper to process my Humminbird Helix 7 data, however, I encountered this error:

Traceback (most recent call last):
  File "C:\Users\henzf\anaconda3\envs\PING\Lib\site-packages\osgeo\__init__.py", line 30, in swig_import_helper
    return importlib.import_module(mname)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\henzf\anaconda3\envs\PING\Lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 676, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 573, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1233, in create_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
ImportError: DLL load failed while importing _gdal: The specified procedure could not be found.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\henzf\MyPythonRepos\PINGMapper\[http://main.py](https://t.co/FBpEkKCDrP)", line 34, in <module>
    from main_readFiles import read_master_func
  File "C:\Users\henzf\MyPythonRepos\PINGMapper\src\main_readFiles.py", line 32, in <module>
    from class_portstarObj import portstarObj
  File "C:\Users\henzf\MyPythonRepos\PINGMapper\src\class_portstarObj.py", line 34, in <module>
    from osgeo import gdal
  File "C:\Users\henzf\anaconda3\envs\PING\Lib\site-packages\osgeo\__init__.py", line 35, in <module>
    _gdal = swig_import_helper()
            ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\henzf\anaconda3\envs\PING\Lib\site-packages\osgeo\__init__.py", line 32, in swig_import_helper
    return importlib.import_module('_gdal')
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\henzf\anaconda3\envs\PING\Lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ModuleNotFoundError: No module named '_gdal'

Could you please give me a solution regarding this.

Thank you.

CameronBodine, 2023-06-06:
Thanks for reporting! It appears there is an issue with the installation of the 'ping' conda environment. Can you confirm that the environment is activated and try to install gdal with the following:

conda install gdal

henzb72, 2023-06-07:
I have activated PING environment and when I checked  the conda list there is gdal                      3.7.0           py311hf43951c_1    conda-forge. I also had no issue when testing using test data, however the error came up when I run the main.py using my data

CameronBodine, 2023-07-25:
@henzb72 were you able to resolve your issue? I'm still willing to take a look at your recordings if you can share them with me.

henzb72, 2023-07-26:
Yes, I have uploaded the resulting Geotiff image to your google drive.
Please have a look and suggest what things I need to improve (regarding the acquisition and processing), since I think the image is still imperfect.

Cheers.

CameronBodine, 2023-07-26:
Thanks and I'm happy you were able to get up and running. I do have a few suggestions and comments. I want to first note that the recordings look pretty good and most of the issues in terms of quality are likely due to environmental factors.

- Survey speed is good. I typically aim for 6-8 km/h. This survey is about 5 km/h, and that is fine.
- I recommend breaking up your survey transects into separate recordings. In this case, a recording for downstream and a second for upstream. It's easier for PINGMapper to process and gives you more options for viewing and post processing.
- If possible, time your recordings when you expect the water to be free of anything that may cause interference, such as turbidity, waves, etc.
- There is a lot of noise visible in the water column. This could also be cause by locating the transducer next to the boat motor, or perhaps it isn't deep enough. Mount the transducer at or near the bow of the boat. Make sure it is 0.5-1 ft beneath the surface and that there are no further obstructions.
- From the bedpick plots, the auto depth doesn't perform as well as the sensor depth. Use the sensor depth to process wcr images.
- You may not want to remove shadows in this case. The low contrast of the image is causing the shadow model to mask too much of the imagery.

Hope this helps. I think the gain normalization feature coming (soon) in v2.0 of the software will help enhance the contrast and make the imagery more interpretable.

henzb72, 2023-07-31:
Thank you for the suggestions.
I have several inquiries regarding the image.
1.  There is different brightness in the image, particularly in the overlapping area (the brightness of the overlapped area doesn't match each other) so this lead to difficult interpretation if any object exists in this area
2.  There are a lot of blank spots in the image (what is it due? and how to handle this?)
3.  Can we compensate the distortion of the mosaic due to the vessel's movement up and down?

Cheers.

CameronBodine, 2023-07-31:
> There is different brightness in the image, particularly in the overlapping area (the brightness of the overlapped area doesn't match each other) so this lead to difficult interpretation if any object exists in this area
This is typical of side scan sonar imagery. The sound attenuates as the range increases, resulting in a reduction of signal returned to the transducer. No information is provided in the Humminbird file to correct for this. That is why I have developed a correction that is coming in the second version of PINGMapper called Empirical Gain Normalization. I included some examples below.
[image]
[image]

> There are a lot of blank spots in the image (what is it due? and how to handle this?)
This is the result of the shadow detection algorithm included in the software. You can disable this option by setting `remShadow=False`. You can find other information about the parameters [here](https://cameronbodine.github.io/PINGMapper/docs/gettingstarted/Exports.html).

> Can we compensate the distortion of the mosaic due to the vessel's movement up and down?
Are you referring to vessel heave? If so, none of this information is stored in the Humminbird file allowing for correction. If you had an intertial momentum unit, you could theoretically correct for this. But there are no options to apply these corrections in PINGMapper. This has been disscussed on #84. You can find out what information is stored in the Humminbird file [here](https://cameronbodine.github.io/PINGMapper/docs/advanced/HumFileStructure.html).

I hope this helps. 
P.S. I will respond to your inquiry on Twitter this week.

CameronBodine, 2023-09-18:
Hi Hendra,
An alpha release of v2.0.0 is now available, which includes the gain normalization routine. Go ahead and [update](https://cameronbodine.github.io/PINGMapper/docs/gettingstarted/UpdateInstallation.html) your version of PINGMapper and let me know how it goes.

henzb72, 2023-09-25:
> Hi Hendra, An alpha release of v2.0.0 is now available, which includes the gain normalization routine. Go ahead and [update](https://cameronbodine.github.io/PINGMapper/docs/gettingstarted/UpdateInstallation.html) your version of PINGMapper and let me know how it goes.

It's great to hear that, I would try to re-process the data.
Would you please reply to my inquiry on the Twitter?

Cheers.

CameronBodine, 2023-09-25:
So sorry! Just sent you a reply!

CameronBodine, 2024-01-22:
I am assuming this is fixed. Please reopen if the issue persists.

closed 2024-01-22 by CameronBodine