CameronBodine/PINGMapper #78: PingMapper not producing Mosaics

BGunnett8, 2023-06-12:
Hello,

This software has worked wonderfully so far but with the last batch of data it failed to give me the mosaic images for both water column present and absent. Everything else is there. In the miniconda window I do the following:

1.cd PingMapper
2. conda activate ping
3. python_batchDirectory.py

and it begins running my batch of files like normal but no mosaics. I've attached images of the files that it does produce and the code that I'm using. Usually under the files that you see in the screenshot is where it puts the mosaic images but not this time. Any help would be appreciated. Thank you!
[image]
[image]

CameronBodine, 2023-06-12:
Thanks for the detailed report! I assume you are trying to batch process multiple recordings? Does the mosaic process only fail on the last recording in the batch? If so, can you try running the `main.py` script on only that recording to see if the issue persists? Also, can you copy the output from the console?

BGunnett8, 2023-06-12:
I am trying to batch process several different recordings and it worked just fine when I did it a few weeks ago. This the output I get for the first recording and it continues to do this. It's processing everything normally but seems to skip over the step of creating the mosaics. It does this for every recording. 

+++++++++++++++++++++++++++++++++++++++++++
+++++++++++++++++++++++++++++++++++++++++++
***** Working On *****
./BenData2\R00021.DAT
Start Time:  2023-06-12 17:11

===========================================
===========================================
***** READING *****

Current CPU/RAM Usage:
________________________
CPU % | RAM % | RAM [GB]
________________________
0.0   | 38.5  | 6.4
________________________

Getting DAT Metadata...
Done!
Time (s): 0.1

Current CPU/RAM Usage:
________________________
CPU % | RAM % | RAM [GB]
________________________
0.0   | 38.5  | 6.4
________________________

Getting Header Structure...
Header Length: 72
ds_highfreq : Done!
ss_port : Done!
ss_star : Done!
ds_vhighfreq : Done!
Time (s): 0.0

Current CPU/RAM Usage:
________________________
CPU % | RAM % | RAM [GB]
________________________
3.5   | 38.5  | 6.4
________________________

Getting SON file header metadata...
[Parallel(n_jobs=4)]: Using backend LokyBackend with 4 concurrent workers.
[Parallel(n_jobs=4)]: Done   1 tasks      | elapsed:    6.2s
[Parallel(n_jobs=4)]: Done   2 out of   4 | elapsed:    6.2s remaining:    6.2s
[Parallel(n_jobs=4)]: Done   4 out of   4 | elapsed:    6.3s remaining:    0.0s
[Parallel(n_jobs=4)]: Done   4 out of   4 | elapsed:    6.3s finished

Current CPU/RAM Usage:
________________________
CPU % | RAM % | RAM [GB]
________________________
7.8   | 42.4  | 7.1
________________________

Locating missing pings and adding NoData...
[Parallel(n_jobs=8)]: Using backend LokyBackend with 8 concurrent workers.
[Parallel(n_jobs=8)]: Done   2 out of   8 | elapsed:   15.2s remaining:   45.8s
[Parallel(n_jobs=8)]: Done   3 out of   8 | elapsed:   15.3s remaining:   25.5s
[Parallel(n_jobs=8)]: Done   4 out of   8 | elapsed:   15.3s remaining:   15.3s
[Parallel(n_jobs=8)]: Done   5 out of   8 | elapsed:   15.3s remaining:    9.2s
[Parallel(n_jobs=8)]: Done   6 out of   8 | elapsed:   15.4s remaining:    5.1s
[Parallel(n_jobs=8)]: Done   8 out of   8 | elapsed:   15.7s remaining:    0.0s
[Parallel(n_jobs=8)]: Done   8 out of   8 | elapsed:   15.7s finished

Current CPU/RAM Usage:
[... 279 lines cut ...]
[Parallel(n_jobs=8)]: Done  10 out of  15 | elapsed:  2.0min remaining:  1.0min
[Parallel(n_jobs=8)]: Done  12 out of  15 | elapsed:  2.0min remaining:   30.1s
[Parallel(n_jobs=8)]: Done  15 out of  15 | elapsed:  2.0min finished

        Exporting 30 sonograms for ss_star
[Parallel(n_jobs=8)]: Using backend LokyBackend with 8 concurrent workers.
[Parallel(n_jobs=8)]: Done   2 out of  15 | elapsed:  1.0min remaining:  6.6min
[Parallel(n_jobs=8)]: Done   4 out of  15 | elapsed:  1.1min remaining:  3.0min
[Parallel(n_jobs=8)]: Done   6 out of  15 | elapsed:  1.1min remaining:  1.7min
[Parallel(n_jobs=8)]: Done   8 out of  15 | elapsed:  1.1min remaining:   59.4s
[Parallel(n_jobs=8)]: Done  10 out of  15 | elapsed:  1.9min remaining:   57.0s
[Parallel(n_jobs=8)]: Done  12 out of  15 | elapsed:  1.9min remaining:   28.5s
[Parallel(n_jobs=8)]: Done  15 out of  15 | elapsed:  2.0min finished

        Exporting 15 sonograms for ds_vhighfreq
[Parallel(n_jobs=8)]: Using backend LokyBackend with 8 concurrent workers.
[Parallel(n_jobs=8)]: Done   2 out of  15 | elapsed:   14.1s remaining:  1.5min
[Parallel(n_jobs=8)]: Done   4 out of  15 | elapsed:   14.2s remaining:   39.3s
[Parallel(n_jobs=8)]: Done   6 out of  15 | elapsed:   14.5s remaining:   21.9s
[Parallel(n_jobs=8)]: Done   8 out of  15 | elapsed:   14.6s remaining:   12.8s
[Parallel(n_jobs=8)]: Done  10 out of  15 | elapsed:   27.2s remaining:   13.6s
[Parallel(n_jobs=8)]: Done  12 out of  15 | elapsed:   27.6s remaining:    6.8s
[Parallel(n_jobs=8)]: Done  15 out of  15 | elapsed:   27.8s finished
Done!
Time (s): 296.5

CameronBodine, 2023-06-12:
Ok, it appears there is an issue during the rectification process. This is usually due to some issue with the recording itself but there may be an issue with the script also. Can you provide me with the problematic recording? You can upload it to my gdrive:

https://drive.google.com/open?id=1WdkzIydG67zWYf5rW7g1fWFQWnM8BsLx&usp=drive_fs

CameronBodine, 2023-06-14:
Hi @BGunnett8, you were experiencing an issue rectifying and mosaicing your imagery because the range was set to a very large value (looks like 180 meters?). An example of a bedpick plot is shown below.

[image]

 This makes the image prohibitively large to rectify and can't be fit into memory. I added a new option called `cropRange` to crop the range to a value, in meters, you can specify. 

https://github.com/CameronBodine/PINGMapper/blob/a019b5598068a680bd3d238be84c98bf55f14582/main_batchDirectory.py#L48

I tested on a value of `30` and was able to successfully export mosaics. Here's the same bedpick plot I showed above cropped to 30 m:

[image]

To update PING-Mapper, navigate to the PING-Mapper directory and run `git pull` in the console.

When you collect recordings in the future, I recommend setting the range to the maximum extent that you are able to see returns from the bed. When we scan rivers, I find a value of 30-50 meters per side is about the maximum usable extent.

BGunnett8, 2023-06-15:
Thank you very much for making this change? Was it an error that we made when recording with the Helix 7? There's a setting we should change? Also, when I ran git pull in the console, I got this which seems to be an error. 

error: Your local changes to the following files would be overwritten by merge:
        main.py
        main_batchDirectory.py
Please commit your changes or stash them before you merge.
Aborting
Updating 73aca8b..a019b55

BGunnett8, 2023-06-15:
That worked, I'll run all my files again with crop range and let you know how it goes. Thanks for the info on the helix as well.

BGunnett8, 2023-06-18:
Most of them worked this time but a few still didn't produce mosaics. I posted recordings 22, 24, 29, 40, 41 and 42 in your google drive. These still had the same error of showing all of the data separately but failing to produce mosaics. The whole batch was run with CropRange = 30.

CameronBodine, 2023-06-18:
Can you run only one of the problematic recordings with main.py and post the error? Thanks!

CameronBodine, 2023-06-18:
I was just able to process mosaics from the new set of recordings you provided. I believe you are experiencing an OOM (out of memor) error. I noticed from your original console output that your computer has 16 GB of RAM. My computer has more and is thus able to export the mosaics. There are two parameters you can adjust on your computer to successfully export these mosaics: 

1. The first is the the `nchunk` parameter. Specifying a smaller `nchunk` value will reduce the number of pings that are loaded into memory at once. You could try a value of `150` or `250`. 

2. The second parameter is `threadCnt`. PING-Mapper uses multithreading to speed up the export process. A value of `0` will use all threads available on your computer, meaning loading more data into memory at once. Using a negative value will reduce the number of threads to use by that much, or you can fix it at a smaller value, say 1 to 4, to specify the exact number of threads.

If you run into further issues, please try processing the problematic recording with `main.py` instead of the batch processing script. There will be an error printed which you can provide to help me identify the issue. The batch script is designed to process what it can, even if there is an error, and will attempt to process all the recordings in the parent directory.

Let me know how it goes for you!

CameronBodine, 2023-06-26:
Thank you for the console output! It's an out of memory issue, as I suspected. Did you try adjusting the `nchunk` or `threadCnt` parameters? If you did, I recommend trying smaller values to see if you can get the rectified images to export.

CameronBodine, 2023-06-26:
FYI, the last portion of your console output indicates that there is not enough memory available to allocate the array:

```
File "C:\Users\BGunn\miniconda3\envs\ping\Lib\concurrent\futures_base.py", line 401, in __get_result
raise self._exception
numpy.core._exceptions._ArrayMemoryError: Unable to allocate 1.20 GiB for an array with shape (80553616, 2) and data type float64
```

BGunnett8, 2023-07-06:
I made the alterations and I was able to get images for everything! Thank you so much for all of your help and responsiveness!

CameronBodine, 2023-07-06:
Glad to hear it! Happy scanning and mapping!

closed 2023-07-06 by CameronBodine

CameronBodine, 2023-07-07:
I found the memory leak! You _should_ be able to process the problematic recordings without modifying `nchunk` or `threadCnt` after updating with `git pull`. It is possible you may still get OOM error, but this fix should lower the chances of that.

commit 1b9500a8 (2023-07-07): Massive mem leak #38 should fix #78 too. Don't return self!
commit 5e74acf0 (2023-07-07): Merge pull request #86 from CameronBodine/test-1

Massive mem leak #38 should fix #78 too. Don't return self!