CameronBodine/PINGMapper #83: Settings for vegetation

dfcscience, 2023-06-30:
Would greatly appreciate guidance on settings for mapping shallow (~2 m) marine vegetation (mainly seagrass), both with the Humminbird  and in PINGMapper.

CameronBodine, 2023-06-30:
I don't have a lot of experience mapping vegetation but I can share what insight I do have. I have found both the 455 khz and 1.2 mhz frequencies to be suitable for mapping veg. The functional range of the imagery is decreased with the higher frequency but that may not be a big factor at your relatively shallow depths. The higher frequency may or may not help in identifying the veg, so that is something to tinker with. One thing to consider is the height of the veg. If the veg is dense and tall, it will block much of the signal, limiting the swath of the side scan. I did some mapping of eel grass on the St Johns river in Florida and the grass was thick and grew to the surface, so I was only able to map the outer extent of the veg.

If you have other questions, don't hesitate to ask. I'd be interested to hear how it goes and what you learned from the process.

dfcscience, 2023-07-29:
Thanks for the suggestions. 

I didn't formulate my question properly. The Helix 7 has A LOT of options for side scan imaging. When I change a setting on the display, does it change the data that's recorded on the SD card? 

For example, when I change the swath width on the display, I still seem to get the full swath width when I process the imagery in PINGmapper.

CameronBodine, 2023-07-31:
Thanks for the clarification. I'm not completely certain what settings made on the Humminbird will be reflected on the sonar recording. One setting that is reflected on the recording is the range (swath) set during data collection. At 455 khz, the theoretical swath width/depth is ~800 ft which I am certain is not the default of the data stored on the file. There are also settings to adjust gain/brightness which _I think_ only affect the imagery displayed on the screen, but don't know for sure.

If you want to do some tests, noting the settings you make then adjusting mid-recording and noting the time, I'd be interested to investigate further with you.

CameronBodine, 2024-01-22:
I am going to convert this to a discussion to clear out open issues.