CameronBodine/PINGMapper #84: Integration with Ardupilot compatible autopilots

dfcscience, 2023-06-30:
I am using a Humminbird Helix 7 G4 MEGA SI on an autonomous boogie board boat, which uses a Pixhawk for waypoint navigation. I can send the position data from the Pixhawk's GPS to the Humminbird but I would like to use the pitch, roll, and yaw data in the Pixhawk log file to improve georectification

CameronBodine, 2023-06-30:
Thanks for your inquiry! I really dig your autonomous boogie board boat!

Correcting georectification using pitch, roll, and yaw data is a great idea. Unfortunately, that functionality is not available in PINGMapper and I don't have a lot of time to work on it at the moment. I have been giving it a think and can imagine ways to correct for each of those factors individually, but am in the dark as to how to resolve them to a final correction. If you had ideas on how the math and workflow might work, that would certainly help. I'll keep this issue open to document any ideas that come up.

CameronBodine, 2024-01-22:
I am going to convert this to a discussion to clear out open issues. 

I may be able to start to work on this in the coming months. Do you have an example recording and the Pixhawk log file that I might be able to test any workflows that I develop?