CameronBodine/PINGMapper #85: Got an exception when Running "test_PINGMapper.py 1“

gushangzao007, 2023-07-07:
I try to run the test function to verify my installation, it got several errors. 
error message like that:   AttributeError: module 'numpy' has no attribute 'int'.
[error.txt](https://github.com/CameronBodine/PINGMapper/files/11980821/error.txt)

gushangzao007, 2023-07-07:
by the way , expected data has been generated. I'm not sure if those data is ok.
[image]

CameronBodine, 2023-07-07:
Can you provide the full error from the console? Thanks!

gushangzao007, 2023-07-07:
full error is in the attached file within my first post, can you see it?

CameronBodine, 2023-07-07:
Sorry I missed it. Ok, just posted an update. Run `git pull` and try the test again. Please report back!

CameronBodine, 2023-07-25:
@gushangzao007 was your issue resolved?

gushangzao007, 2023-07-30:
sorry, have not yet, I've changed my solution for my hobby project, thanks for your advice about how to use SL3Reader to generate a point cloud, :)

CameronBodine, 2023-07-31:
Great I will close the issue then.

closed 2023-07-31 by CameronBodine