CameronBodine/PINGMapper #92: Problem with Tensorflow Module upon Attempted Update

AdrianPinchbeck, 2023-09-03:
I attempted to update to new version, but seemed to have an issue with updating the environment.  I tried a new install in a different directory using the libmamba solver, but also had issues, and installed using classic instead.  Unfortunately I now seem to be having a problem with the module tensorflow.keras.  I've been down a number of rabbit holes trying to correct the issue without success.  Just wondered if you have any pointers?
Adrian
[image]

CameronBodine, 2023-09-03:
Thanks for reporting Adriane. Do you remember what the issues were with libmamba?

It looks like your environment is either missing Keras or there is a mismatch in versions between tensorflow and keras. https://stackoverflow.com/questions/73796400/error-no-module-named-tensorflow-keras

Can you run 'conda list' in the ping environment and check that keras and tensorflow versions match? And report the output please.

AdrianPinchbeck, 2023-09-03:
Sorry, I don't recall the actual issue, I did attempt to record some of the screen errors, but the capture didn't work!  Conda List reports thus:
[image]

and Keras:
[image]

CameronBodine, 2023-09-03:
Ok give this a try:
```
conda install keras=2.13.0
```

If that doesn't work, try removing the ping environment with
```
conda remove --name ping --all
```

then clean miniconda
```
conda clean --all
```

then update miniconda
```
conda update -n base conda
```

clean again
```
conda clean --all
```

update pip
```
pip install --upgrade pip
```

get libmamba up and running again
```
conda install -n base conda-libmamba-solver
conda config --set solver libmamba
```

then try reinstalling ping
```
conda env create --file conda/PINGMapper.yml
```

If you can keep track of errors as they arise and report them, that will help a lot! Thanks for helping me debug!!!!

CameronBodine, 2023-09-04:
Hmm, give this a try:

conda install pip

You should have write access to that folder. Strange indeed!

AdrianPinchbeck, 2023-09-04:
Just realised that I forgot the second clean --all after updating conda.  I could clean again before your above conda install pip?

AdrianPinchbeck, 2023-09-04:
Checked the Attributes for the Site Packages folder and found that it is marked up as 'Read Only', together with all it's sub-directories.  I'm going to uncheck the Attributes bullet, and try pip install --upgrade pip again...  Wish me luck!

AdrianPinchbeck, 2023-09-04:
Having no luck so far!  Changing attributes via Windows, making sure all permissions etc. are available to the top of the tree, and having default administrator rights anyway, I'm running in to failure so far.  Attributes are checked in grey for Anaconda3 directory and sub-directories (not black), further up the tree in black. Unfortunately I don't have much time this week to progress through the issue during the day, but I'll keep you informed of progress tomorrow evening again.  Almost time for bed now, but thanks again for your help to date.

CameronBodine, 2023-09-04:
One thing could try is uninstalling miniconda and reinstalling, making sure to select Install for: Just Me (recommended). Also, maybe you are using Anaconda instead of miniconda? I have had a lot more success with Miniconda. Keep me posted and I will help as much as possible!

[image]

AdrianPinchbeck, 2023-09-05:
I've been using the full Anaconda3 installation, so potentially run into some conflict/s somewhere.  Have uninstalled it, and currently installing Miniconda3 as above...

CameronBodine, 2023-09-05:
My bad. Try running `git fetch`. Also, there is a simple GUI you could try out, if interested. Run either `python gui_main.py` or `python gui_main_batchDirectory.py`.

AdrianPinchbeck, 2023-09-05:
Will take a look at the GUI once have it working if that's okay!! Similar error results after running git fetch (didn't try remove cmocean #92):
[image]

AdrianPinchbeck, 2023-09-05:
PINGMapper test folder:
[image]

CameronBodine, 2023-09-05:
Ok that helps. Looks like it isn't updating. Go ahead and delete the folder, then download the latest version with `git clone --depth 1 https://github.com/CameronBodine/PINGMapper`. You may need to update the environment again with this `conda env update --file conda/PINGMapper.yml --prune`.

AdrianPinchbeck, 2023-09-05:
I've placed hash on line 39 within class_rectObj.py to form a comment, then saved, but still getting same error when running the test:
 
[image]

AdrianPinchbeck, 2023-09-05:
Ah okay, I may place it in a new folder altogether.

CameronBodine, 2023-09-05:
I was also wondering if you may be editing one version of PINGMapper then running a version in a different folder.

AdrianPinchbeck, 2023-09-05:
I did wonder that myself, but checking the navigation all seemed to correspond correctly.  Might be wise to relocate any old directories within a separate archive directory to change the path for better housekeeping?

AdrianPinchbeck, 2023-09-05:
The test is running.  I did notice a message during the mosaic section that I'll check over.
We're back in business though - I may need a little lie down!

[image]

CameronBodine, 2023-09-05:
Good to hear! I'm sorry it was such a pain! Yes, I need to get that warning taken care of. It won't affect anything for you though.

AdrianPinchbeck, 2023-09-05:
Thanks for all your input Cameron, it must have been a pain for you too!  Yes, both mosaics seem to have been created successfully.  I'll probably go through it all in more detail tomorrow, so I'll go now and have a stiff drink before bed - rum.  Thanks again, and I'm planning to place a good result post on here in connection with that ghost gear I mapped the other week....

CameronBodine, 2023-09-05:
It's always an adventure getting folks up and running! Looking forward to your post! Cheers!

I'll close this issue now. Please make another if needed.

closed 2023-09-05 by CameronBodine