CameronBodine/PINGMapper #100: No substrate data produced

BGunnett8, 2023-10-23:
Hello,

We have messaged before about this same issue however back then it was with PingMapper not producing mosaics. You had me alter the nchunk to 250 and croprange to 30 and the issue was solved. Now that PingMapper 2.0 is out, I decided to run the same files to get substrate classification. The same exact set of files that were not producing mosaics are now not returning any substrate data, it doesn't even make a folder labelled substrate like every other file. I did alter the nchunk to 250 in the substrate section of code. I'm using main_batchDirectory.py for these files. Thanks for your help in the past and hopefully you know the solution to this problem as well. This is what the code looks like for me: 

# Part of PING-Mapper software
#
# Co-Developed by Cameron S. Bodine and Dr. Daniel Buscombe
#
# Inspired by PyHum: https://github.com/dbuscombe-usgs/PyHum
#
# MIT License
#
# Copyright (c) 2022-23 Cameron S. Bodine
#
# Permission is hereby granted, free of charge, to any person obtaining a copy
# of this software and associated documentation files (the "Software"), to deal
# in the Software without restriction, including without limitation the rights
# to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
# copies of the Software, and to permit persons to whom the Software is
# furnished to do so, subject to the following conditions:
#
# The above copyright notice and this permission notice shall be included in all
# copies or substantial portions of the Software.
#
# THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
# IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
# FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
# AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
# LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
# OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
# SOFTWARE.

import sys
sys.path.insert(0, 'src')

from funcs_common import *
from main_readFiles import read_master_func
from main_rectify import rectify_master_func
from main_mapSubstrate import map_master_func

import time
import datetime

# Get processing script's dir so we can save it to file
scriptDir = os.getcwd()

# For the logfile
oldOutput = sys.stdout

#============================================

#######################
# Start User Parameters
#######################

# Path to data/output
inDir = r'./BenData4' # Parent folder of sonar recordings
outDir = r'./BenDataOut4' # Parent folder for export files

# *** IMPORTANT ****
# Export Mode: project_mode
## 0==NEW PROJECT: Create a new project. [DEFAULT]
##      If project already exists, program will exit without any project changes.
##
## 1==OVERWRITE MODE: Create new project, regardless of previous project state.
##      If project exists, it will be DELETED and reprocessed.
##      If project does not exist, a new project will be created.
project_mode = 1

# General Parameters
[... 191 lines cut ...]
            print('\n===========================================')
            print('===========================================')
            print('***** RECTIFYING *****')
            rectify_master_func(**params)
            # rectify_master_func(sonFiles, humFile, projDir, nchunk, rect_wcp, rect_wcr, mosaic, threadCnt)

        #==================================================
        #==================================================
        if pred_sub or map_sub or export_poly or pltSubClass:
            print('\n===========================================')
            print('===========================================')
            print('***** MAPPING SUBSTRATE *****')
            print("working on "+projDir)
            map_master_func(**params)

        sys.stdout.log.close()

    except Exception as Argument:
        unableToProcessError(logfilename)
        print('\n\nCould not process:', datFile)

    sys.stdout = oldOutput

    gc.collect()
    print("\n\nTotal Processing Time: ",datetime.timedelta(seconds = round(time.time() - start_time, ndigits=0)))

CameronBodine, 2023-10-23:
Sorry for the issue. It should be re-enabled now. Run `git stash` then `git pull` and try running the script again. Let me know how it goes!

BGunnett8, 2023-10-26:
Thanks for the help! It worked and I have everything!

closed 2023-10-26 by BGunnett8