CameronBodine/PINGMapper #101: Install issues

mjhawker, 2023-10-31:
Hi Cameron,

I've had some installation issues and finally surrendered. I can't get past the final installation step of creating the environment. Here's what I see each time I've tried. 

Pip subprocess error:
ERROR: Could not find a version that satisfies the requirement tensorflow (from versions: none)
ERROR: No matching distribution found for tensorflow 

failed

CondaEnvException: Pip failed

I've tried updating pip, updating tensorflow, uninstalling, reinstalling, etc. I also tried reverting to an earlier Python version 3.6.  Was there a version update that broke compatibility or am I simply doing something wrong?

CameronBodine, 2023-10-31:
I'm sorry to hear this @mjhawker. I have not seen this issue yet. Can you try editing the yml file in the PINGMapper conda folder and specify Tensorflow==12.0 instead of Tensorflow?

CameronBodine, 2023-10-31:
I did see [this](https://stackoverflow.com/a/41084963). Perhaps you installed the wrong version of Miniconda? Can you confirm that you installed the 64-bit version rather then the 32-bit version? In the Anaconda prompt, first run `python` then type `import struct;print(struct.calcsize("P") * 8)` and confirm that the output is `64`, not `32`. For reference, this is what my environment looks like after updating in the past month or so:

<details>
<summary>Click to expand</summary>

```
(base) PS C:\> conda activate ping
(ping) PS C:\> conda list
# packages in environment at C:\Users\csb67\AppData\Local\miniconda3\envs\ping:
#
# Name                    Version                   Build  Channel
absl-py                   2.0.0                    pypi_0    pypi
affine                    2.4.0              pyhd8ed1ab_0    conda-forge
aom                       3.6.1                h63175ca_0    conda-forge
astunparse                1.6.3                    pypi_0    pypi
attrs                     23.1.0             pyh71513ae_1    conda-forge
blosc                     1.21.5               hdccc3a2_0    conda-forge
branca                    0.6.0              pyhd8ed1ab_0    conda-forge
brotli                    1.1.0                hcfcfb64_1    conda-forge
brotli-bin                1.1.0                hcfcfb64_1    conda-forge
brotli-python             1.1.0           py311h12c1d0e_1    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
c-ares                    1.19.1               hcfcfb64_0    conda-forge
c-blosc2                  2.10.4               h183a6f4_0    conda-forge
ca-certificates           2023.7.22            h56e8100_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cachetools                5.3.1                    pypi_0    pypi
cairo                     1.16.0            h412253b_1017    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
cfitsio                   4.3.0                h9b0cee5_0    conda-forge
charls                    2.4.2                h1537add_0    conda-forge
charset-normalizer        3.3.0              pyhd8ed1ab_0    conda-forge
click                     8.1.7           win_pyh7428d3b_0    conda-forge
click-plugins             1.1.1                      py_0    conda-forge
cligj                     0.7.2              pyhd8ed1ab_1    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
contourpy                 1.1.1           py311h005e61a_1    conda-forge
cycler                    0.12.0             pyhd8ed1ab_0    conda-forge
dav1d                     1.2.1                hcfcfb64_0    conda-forge
doodleverse-utils         0.0.35                   pypi_0    pypi
expat                     2.5.0                h63175ca_1    conda-forge
filelock                  3.12.4                   pypi_0    pypi
fiona                     1.9.4           py311h4e4dc46_0    conda-forge
flatbuffers               23.5.26                  pypi_0    pypi
folium                    0.14.0             pyhd8ed1ab_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               hbde0cde_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.43.0          py311ha68e1ae_0    conda-forge
freetype                  2.12.1               hdaf720e_2    conda-forge
freexl                    2.0.0                h8276f4a_0    conda-forge
fsspec                    2023.9.2                 pypi_0    pypi
gast                      0.5.4                    pypi_0    pypi
gdal                      3.7.2           py311h9601e46_4    conda-forge
geopandas                 0.14.0             pyhd8ed1ab_0    conda-forge
geopandas-base            0.14.0             pyha770c72_0    conda-forge
geos                      3.12.0               h1537add_0    conda-forge
geotiff                   1.7.1               h2e186c9_13    conda-forge
gettext                   0.21.1               h5728263_0    conda-forge
giflib                    5.2.1                h64bf75a_3    conda-forge
glib                      2.78.0               h12be248_0    conda-forge
glib-tools                2.78.0               h12be248_0    conda-forge
google-auth               2.23.2                   pypi_0    pypi
google-auth-oauthlib      1.0.0                    pypi_0    pypi
google-pasta              0.2.0                    pypi_0    pypi
[... 160 lines cut ...]
typing-extensions         4.8.0                    pypi_0    pypi
tzdata                    2023c                h71feb2d_0    conda-forge
ucrt                      10.0.22621.0         h57928b3_0    conda-forge
uriparser                 0.9.7                h1537add_1    conda-forge
urllib3                   2.0.6              pyhd8ed1ab_0    conda-forge
vc                        14.3                h64f974e_17    conda-forge
vc14_runtime              14.36.32532         hdcecf7f_17    conda-forge
[... cut ...]
xyzservices               2023.10.0          pyhd8ed1ab_0    conda-forge
xz                        5.2.6                h8d14728_0    conda-forge
zfp                       1.0.0                h63175ca_4    conda-forge
zipp                      3.17.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               hcfcfb64_5    conda-forge
zlib-ng                   2.0.7                hcfcfb64_0    conda-forge
zstd                      1.5.5                h12be248_0    conda-forge
```
</details>

mjhawker, 2023-10-31:
Thanks for the quick response. I confirmed Miniconda 64 bit version using your command. The output for my environment matches yours except none of the pipy channels, just conda-forge. 

I followed your directions to the T (several times).  I'm somewhat novice to Python, but learning. My bathymetry tool box is much deeper!

CameronBodine, 2023-11-01:
> I did see [this](https://stackoverflow.com/a/41084963). Perhaps you installed the wrong version of Miniconda? Can you confirm that you installed the 64-bit version rather then the 32-bit version?

I figured that was a stab in the dark!

Ok, yes, that information helps. Seems Python has a new version I didn't know about. Tensorflow currently supports Python 3.8-3.11. I updated the .yml. Go ahead and update by running the following, then try the install again.

`git stash` then `git pull`. 

Let me know how it goes!

mjhawker, 2023-11-01:
Ok, that seemed to work to complete the install. I also updated everything else with your update page directions.  However, I tried to run your test data set and got this: 

```
(ping) PS C:\Users\mjhaw\PythonPackages\PINGMapper> python test_PINGMapper.py 1
Traceback (most recent call last):
  File "C:\Users\mjhaw\PythonPackages\PINGMapper\test_PINGMapper.py", line 34, in <module>
    from funcs_common import *
  File "C:\Users\mjhaw\PythonPackages\PINGMapper\src\funcs_common.py", line 43, in <module>
    import rasterio
  File "C:\Users\mjhaw\miniconda3\envs\ping\Lib\site-packages\rasterio\__init__.py", line 28, in <module>
    from rasterio._version import gdal_version, get_geos_version, get_proj_version
ImportError: DLL load failed while importing _version: The specified procedure could not be found.
(ping) PS C:\Users\mjhaw\PythonPackages\PINGMapper>
```

CameronBodine, 2023-11-01:
I suspect you are in a limbo state after trying to do an install and everything failing. Sorry for the issue but this tends to be the situation with this software. Though it is tedious, I recommend starting completely from scratch as there has been unanticipated issues:  

- Delete the PINGMapper folder
- Uninstall miniconda
- Delete any conda or miniconda folders in your user directory 
- Start the installation process over again following these directions: https://cameronbodine.github.io/PINGMapper/docs/gettingstarted/InstallationDetailed.html

Any errors you may encounter, please post the complete output from the console. 

Again, I apologize for the inconvenience. It's very difficult to anticipate what is causing the issue remotely.

mjhawker, 2023-11-01:
Ok, uninstalled everything and made sure any folders were deleted. Followed the installation directions and it seemed successful, but got the same output as above. I'll try on a fresh pc in the morning.

```
(base) PS C:\Users\mjhaw\PythonPackages\PINGMapper> conda activate ping
(ping) PS C:\Users\mjhaw\PythonPackages\PINGMapper> python test_PINGMapper.py 1
Traceback (most recent call last):
  File "C:\Users\mjhaw\PythonPackages\PINGMapper\test_PINGMapper.py", line 34, in <module>
    from funcs_common import *
  File "C:\Users\mjhaw\PythonPackages\PINGMapper\src\funcs_common.py", line 43, in <module>
    import rasterio
  File "C:\Users\mjhaw\miniconda3\envs\ping\Lib\site-packages\rasterio\__init__.py", line 28, in <module>
    from rasterio._version import gdal_version, get_geos_version, get_proj_version
ImportError: DLL load failed while importing _version: The specified procedure could not be found.
```

CameronBodine, 2023-11-01:
Thank you for trying and I appreciate your patience! I will dig into this further today. Seems to be some other package version issue between rasterio and gdal.

CameronBodine, 2023-11-01:
Unfortunately, I am not able to replicate your issue on my end. For documentation purposes, here are the steps that I took:

1. I did not remove and reinstall miniconda. I recently did a reinstall.
2. I did the update and maintenance as described in Step 1 to Step 5 from the [update instructions](https://cameronbodine.github.io/PINGMapper/docs/gettingstarted/UpdateInstallation.html)
3. Just to be sure, I renamed the environment from `ping` to `ping2` and installed the environment as described in[ Step 9](https://cameronbodine.github.io/PINGMapper/docs/gettingstarted/Installation.html#step-9) of the install instructions.
4. Ran the test script with no errors. I attached the processing logs found in `PINGMapper/procData/PINGMapper-Test-Small-DS/meta/logs/` 
[log_2023-11-01_0504.txt](https://github.com/CameronBodine/PINGMapper/files/13226783/log_2023-11-01_0504.txt)

@mjhawker I will look to see if I can find info about your error.

CameronBodine, 2023-11-01:
I tried a small fix referenced [here](https://stackoverflow.com/a/61460363). @mjhawker can you do a `git pull` to update PINGMapper then try the test again? If it fails, can you provide the following:

1. Attach any logs from `PINGMapper/procData/PINGMapper-Test-Small-DS/meta/logs/`
2. The complete console output from running `conda list` in the activated `ping` environment.

mjhawker, 2023-11-01:
Well, the errors changed a bit after the update, but still no output. Here's the console message and conda list. 

There's no file other than the "OutputsHere" text file in that location.

(ping) PS C:\Users\mjhaw\PythonPackages\PINGMapper> python test_PINGMapper.py 1
Version:  2.14.0
Eager mode:  True
Traceback (most recent call last):
  File "C:\Users\mjhaw\PythonPackages\PINGMapper\test_PINGMapper.py", line 35, in <module>
    from main_readFiles import read_master_func
  File "C:\Users\mjhaw\PythonPackages\PINGMapper\src\main_readFiles.py", line 32, in <module>
    from class_portstarObj import portstarObj
  File "C:\Users\mjhaw\PythonPackages\PINGMapper\src\class_portstarObj.py", line 31, in <module>
    from funcs_model import *
  File "C:\Users\mjhaw\PythonPackages\PINGMapper\src\funcs_model.py", line 51, in <module>
    from doodleverse_utils.prediction_imports import *
  File "C:\Users\mjhaw\miniconda3\envs\ping\Lib\site-packages\doodleverse_utils\prediction_imports.py", line 32, in <module>
    from tkinter import filedialog
  File "C:\Users\mjhaw\miniconda3\envs\ping\Lib\tkinter\__init__.py", line 38, in <module>
    import _tkinter # If this fails your Python may not be configured for Tk
    ^^^^^^^^^^^^^^^
ImportError: DLL load failed while importing _tkinter: The specified procedure could not be found.

```
(ping) PS C:\Users\mjhaw\PythonPackages\PINGMapper> conda list
# packages in environment at C:\Users\mjhaw\miniconda3\envs\ping:
#
# Name                    Version                   Build  Channel
absl-py                   2.0.0                    pypi_0    pypi
affine                    2.4.0              pyhd8ed1ab_0    conda-forge
aom                       3.6.1                h63175ca_0    conda-forge
astunparse                1.6.3                    pypi_0    pypi
attrs                     23.1.0             pyh71513ae_1    conda-forge
blosc                     1.21.5               hdccc3a2_0    conda-forge
branca                    0.6.0              pyhd8ed1ab_0    conda-forge
brotli                    1.1.0                hcfcfb64_1    conda-forge
brotli-bin                1.1.0                hcfcfb64_1    conda-forge
brotli-python             1.1.0           py311h12c1d0e_1    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
c-ares                    1.20.1               hcfcfb64_1    conda-forge
c-blosc2                  2.10.5               h183a6f4_0    conda-forge
ca-certificates           2023.7.22            h56e8100_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cachetools                5.3.2                    pypi_0    pypi
cairo                     1.18.0               h1fef639_0    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
cfitsio                   4.3.0                h9b0cee5_0    conda-forge
charls                    2.4.2                h1537add_0    conda-forge
charset-normalizer        3.3.1              pyhd8ed1ab_0    conda-forge
click                     8.1.7           win_pyh7428d3b_0    conda-forge
click-plugins             1.1.1                      py_0    conda-forge
cligj                     0.7.2              pyhd8ed1ab_1    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
contourpy                 1.1.1           py311h005e61a_1    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
dav1d                     1.2.1                hcfcfb64_0    conda-forge
doodleverse-utils         0.0.35                   pypi_0    pypi
expat                     2.5.0                h63175ca_1    conda-forge
filelock                  3.13.1                   pypi_0    pypi
fiona                     1.9.5           py311h4e4dc46_0    conda-forge
flatbuffers               23.5.26                  pypi_0    pypi
folium                    0.14.0             pyhd8ed1ab_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               hbde0cde_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
[... 177 lines cut ...]
transformers              4.34.1                   pypi_0    pypi
typing-extensions         4.8.0                    pypi_0    pypi
tzdata                    2023c                h71feb2d_0    conda-forge
ucrt                      10.0.22621.0         h57928b3_0    conda-forge
uriparser                 0.9.7                h1537add_1    conda-forge
urllib3                   2.0.7              pyhd8ed1ab_0    conda-forge
vc                        14.3                h64f974e_17    conda-forge
vc14_runtime              14.36.32532         hdcecf7f_17    conda-forge
versioneer                0.29                     pypi_0    pypi
vs2015_runtime            14.36.32532         h05e6639_17    conda-forge
werkzeug                  3.0.1                    pypi_0    pypi
wheel                     0.41.3             pyhd8ed1ab_0    conda-forge
[... cut ...]
xyzservices               2023.10.1          pyhd8ed1ab_0    conda-forge
xz                        5.2.6                h8d14728_0    conda-forge
zfp                       1.0.0                h63175ca_4    conda-forge
zipp                      3.17.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               hcfcfb64_5    conda-forge
zlib-ng                   2.0.7                hcfcfb64_0    conda-forge
zstd                      1.5.5                h12be248_0    conda-forge
```

CameronBodine, 2023-11-01:
Ok, we are making progress at least! Do you happen to have ArcGIS installed on your computer? Do you happen to have a Python27 folder at `C:\Python27`? It's possible your computer environments may be mixing up python installations, but I'm not sure. Also, are you running the test script from the Anaconda Powershell Prompt?

Regardless of answers to above, in the activated `ping` environment, can you try reinstalling tkinter with `conda install tk` then trying to run the test script again?

mjhawker, 2023-11-01:
Yes!  I run ArcGIS Pro all the time. I don't have a python27 file under C: or anywhere else I looked. 

I installed tk and reran the test script.  No joy. 

Just for clarification, after I do an package install or run a git command, do I need to refresh the env?

```
(ping) PS C:\Users\mjhaw\PythonPackages\PINGMapper> python test_PINGMapper.py 1
Version:  2.14.0
Eager mode:  True
Traceback (most recent call last):
  File "C:\Users\mjhaw\PythonPackages\PINGMapper\test_PINGMapper.py", line 35, in <module>
    from main_readFiles import read_master_func
  File "C:\Users\mjhaw\PythonPackages\PINGMapper\src\main_readFiles.py", line 32, in <module>
    from class_portstarObj import portstarObj
  File "C:\Users\mjhaw\PythonPackages\PINGMapper\src\class_portstarObj.py", line 31, in <module>
    from funcs_model import *
  File "C:\Users\mjhaw\PythonPackages\PINGMapper\src\funcs_model.py", line 51, in <module>
    from doodleverse_utils.prediction_imports import *
  File "C:\Users\mjhaw\miniconda3\envs\ping\Lib\site-packages\doodleverse_utils\prediction_imports.py", line 32, in <module>
    from tkinter import filedialog
  File "C:\Users\mjhaw\miniconda3\envs\ping\Lib\tkinter\__init__.py", line 38, in <module>
    import _tkinter # If this fails your Python may not be configured for Tk
    ^^^^^^^^^^^^^^^
ImportError: DLL load failed while importing _tkinter: The specified procedure could not be found.
```

CameronBodine, 2023-11-01:
Good to know. Do you happen to have a `C:\Python3*` folder? There shouldn't be an issue with Pro, but I suppose it's possible.

As long as you issue the `conda install *` and press enter (or hit `y` then enter) to accept the install, it should all work. I'll keep digging...

CameronBodine, 2023-11-01:
Just for kicks, can you try installing tk from the anaconda channel with `conda install -c anaconda tk`?

mjhawker, 2023-11-01:
Installed from the anaconda channel, no change was required. Same error. I don't see any Python folders except for C:\Program Files\ArcGIS\Pro\bin\Python

I also have QGIS installed.

CameronBodine, 2023-11-01:
Can you share the console output from `conda info -vv`?

CameronBodine, 2023-11-01:
Here is something else to try:

```
python.exe -m pip install tk-tools
```

mjhawker, 2023-11-01:
```
(ping) PS C:\Users\mjhaw\PythonPackages\PINGMapper> conda info -vv

     active environment : ping
    active env location : C:\Users\mjhaw\miniconda3\envs\ping
            shell level : 2
       user config file : C:\Users\mjhaw\.condarc
 populated config files : C:\Users\mjhaw\.condarc
          conda version : 23.9.0
    conda-build version : not installed
         python version : 3.11.5.final.0
       virtual packages : __archspec=1=x86_64
                          __win=0=0
       base environment : C:\Users\mjhaw\miniconda3  (writable)
      conda av data dir : C:\Users\mjhaw\miniconda3\etc\conda
  conda av metadata url : None
           channel URLs : https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
          package cache : C:\Users\mjhaw\miniconda3\pkgs
                          C:\Users\mjhaw\.conda\pkgs
                          C:\Users\mjhaw\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\mjhaw\miniconda3\envs
                          C:\Users\mjhaw\.conda\envs
                          C:\Users\mjhaw\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/23.9.0 requests/2.31.0 CPython/3.11.5 Windows/10 Windows/10.0.22621 solver/libmamba conda-libmamba-solver/23.9.3 libmambapy/1.5.1
          administrator : False
             netrc file : None
           offline mode : False

# conda environments:
#
base                     C:\Users\mjhaw\miniconda3
ping                  *  C:\Users\mjhaw\miniconda3\envs\ping

sys.version: 3.11.5 | packaged by Anaconda, Inc. | (m...
sys.prefix: C:\Users\mjhaw\miniconda3
sys.executable: C:\Users\mjhaw\miniconda3\python.exe
conda location: C:\Users\mjhaw\miniconda3\Lib\site-packages\conda
conda-build: None
conda-content-trust: C:\Users\mjhaw\miniconda3\Scripts\conda-content-trust.exe
conda-env: C:\Users\mjhaw\miniconda3\Scripts\conda-env.exe
user site dirs:

CIO_TEST: <not set>
CONDA_DEFAULT_ENV: ping
CONDA_EXE: C:\Users\mjhaw\miniconda3\Scripts\conda.exe
CONDA_PREFIX: C:\Users\mjhaw\miniconda3\envs\ping
CONDA_PREFIX_1: C:\Users\mjhaw\miniconda3
CONDA_PROMPT_MODIFIER: (ping)
CONDA_PYTHON_EXE: C:\Users\mjhaw\miniconda3\python.exe
CONDA_ROOT: C:\Users\mjhaw\miniconda3
CONDA_SHLVL: 2
CURL_CA_BUNDLE: <not set>
GDAL_DRIVER_PATH: C:\Users\mjhaw\miniconda3\envs\ping\Library\lib\gdalplugins
HOMEPATH: \Users\mjhaw
LD_PRELOAD: <not set>
PATH: C:\Users\mjhaw\miniconda3\envs\ping;C:\Users\mjhaw\miniconda3\envs\ping\Library\mingw-w64\bin;C:\Users\mjhaw\miniconda3\envs\ping\Library\usr\bin;C:\Users\mjhaw\miniconda3\envs\ping\Library\bin;C:\Users\mjhaw\miniconda3\envs\ping\Scripts;C:\Users\mjhaw\miniconda3\envs\ping\bin;C:\Users\mjhaw\miniconda3\condabin;C:\Program Files\Microsoft MPI\Bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0;C:\WINDOWS\System32\OpenSSH;C:\GDAL;C:\Program Files\GDAL;C:\Program Files\TauDEM\TauDEM5Exe;C:\Program Files\dotnet;C:\Program Files (x86)\dotnet;C:\Users\mjhaw\AppData\Local\Microsoft\WindowsApps;C:\Users\mjhaw\.dotnet\tools
PSMODULEPATH: C:\Users\mjhaw\Documents\WindowsPowerShell\Modules;C:\Program Files\WindowsPowerShell\Modules;C:\WINDOWS\system32\WindowsPowerShell\v1.0\Modules
REQUESTS_CA_BUNDLE: <not set>
SSL_CERT_FILE: <not set>
```

CameronBodine, 2023-11-01:
I tried a complete reinstall (see steps below) but I am unable to reproduce the error.

1. Install Miniconda according to the install directions
2. Update base conda
3. Clean conda
4. Upgrade pip
5. Install libmamba and set as default solver
6. Clone PINGMapper repo
7. Install `ping` environment

Something else to try: I saw that issues with not locating DLL's might be fixed with installing Visual C++ Redistributable (see [this](https://stackoverflow.com/a/38350913) for more information). I recommend downloading the redistributable for Visual Studio 2015, 2017, 2019, and 2022 [linked here](https://learn.microsoft.com/en-us/cpp/windows/latest-supported-vc-redist?view=msvc-170#visual-studio-2015-2017-2019-and-2022). I would install the X64 version. If this doesn't do it, I am really at a loss of any additional fixes. There is the option of trying to install `tk-tools` which I mentioned above.

mjhawker, 2023-11-01:
I was able to get another, older pc today (Win 10 if that makes any difference). Similar software setup with GIS's.

It looks like your program worked as it should. The output folders are full. 

Sorry for the goose chase, very frustrating I'm sure, but I learned a bunch along the way, anyway.

I'm really not sure what is going on with my other pc - it is my daily Dell XPS and no problems otherwise.  There's obviously a ghost in there somewhere! 

Thanks for taking more time than you needed to.  I'll try to follow your progress and updates. 

Looking forward to processing some data and using the output.

CameronBodine, 2023-11-01:
Thanks for the update and glad you got it working on your other computer. Is your newer one running Windows 11? I haven't tested anything on that OS yet. 

I'm really interested in making this work for folks beyond my own research, so that's why I spend the time. And I always learn something along the way.

I will go ahead and close as unresolved. Feel free to reopen if you find or try something else.

collinjroland, 2024-01-06:
I am encountering the same `tkinter` issues on Windows 11. I'm still down the StackExchange rabbit hole on this, with no end in site. Thus far the best explanation I've found is this (https://stackoverflow.com/questions/76105218/why-does-tkinter-or-turtle-seem-to-be-missing-or-broken-shouldnt-it-be-part), I've attempted installs using Anaconda, miniconda, and miniforge, all with identical results. Running
`import tkinter
tkinter._test()`
yields a popup window like it should. 

`(ping) C:\Users\colli\OneDrive\Documents\Software\PINGMapper>python test_PINGMapper.py 1
WARNING:tensorflow:From C:\Users\colli\miniforge3\envs\ping\Lib\site-packages\keras\src\losses.py:2976: The name tf.losses.sparse_softmax_cross_entropy is deprecated. Please use tf.compat.v1.losses.sparse_softmax_cross_entropy instead.

Version:  2.15.0
Eager mode:  True
Traceback (most recent call last):
  File "C:\Users\colli\OneDrive\Documents\Software\PINGMapper\test_PINGMapper.py", line 35, in <module>
    from main_readFiles import read_master_func
  File "C:\Users\colli\OneDrive\Documents\Software\PINGMapper\src\main_readFiles.py", line 32, in <module>
    from class_portstarObj import portstarObj
  File "C:\Users\colli\OneDrive\Documents\Software\PINGMapper\src\class_portstarObj.py", line 31, in <module>
    from funcs_model import *
  File "C:\Users\colli\OneDrive\Documents\Software\PINGMapper\src\funcs_model.py", line 51, in <module>
    from doodleverse_utils.prediction_imports import *
  File "C:\Users\colli\miniforge3\envs\ping\Lib\site-packages\doodleverse_utils\prediction_imports.py", line 32, in <module>
    from tkinter import filedialog
  File "C:\Users\colli\miniforge3\envs\ping\Lib\tkinter\__init__.py", line 38, in <module>
    import _tkinter # If this fails your Python may not be configured for Tk
    ^^^^^^^^^^^^^^^
ImportError: DLL load failed while importing _tkinter: The specified procedure could not be found.`

collinjroland, 2024-01-06:
> I am encountering the same `tkinter` issues on Windows 11. I'm still down the StackExchange rabbit hole on this, with no end in site. Thus far the best explanation I've found is this (https://stackoverflow.com/questions/76105218/why-does-tkinter-or-turtle-seem-to-be-missing-or-broken-shouldnt-it-be-part), I've attempted installs using Anaconda, miniconda, and miniforge, all with identical results. Running `import tkinter tkinter._test()` yields a popup window like it should.
> 
> I successfully installed and tested PINGMapper on a machine running macOS Sononma 14.2.1, but that is not a viable long-term solution for me.
> 
> ```
> 
> (ping) C:\Users\colli\OneDrive\Documents\Software\PINGMapper>python test_PINGMapper.py 1
> WARNING:tensorflow:From C:\Users\colli\miniforge3\envs\ping\Lib\site-packages\keras\src\losses.py:2976: The name tf.losses.sparse_softmax_cross_entropy is deprecated. Please use tf.compat.v1.losses.sparse_softmax_cross_entropy instead.
> 
> Version:  2.15.0
> Eager mode:  True
> Traceback (most recent call last):
>   File "C:\Users\colli\OneDrive\Documents\Software\PINGMapper\test_PINGMapper.py", line 35, in <module>
>     from main_readFiles import read_master_func
>   File "C:\Users\colli\OneDrive\Documents\Software\PINGMapper\src\main_readFiles.py", line 32, in <module>
>     from class_portstarObj import portstarObj
>   File "C:\Users\colli\OneDrive\Documents\Software\PINGMapper\src\class_portstarObj.py", line 31, in <module>
>     from funcs_model import *
>   File "C:\Users\colli\OneDrive\Documents\Software\PINGMapper\src\funcs_model.py", line 51, in <module>
>     from doodleverse_utils.prediction_imports import *
>   File "C:\Users\colli\miniforge3\envs\ping\Lib\site-packages\doodleverse_utils\prediction_imports.py", line 32, in <module>
>     from tkinter import filedialog
>   File "C:\Users\colli\miniforge3\envs\ping\Lib\tkinter\__init__.py", line 38, in <module>
>     import _tkinter # If this fails your Python may not be configured for Tk
>     ^^^^^^^^^^^^^^^
> ImportError: DLL load failed while importing _tkinter: The specified procedure could not be found.
> ```

Update: I successfully installed and tested using WSL2 - Ubuntu; I cannot run gui_main.py, error message contained below. I should be able to use the script version though. Thanks for the great package Cameron!

```
python gui_main.py
Version:  2.15.0
Eager mode:  True
Traceback (most recent call last):
  File "/home/croland/PINGMapper/gui_main.py", line 92, in <module>
    event, values = window.read()
                    ^^^^^^^^^^^^^
  File "/home/croland/mambaforge/envs/ping/lib/python3.11/site-packages/PySimpleGUI/PySimpleGUI.py", line 10079, in read
    results = self._read(timeout=timeout, timeout_key=timeout_key)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/croland/mambaforge/envs/ping/lib/python3.11/site-packages/PySimpleGUI/PySimpleGUI.py", line 10150, in _read
    self._Show()
  File "/home/croland/mambaforge/envs/ping/lib/python3.11/site-packages/PySimpleGUI/PySimpleGUI.py", line 9890, in _Show
    StartupTK(self)
  File "/home/croland/mambaforge/envs/ping/lib/python3.11/site-packages/PySimpleGUI/PySimpleGUI.py", line 16821, in StartupTK
    _get_hidden_master_root()
  File "/home/croland/mambaforge/envs/ping/lib/python3.11/site-packages/PySimpleGUI/PySimpleGUI.py", line 16708, in _get_hidden_master_root
    Window.hidden_master_root = tk.Tk()
                                ^^^^^^^
  File "/home/croland/mambaforge/envs/ping/lib/python3.11/tkinter/__init__.py", line 2326, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
_tkinter.TclError: no display name and no $DISPLAY environment variable
```

CameronBodine, 2024-01-07:
Thanks for report Collin. This is a strange one. I wonder if you can try something for me? With `ping` environment activated, run the following two commands:

```
conda remove -n ping tkinter
```

then reinstall

```
conda install tkinter
```

collinjroland, 2024-01-07:
Indeed! I have the package `tk `installed. When I run` conda install tkinter` I get a `PackagesNotFoundError`.

```
conda install tkinter
Channels:
 - defaults
 - conda-forge
Platform: win-64
Collecting package metadata (repodata.json): done
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - tkinter

Current channels:

  - defaults
  - https://conda.anaconda.org/conda-forge

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
```

CameronBodine, 2024-01-07:
Whoops! How about this:

conda install conda-forge::tk

collinjroland, 2024-01-07:
It seems like the tk package cannot be removed, I assume because it is part of the base distribution. 

```
mamba remove -n ping tk

                  __    __    __    __
                 /  \  /  \  /  \  /  \
                /    \/    \/    \/    \
███████████████/  /██/  /██/  /██/  /████████████████████████
              /  / \   / \   / \   / \  \____
             /  /   \_/   \_/   \_/   \    o \__,
            / _/                       \_____/  `
            |/
        ███╗   ███╗ █████╗ ███╗   ███╗██████╗  █████╗
        ████╗ ████║██╔══██╗████╗ ████║██╔══██╗██╔══██╗
        ██╔████╔██║███████║██╔████╔██║██████╔╝███████║
        ██║╚██╔╝██║██╔══██║██║╚██╔╝██║██╔══██╗██╔══██║
        ██║ ╚═╝ ██║██║  ██║██║ ╚═╝ ██║██████╔╝██║  ██║
        ╚═╝     ╚═╝╚═╝  ╚═╝╚═╝     ╚═╝╚═════╝ ╚═╝  ╚═╝

        mamba (1.4.2) supported by @QuantStack

        GitHub:  https://github.com/mamba-org/mamba
        Twitter: https://twitter.com/QuantStack

█████████████████████████████████████████████████████████████

Removing specs: ['tk']
Transaction

  Prefix: C:\Users\colli\miniforge3\envs\ping

  Removing specs:

   - tk

  Package                                 Version  Build                    Channel         Size
--------------------------------------------------------------------------------------------------
  Remove:
--------------------------------------------------------------------------------------------------

  - __win                                       0  0                        installed
  - affine                                  2.4.0  pyhd8ed1ab_0             conda-forge
  - aom                                     3.7.1  h63175ca_0               conda-forge
  - asttokens                               2.4.1  pyhd8ed1ab_0             conda-forge
  - attrs                                  23.2.0  pyh71513ae_0             conda-forge
  - azure-core-cpp                         1.10.3  h249a519_0               conda-forge
  - azure-storage-blobs-cpp               12.10.0  h91493d7_0               conda-forge
  - azure-storage-common-cpp               12.5.0  h91493d7_2               conda-forge
  - blosc                                  1.21.5  hdccc3a2_0               conda-forge
  - branca                                  0.7.0  pyhd8ed1ab_1             conda-forge
  - brotli                                  1.1.0  hcfcfb64_1               conda-forge
  - brotli-bin                              1.1.0  hcfcfb64_1               conda-forge
  - brotli-python                           1.1.0  py311h12c1d0e_1          conda-forge
  - bzip2                                   1.0.8  hcfcfb64_5               conda-forge
  - c-ares                                 1.24.0  hcfcfb64_0               conda-forge
  - c-blosc2                               2.12.0  h183a6f4_0               conda-forge
  - cached-property                         1.5.2  hd8ed1ab_1               conda-forge
  - cached_property                         1.5.2  pyha770c72_1             conda-forge
  - cairo                                  1.18.0  h1fef639_0               conda-forge
  - certifi                            2023.11.17  py311haa95532_0          pkgs/main
  - cfitsio                                 4.3.1  h9b0cee5_0               conda-forge
  - charls                                  2.4.2  h1537add_0               conda-forge
  - charset-normalizer                      3.3.2  pyhd8ed1ab_0             conda-forge
  - click                                   8.1.7  win_pyh7428d3b_0         conda-forge
  - click-plugins                           1.1.1  py_0                     conda-forge
  - cligj                                   0.7.2  pyhd8ed1ab_1             conda-forge
  - cloudpickle                             3.0.0  pyhd8ed1ab_0             conda-forge
  - colorama                                0.4.6  pyhd8ed1ab_0             conda-forge
  - comm                                    0.2.1  pyhd8ed1ab_0             conda-forge
[... 309 lines cut ...]
                          __win=0=0
       base environment : C:\Users\colli\miniforge3  (writable)
      conda av data dir : C:\Users\colli\miniforge3\etc\conda
  conda av metadata url : None
           channel URLs : https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
                          https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : C:\Users\colli\miniforge3\pkgs
                          C:\Users\colli\.conda\pkgs
                          C:\Users\colli\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\colli\miniforge3\envs
                          C:\Users\colli\.conda\envs
                          C:\Users\colli\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/23.3.1 requests/2.31.0 CPython/3.10.12 Windows/10 Windows/10.0.22621 solver/libmamba co2          administrator : False
             netrc file : C:\Users\colli/.netrc
           offline mode : False

An unexpected error has occurred. Conda has prepared the above report.
```

CameronBodine, 2024-01-07:
Shoot. I wondered if that might be the case. I did find [this explanation](https://stackoverflow.com/a/76105219) describing why `tkinter` may not work. I have also wondered if having ArcGIS installed might mess things up, but that's just a hunch. Any chance you have ArcGIS on your machine?

I know you are up and running with WSL2, but if you are will to keep playing, I was wondering if you could try something else for me. Can you try following the [update installation](https://cameronbodine.github.io/PINGMapper/docs/gettingstarted/UpdateInstallation.html) to see if that might fix the issue?

collinjroland, 2024-01-08:
I attempted the update installation instructions but am still receiving the `tkinter` error. This PC does have ArcGIS installed. I will attempt it on a PC without ArcGIS installed later today. 

I will open a separate issue for this after a bit more troubleshooting, but wanted to mention that I have repeatedly encountered this (memory overload?) error when processing files where each channel is on the order of 30 mB. I an running this  on a machine with 64GB of RAM. 
```
Exporting Banklines...

        Exporting to shapefile...
[Parallel(n_jobs=2)]: Using backend LokyBackend with 2 concurrent workers.
Version:  2.15.0
Version:  2.15.0
Eager mode:  True
Eager mode:  True

!!!!!!!!!!!!!!!!!
An Error Occured.

Please consult the logfile at:
 procData/R00098/logs/error_2024-01-08_1227.txt
```
which yields
```

Error Thrown:
Traceback (most recent call last):
  File "/home/croland/PINGMapper/main_R00057.py", line 262, in <module>
    rectify_master_func(**params)
  File "/home/croland/PINGMapper/src/main_rectify.py", line 417, in rectify_master_func
    psObj._exportBanklines(threadCnt, mosaic_nchunk)
  File "/home/croland/PINGMapper/src/class_portstarObj.py", line 2355, in _exportBanklines
    _ = Parallel(n_jobs= np.min([len(rasterFiles), threadCnt]), verbose=10)(delayed(self._createBanklinePolygon)(f, outDir) for f in rasterFiles)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/croland/miniforge3/envs/ping/lib/python3.11/site-packages/joblib/parallel.py", line 1952, in __call__
    return output if self.return_generator else list(output)
                                                ^^^^^^^^^^^^
  File "/home/croland/miniforge3/envs/ping/lib/python3.11/site-packages/joblib/parallel.py", line 1595, in _get_outputs
    yield from self._retrieve()
  File "/home/croland/miniforge3/envs/ping/lib/python3.11/site-packages/joblib/parallel.py", line 1699, in _retrieve
    self._raise_error_fast()
  File "/home/croland/miniforge3/envs/ping/lib/python3.11/site-packages/joblib/parallel.py", line 1734, in _raise_error_fast
    error_job.get_result(self.timeout)
  File "/home/croland/miniforge3/envs/ping/lib/python3.11/site-packages/joblib/parallel.py", line 736, in get_result
    return self._return_or_raise()
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/croland/miniforge3/envs/ping/lib/python3.11/site-packages/joblib/parallel.py", line 754, in _return_or_raise
    raise self._result
joblib.externals.loky.process_executor.TerminatedWorkerError: A worker process managed by the executor was unexpectedly terminated. This could be caused by a segmentation fault while calling the function or by an excessive memory usage causing the Operating System to kill the worker.

The exit codes of the workers are {SIGSEGV(-11), SIGSEGV(-11)}
```

CameronBodine, 2024-01-22:
Sorry to not get back to you sooner. I was wondering if you are behind a government firewall? This may be causing the issue. If possible, are you able to install, both miniconda and PINGMapper, on another network?

collinjroland, 2024-01-22:
Hey Cameron, no worries. I've been trying this on a PC that is not the government network as well as a government PC. Maybe this is a Windows 11 issue based on the above dialogue with mjhawker, or are you aware of users successfully installing on Windows 11 machines? Thanks for doing your best to troubleshoot.

CameronBodine, 2024-01-22:
I thought for sure it would be a network or government pc issue. Bummer. I have successfully installed and run on a Windows 11 pc with miniconda. The computer was fresh out of the box though. Any chance your computers were updated from Windows 10 to 11? I have had unrelated issues making those types of upgrades...

I suppose as long as you are able to run on WSL2, then you are good to go. And I can recommend this workaround for others experiencing similar issues. 

If, and only if, you are still interested in trying to get it to run in native windows, you could try lowering the python version in the `PIMGMapper/conda/PINGMapper.yml` file. For example, `python==3.10.*` instead of `python==3.11.*`. Then reinstall the environment. I will keep looking around to see if there might be other solutions.

CameronBodine, 2024-01-31:
@collinjroland keep me posted on the model training. If you are interested, it would be neat to make it available in a future version of PINGMapper! Definitely reach out as you move forward.

closed 2024-01-31 by CameronBodine