CameronBodine/PINGMapper #105: TypeError error and missing package "sumpy" while running "test_PINGMapper.py"

mihykavetu, 2024-01-22:
I installed the program according to the instructions. The only difference was that I already had an anaconda installation which I built the environment in. The environment was built successfully, but running the "test_PINGMapper.py" for the small dataset first returned an error:

```
> Traceback (most recent call last):
>   File "C:\Users\mikko\PINGMapper\test_PINGMapper.py", line 35, in <module>
>     from main_readFiles import read_master_func
>   File "C:\Users\mikko\PINGMapper\src\main_readFiles.py", line 32, in <module>
>     from class_portstarObj import portstarObj
>   File "C:\Users\mikko\PINGMapper\src\class_portstarObj.py", line 31, in <module>
>     from funcs_model import *
>   File "C:\Users\mikko\PINGMapper\src\funcs_model.py", line 42, in <module>
>     from transformers import TFSegformerForSemanticSegmentation
>   File "C:\Users\mikko\.conda\envs\ping\Lib\site-packages\transformers\__init__.py", line 26, in <module>
>     from . import dependency_versions_check
>   File "C:\Users\mikko\.conda\envs\ping\Lib\site-packages\transformers\dependency_versions_check.py", line 16, in <module>
>     from .utils.versions import require_version, require_version_core
>   File "C:\Users\mikko\.conda\envs\ping\Lib\site-packages\transformers\utils\__init__.py", line 32, in <module>
>     from .generic import (
>   File "C:\Users\mikko\.conda\envs\ping\Lib\site-packages\transformers\utils\generic.py", line 432, in <module>
>     import torch.utils._pytree as _torch_pytree
>   File "C:\Users\mikko\AppData\Roaming\Python\Python311\site-packages\torch\__init__.py", line 1504, in <module>
>     from . import masked
>   File "C:\Users\mikko\AppData\Roaming\Python\Python311\site-packages\torch\masked\__init__.py", line 3, in <module>
>     from ._ops import (
>   File "C:\Users\mikko\AppData\Roaming\Python\Python311\site-packages\torch\masked\_ops.py", line 11, in <module>
>     from torch._prims_common import corresponding_real_dtype
>   File "C:\Users\mikko\AppData\Roaming\Python\Python311\site-packages\torch\_prims_common\__init__.py", line 23, in <module>
>     import sympy
> ModuleNotFoundError: No module named 'sympy'
```

This was fixed by: pip install sumpy.

I ran the "test_PINGMapper.py" again and now got the following error: 

```
Traceback (most recent call last):
  File "C:\Users\mikko\PINGMapper\test_PINGMapper.py", line 293, in <module>
    read_master_func(**params)
TypeError: read_master_func() got an unexpected keyword argument 'pix_res'
```

It seems that the parameters for "read_master_func()" in "main_readFiles.py" needs to include "pix_res_map" and "pix_res_son" -variables. The issue was solved by removing the "pix_res=pix_res" from Line 277 of "test_PINGMapper.py" and inserting "pix_res_map = pix_res" and "pix_res_son = pix_res" into the arguments.

CameronBodine, 2024-01-22:
Thanks for reporting these issues and the steps you took to resolve them! I added the ability to export sonar mosaics and substrate maps at different pixel resolutions ([d42838](https://github.com/CameronBodine/PINGMapper/commit/40d5855c981b5df5a8cbad07b7c9801045dfc4e7)) but forgot to update the test script.  This has been resolved.

Several folks have been having issues with the installation process (e.g., #101). I added an FAQ to help folks [troubleshoot the error](https://cameronbodine.github.io/PINGMapper/docs/faq.html#modulenotfounderror). Would you be willing to supply additional information to help document this issue?

1. What country is the computer located in?
2. What is the computer's operating system and version? (e.g., Windows 11, Ubuntu 23.10, Mac OS X Leopard. 10.5.8, etc.)
3. What kind of network is the computer connected to? (e.g., Home/Personal, Local/State/Federal Government, University, Private Company, etc.)
4. Does the computer have ArcGIS installed?
5. What version of Anaconda or Miniconda is installed? (run `conda list anaconda` or `conda list miniconda`)

Thank you in advance!

mihykavetu, 2024-01-23:
Glad to be of an assistance. Here is the additional information requested:

1. Finland.
2. Windows 11
3. Private company local network
4. QGIS is installed
5. Output of "conda list anaconda":
```
# packages in environment at C:\ProgramData\anaconda3:
#
# Name                    Version                   Build  Channel
anaconda-anon-usage       0.4.2           py311hfc23b7f_0
anaconda-catalogs         0.2.0           py311haa95532_0
anaconda-client           1.12.1          py311haa95532_0
anaconda-cloud-auth       0.1.4           py311haa95532_0
anaconda-navigator        2.5.2           py311haa95532_0
anaconda-project          0.11.1          py311haa95532_0
```

CameronBodine, 2024-01-23:
Ok, thank you for the info! My best guess it is something to do with the Anaconda install on Windows 11, but I am not sure. But this will help for documentation. Thanks again!

closed 2024-01-23 by CameronBodine