CameronBodine/PINGMapper #111: Linux install??

mmcintosh, 2024-01-25:
Can this project be installed in a linux distro (Ubuntu?) 

TIA

CameronBodine, 2024-01-25:
Certainly! Just follow the [installation instructions](https://cameronbodine.github.io/PINGMapper/docs/gettingstarted/Installation.html). If your Ubuntu version already includes Anaconda, you can proceed from [step 2](https://cameronbodine.github.io/PINGMapper/docs/gettingstarted/Installation.html#step-2).

mmcintosh, 2024-01-28:
I do have anaconda installed version conda 4.12.0 and I have attempted to install and get a few errors that I think are environmental issues one was an upgrade to conda and pip so I back tracked and looked in the ubuntu directory adn tried to follow the directions there:  

This produced and error as well so I tried upgrading pip as expressed at the bottom of the code listed below. I then tried running the first test again.

python test_PINGMapper.py 1

getting this response 
```
Traceback (most recent call last):
  File "/home/siddhartha/anaconda3/PythonPackages/PINGMapper/test_PINGMapper.py", line 31, in <module>
    import sys, os, zipfile, requests
ModuleNotFoundError: No module named 'requests'
```
Attempts to install the missing module request failed as well 

```
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.

PackagesNotFoundError: The following packages are not available from current channels:

  - request

```
I assume I have the module name wrong 
```
python3 -m venv .venv
source .venv/bin/activate
pip install -r ubuntu/requirements.txt

Following from the directions from step 2 seem a bit confusing as they are all windows commands and I am have trouble converting them to linux concepts. I had to remove the majority of the response from th egdal install as it was to long ot include. I can attach it later a file if needed. 

Any assistance would be apprectiated

This installed a load of programs and I was hoping this had solved the issue. (nope but it did install a lot of programs) --> sorry for the amount of noise this creates:

Collecting absl-py==2.0.0
  Downloading absl_py-2.0.0-py3-none-any.whl (130 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 130.2/130.2 KB 742.1 kB/s eta 0:00:00
Collecting affine==2.4.0
  Downloading affine-2.4.0-py3-none-any.whl (15 kB)
Collecting astunparse==1.6.3
  Downloading astunparse-1.6.3-py2.py3-none-any.whl (12 kB)
Collecting attrs==23.1.0
  Downloading attrs-23.1.0-py3-none-any.whl (61 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 61.2/61.2 KB 3.1 MB/s eta 0:00:00
Collecting cachetools==5.3.1
  Downloading cachetools-5.3.1-py3-none-any.whl (9.3 kB)
Collecting certifi==2023.7.22
  Downloading certifi-2023.7.22-py3-none-any.whl (158 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 158.3/158.3 KB 1.3 MB/s eta 0:00:00
Collecting charset-normalizer==3.2.0
  Downloading charset_normalizer-3.2.0-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (202 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 202.1/202.1 KB 1.7 MB/s eta 0:00:00
Collecting click==8.1.7
  Downloading click-8.1.7-py3-none-any.whl (97 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 97.9/97.9 KB 4.4 MB/s eta 0:00:00
Collecting click-plugins==1.1.1
  Downloading click_plugins-1.1.1-py2.py3-none-any.whl (7.5 kB)
Collecting cligj==0.7.2
  Downloading cligj-0.7.2-py3-none-any.whl (7.1 kB)
Collecting contourpy==1.1.1
  Downloading contourpy-1.1.1-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (301 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 301.9/301.9 KB 2.2 MB/s eta 0:00:00
Collecting cycler==0.12.0
  Downloading cycler-0.12.0-py3-none-any.whl (8.2 kB)
Collecting doodleverse-utils==0.0.35
  Downloading doodleverse_utils-0.0.35-py3-none-any.whl (44 kB)
[... 305 lines cut ...]
      copying gdal-utils/osgeo_utils/samples/gdallocationinfo.py -> build/lib.linux-x86_64-3.9/osgeo_utils/samples
      copying gdal-utils/osgeo_utils/samples/build_jp2_from_xml.py -> build/lib.linux-x86_64-3.9/osgeo_utils/samples
      copying gdal-utils/osgeo_utils/samples/gdal_vrtmerge.py -> build/lib.linux-x86_64-3.9/osgeo_utils/samples
      copying gdal-utils/osgeo_utils/samples/vec_tr.py -> build/lib.linux-x86_64-3.9/osgeo_utils/samples
      copying gdal-utils/osgeo_utils/samples/jpeg_in_tiff_extract.py -> build/lib.linux-x86_64-3.9/osgeo_utils/samples
      copying gdal-utils/osgeo_utils/samples/dump_jp2.py -> build/lib.linux-x86_64-3.9/osgeo_utils/samples
      copying gdal-utils/osgeo_utils/samples/load2odbc.py -> build/lib.linux-x86_64-3.9/osgeo_utils/samples
      copying gdal-utils/osgeo_utils/samples/gdalident.py -> build/lib.linux-x86_64-3.9/osgeo_utils/samples
      copying gdal-utils/osgeo_utils/samples/epsg_tr.py -> build/lib.linux-x86_64-3.9/osgeo_utils/samples
      copying gdal-utils/osgeo_utils/samples/validate_jp2.py -> build/lib.linux-x86_64-3.9/osgeo_utils/samples
      running build_ext
 
      error: command '/usr/bin/gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> GDAL

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
WARNING: You are using pip version 22.0.4; however, version 23.3.2 is available.
```

CameronBodine, 2024-01-29:
Thanks for the report and sorry that you are having issues. Note that I do most of the development and testing of PING-Mapper on Ubuntu, so I can confirm that it does work.

I confess I am a bit confused about your troubleshooting steps. It seems you had to bounce around and tried a few things and it is not clear what caused an error and what the error was. Providing the exact commands issued and the resulting error will help me troubleshoot this with you. Let's see if we can get it figured out.

There seems to be something off with your Anaconda install. Open your terminal and update conda. On ubuntu, conda should be available in the standard terminal. You will know that this is the case if you open the terminal and see `(base)` before your username and current directory. 

Now use the terminal to navigate to the location where you want to install pingmapper. This must be a directory where you have write permissions.

Now proceed with [Step 3](https://cameronbodine.github.io/PINGMapper/docs/gettingstarted/Installation.html#step-3).

If the gdal error you previously reported persists, look to see if the line below is in the error:

```
Using legacy 'setup.py install' for GDAL, since package 'wheel' is not installed.
```

If it is, try installing wheel with:

```
conda install conda-forge::wheel
```

Then try install the pingmapper environment. Let me know how it goes! If you run into issues, please report the exact command and the resulting error. If the error is too long, please attach as txt.

mmcintosh, 2024-01-30:
Cameron,

Will do and let you know how it goes. I will get back to it in the next day
or so and will alot more time to both document and troubleshoot the issues.
Thanks for your time.

Mark

mmcintosh, 2024-01-31:
In my attempts to fix my installation I decided to just start over from scratch. 

Downloaded newest version from Anaconda --> Anaconda3-2023.09-0-Linux-x86_64.sh

initiated install via bash command: ---> bash Anaconda3-2023.09-0-Linux-x86_64.sh

installation finished.
Do you wish to update your shell profile to automatically initialize conda?
This will activate conda on startup and change the command prompt when activated.
If you'd prefer that conda's base environment not be activated on startup,
   run the following command when conda is activated:

#conda config --set auto_activate_base false

You can undo this by running `conda init --reverse $SHELL`? [yes|no]
[no] >>> yes
no change     /home/siddhartha/anaconda3/condabin/conda
no change     /home/siddhartha/anaconda3/bin/conda
no change     /home/siddhartha/anaconda3/bin/conda-env
no change     /home/siddhartha/anaconda3/bin/activate
no change     /home/siddhartha/anaconda3/bin/deactivate
no change     /home/siddhartha/anaconda3/etc/profile.d/conda.sh
no change     /home/siddhartha/anaconda3/etc/fish/conf.d/conda.fish
no change     /home/siddhartha/anaconda3/shell/condabin/Conda.psm1
no change     /home/siddhartha/anaconda3/shell/condabin/conda-hook.ps1
no change     /home/siddhartha/anaconda3/lib/python3.11/site-packages/xontrib/conda.xsh
no change     /home/siddhartha/anaconda3/etc/profile.d/conda.csh
no change     /home/siddhartha/.bashrc
No action taken.

So installed over and in a directory I have full control over.

Upon sourcing bash I got base activated 

#source ~/.bashrc

(base) sidhatha@siddharthaX1:~/sonar2$ 

conda list shows packages:

#conda list

# packages in environment at /home/siddhartha/anaconda3:
#
# Name                    Version                   Build  Channel
_anaconda_depends         2023.09             py311_mkl_1  
_libgcc_mutex             0.1                        main  
_openmp_mutex             5.1                       1_gnu  
abseil-cpp                20211102.0           hd4dd3e8_0  
aiobotocore               2.5.0           py311h06a4308_0  
aiofiles                  22.1.0          py311h06a4308_0  
aiohttp                   3.8.5           py311h5eee18b_0  
aioitertools              0.7.1              pyhd3eb1b0_0  
aiosignal                 1.2.0              pyhd3eb1b0_0  
aiosqlite                 0.18.0          py311h06a4308_0  
alabaster                 0.7.12             pyhd3eb1b0_0  
anaconda-anon-usage       0.4.2           py311hfc0e8ea_0  
anaconda-catalogs         0.2.0           py311h06a4308_0  
anaconda-client           1.12.1          py311h06a4308_0  
anaconda-cloud-auth       0.1.3           py311h06a4308_0  
anaconda-navigator        2.5.0           py311h06a4308_0  
anaconda-project          0.11.1          py311h06a4308_0  
anyio                     3.5.0           py311h06a4308_0  
aom                       3.6.0                h6a678d5_0  
appdirs                   1.4.4              pyhd3eb1b0_0  
argon2-cffi               21.3.0             pyhd3eb1b0_0  
argon2-cffi-bindings      21.2.0          py311h5eee18b_0  
arrow                     1.2.3           py311h06a4308_1  
arrow-cpp                 11.0.0               h374c478_2  
astroid                   2.14.2          py311h06a4308_0  
[... 826 lines cut ...]
Installing collected packages: pysimplegui, libclang, flatbuffers, wrapt, werkzeug, versioneer, typing-extensions, tqdm, termcolor, tensorflow-io-gcs-filesystem, tensorflow-estimator, tensorboard-data-server, safetensors, regex, pyyaml, pyasn1, psutil, protobuf, opt-einsum, oauthlib, ml-dtypes, markdown, keras, grpcio, google-pasta, gast, fsspec, filelock, cachetools, astunparse, absl-py, rsa, requests-oauthlib, pyasn1-modules, huggingface-hub, doodleverse_utils, tokenizers, google-auth, transformers, google-auth-oauthlib, tensorboard, tensorflow
Successfully installed absl-py-2.1.0 astunparse-1.6.3 cachetools-5.3.2 doodleverse_utils-0.0.35 filelock-3.13.1 flatbuffers-23.5.26 fsspec-2023.12.2 gast-0.5.4 google-auth-2.27.0 google-auth-oauthlib-1.2.0 google-pasta-0.2.0 grpcio-1.60.0 huggingface-hub-0.20.3 keras-2.15.0 libclang-16.0.6 markdown-3.5.2 ml-dtypes-0.2.0 oauthlib-3.2.2 opt-einsum-3.3.0 protobuf-4.23.4 psutil-5.9.8 pyasn1-0.5.1 pyasn1-modules-0.3.0 pysimplegui-4.60.5 pyyaml-6.0.1 regex-2023.12.25 requests-oauthlib-1.3.1 rsa-4.9 safetensors-0.4.2 tensorboard-2.15.1 tensorboard-data-server-0.7.2 tensorflow-2.15.0.post1 tensorflow-estimator-2.15.0 tensorflow-io-gcs-filesystem-0.35.0 termcolor-2.4.0 tokenizers-0.15.1 tqdm-4.66.1 transformers-4.37.2 typing-extensions-4.9.0 versioneer-0.29 werkzeug-3.0.1 wrapt-1.14.1

done
#
# To activate this environment, use
#
#     $ conda activate ping
#
# To deactivate an active environment, use
#
#     $ conda deactivate

now to activate the environment:

conda activate ping
(base) sidhatha@siddharthaX1:~/anaconda3/PythonPackages/PINGMapper$   conda activate ping

response cleared  --> now in the ping environment

(ping) sidhatha@siddharthaX1:~/anaconda3/PythonPackages/PINGMapper$

Will move on to test and report

Thx!

mmcintosh, 2024-01-31:
Test occurred but did get an error: test results--> Version:  2.15.0
Eager mode:  True
Need to download small recording dataset...

Downloaded and extracted ./exampleData/Test_Small_DS.zip
1
['./exampleData/Test-Small-DS/B000.SON', './exampleData/Test-Small-DS/B001.SON', './exampleData/Test-Small-DS/B002.SON', './exampleData/Test-Small-DS/B003.SON']

 ***User Parameters***
| logfilename          : ./procData/PINGMapper-Test-Small-DS/logs/log_2024-01-30_2158.txt |
| project_mode         : 1          |
| script               : ['/home/siddhartha/anaconda3/PythonPackages/PINGMapper/test_PINGMapper.py', 'test_PINGMapper_2024-01-30_2158.py'] |
| humFile              : ./exampleData/Test-Small-DS.DAT |
| sonFiles             : ['./exampleData/Test-Small-DS/B000.SON', './exampleData/Test-Small-DS/B001.SON', './exampleData/Test-Small-DS/B002.SON', './exampleData/Test-Small-DS/B003.SON'] |
| projDir              : ./procData/PINGMapper-Test-Small-DS |
| tempC                : 10         |
| nchunk               : 500        |
| exportUnknown        : False      |
| fixNoDat             : True       |
| threadCnt            : 0          |
| pix_res_son          : 0.05       |
| pix_res_map          : 0.25       |
| x_offset             : 0.0        |
| y_offset             : 0.0        |
| egn                  : True       |
| egn_stretch          : 1          |
| egn_stretch_factor   : 0.5        |
| tileFile             : .jpg       |
| wcp                  : True       |
| wcr                  : True       |
| lbl_set              : 2          |
| spdCor               : 1          |
| maxCrop              : True       |
| USE_GPU              : False      |
| remShadow            : 1          |
| detectDep            : 1          |
| smthDep              : True       |
| adjDep               : 10         |
| pltBedPick           : True       |
| rect_wcp             : True       |
| rect_wcr             : True       |
| son_colorMap         : Greys_r    |
| pred_sub             : 1          |
| map_sub              : 1          |
| export_poly          : True       |
| map_predict          : False      |
| pltSubClass          : True       |
| map_class_method     : max        |
| mosaic_nchunk        : 0          |
| mosaic               : 1          |
| map_mosaic           : 1          |
| banklines            : True       |

===========================================
===========================================
***** READING *****
working on ./procData/PINGMapper-Test-Small-DS

Downloading segmentation models (v1.0):
https://zenodo.org/records/10092289/files/PINGMapperv2.0_SegmentationModelsv1.0.zip?download=1
Model download success!

Current CPU/RAM Usage:
________________________
CPU % | RAM % | RAM [GB]
________________________
6.2   | 26.2  | 2.9  
________________________

Getting DAT Metadata...
[... 1910 lines cut ...]
Version:  2.15.0
Eager mode:  True
config.json: 100%|███████████████████████████████████████████████████████████████████████████████████████████| 70.0k/70.0k [00:00<00:00, 1.55MB/s]
tf_model.h5:   0%|                                                                                                    | 0.00/14.6M [00:00<?, ?B/s]Version:  2.15.0
Eager mode:  True
Version:  2.15.0
Eager mode:  True
tf_model.h5: 100%|███████████████████████████████████████████████████████████████████████████████████████████| 14.6M/14.6M [00:01<00:00, 7.99MB/s]
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
I0000 00:00:1706670326.442588  264859 device_compiler.h:186] Compiled cluster using XLA!  This line is logged at most once for the lifetime of the process.
I0000 00:00:1706670326.442588  264857 device_compiler.h:186] Compiled cluster using XLA!  This line is logged at most once for the lifetime of the process.
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
I0000 00:00:1706670326.533227  264854 device_compiler.h:186] Compiled cluster using XLA!  This line is logged at most once for the lifetime of the process.
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
I0000 00:00:1706670326.929771  264853 device_compiler.h:186] Compiled cluster using XLA!  This line is logged at most once for the lifetime of the process.
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
I0000 00:00:1706670327.071246  264856 device_compiler.h:186] Compiled cluster using XLA!  This line is logged at most once for the lifetime of the process.
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
I0000 00:00:1706670340.829795  264858 device_compiler.h:186] Compiled cluster using XLA!  This line is logged at most once for the lifetime of the process.
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
I0000 00:00:1706670343.067362  264855 device_compiler.h:186] Compiled cluster using XLA!  This line is logged at most once for the lifetime of the process.

!!!!!!!!!!!!!!!!!
An Error Occured.

mmcintosh, 2024-01-31:
Error log attached
[error_2024-01-30_2158.txt](https://github.com/CameronBodine/PINGMapper/files/14106778/error_2024-01-30_2158.txt)

CameronBodine, 2024-01-31:
Excellent! Though it did not finish, it appears everything installed correctly. Thank you for working through this!

The problem now lies with a memory error. It is an issue with PINGMappers workflows. I will investigate for memory leaks and report back.

mmcintosh, 2024-01-31:
KK no worries it does appear to be working now and I am running the larger
test just to see the results. Next is to get some of my own equipment todo
mapping in the Chesapeake bay. I may ping you for some recommendations if
you don’t mind.

Regards and thx,

Mark

mmcintosh, 2024-01-31:
Question? Is it possible to export an environment to pass on to another
person or system?

Mark

CameronBodine, 2024-01-31:
Glad to hear you are up and running!

> Question? Is it possible to export an environment to pass on to another
person or system?

I am not sure, but I would not recommend this. It is better to build a conda environment on each system as the solver will take the new systems settings/configuration into account. I realize the install is a bit tedious, but it will be less so now that you have been through it and provided excellent documentation.

Closing as complete!

For my own notes:
> Error log attached
[error_2024-01-30_2158.txt](https://github.com/CameronBodine/PINGMapper/files/14106778/error_2024-01-30_2158.txt)

This may be due to downloading SegFormer model, not a memory error. Needs further investigation.

closed 2024-01-31 by CameronBodine

mmcintosh, 2024-01-31:
Going to go ahead and run the longer test to see if it errors as well or passes. ----> it also erred out but a different manor:

Error Thrown:
Traceback (most recent call last):
  File "/home/siddhartha/anaconda3/PythonPackages/PINGMapper/test_PINGMapper.py", line 307, in <module>
    rectify_master_func(**params)
  File "/home/siddhartha/anaconda3/PythonPackages/PINGMapper/src/main_rectify.py", line 419, in rectify_master_func
    psObj._exportBanklines(threadCnt, mosaic_nchunk)
  File "/home/siddhartha/anaconda3/PythonPackages/PINGMapper/src/class_portstarObj.py", line 2379, in _exportBanklines
    _ = Parallel(n_jobs= np.min([len(rasterFiles), threadCnt]), verbose=10)(delayed(self._createBanklinePolygon)(f, outDir) for f in rasterFiles)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/siddhartha/anaconda3/envs/ping/lib/python3.11/site-packages/joblib/parallel.py", line 1952, in __call__
    return output if self.return_generator else list(output)
                                                ^^^^^^^^^^^^
  File "/home/siddhartha/anaconda3/envs/ping/lib/python3.11/site-packages/joblib/parallel.py", line 1595, in _get_outputs
    yield from self._retrieve()
  File "/home/siddhartha/anaconda3/envs/ping/lib/python3.11/site-packages/joblib/parallel.py", line 1699, in _retrieve
    self._raise_error_fast()
  File "/home/siddhartha/anaconda3/envs/ping/lib/python3.11/site-packages/joblib/parallel.py", line 1734, in _raise_error_fast
    error_job.get_result(self.timeout)
  File "/home/siddhartha/anaconda3/envs/ping/lib/python3.11/site-packages/joblib/parallel.py", line 736, in get_result
    return self._return_or_raise()
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/siddhartha/anaconda3/envs/ping/lib/python3.11/site-packages/joblib/parallel.py", line 754, in _return_or_raise
    raise self._result
joblib.externals.loky.process_executor.TerminatedWorkerError: A worker process managed by the executor was unexpectedly terminated. This could be caused by a segmentation fault while calling the function or by an excessive memory usage causing the Operating System to kill the worker.

The exit codes of the workers are {SIGKILL(-9)}

mmcintosh, 2024-01-31:
Sorry I did not notice this was closed and that is just fine once again thanks for the encouragement and assistance.

CameronBodine, 2024-01-31:
Looks like you have 12 gb of ram? IYou could try passing a -2 or -4 to the Thread Count (`threadCnt`) parameter. This will reduce the number of threads used during processing and thus, the amount of ram needed. If this continues to be an issue, please create a new issue.