CameronBodine/PINGMapper #113: Example Data use in HumViewer

mmcintosh, 2024-02-08:
While we get set up to use PINGMapper we wanted to test the viewer used for your example [HumViewer](https://humviewer.cm-johansen.dk/) as this would allow us to see how to manipulate data and views. We looked at and had hoped to use the example data that was downloaded for the test (python test_PINGMapper.py 1 & 2) but it fails to import into HumViewer.

What we would like to know is can we use that data and import it into the viewer or is there data available that would work with both PINGMapper and Humviewer that can be added to the example data or that we can download from the project page?

We have been looking elsewhere for data that we could practice with but have not been successful in finding any that has worked. Does anyone have nay or know where we can get some test data that would help?

Thanks for any assistance,

Mark

CameronBodine, 2024-02-09:
Thanks for your inquiry. I believe the short example recording should work with HumViewer. If you run the [test](https://cameronbodine.github.io/PINGMapper/docs/gettingstarted/Testing.html#small-dataset-test), it will download to the `exampleData` folder. Full disclaimer: HumViewer is not my software, so I do not have the ability to make any modifications to the software. Hope this helps!

mmcintosh, 2024-02-10:
I will test and update thanks!

CameronBodine, 2024-02-10:
Here is another recording from the Suwannee River in Florida, USA that I just confirmed will work with HumViewer:

https://drive.google.com/file/d/1zjq8OFuNNJKt6SgrLLAD7w37GkqupJfB/view?usp=drive_link

mmcintosh, 2024-02-10:
That one worked thx!! The small one did not but I do not know if that is also because the test itself failed or it was just not compatible.

closed 2024-02-10 by CameronBodine

CameronBodine, 2024-02-10:
FYI A word of caution: PING-Mapper's substrate model was not trained from the substrates present in the recording I provided. For example, the recording I provided likely has aquatic vegetation, but the current substrate model does not have a class for that. So be careful if you are making comparisons between your interpretation of the imagery and PING-Mappers classification.

mmcintosh, 2024-02-10:
Thank you for that information we will keep it in mind.