CameronBodine/PINGMapper #114: Improve coordinates of track outputs

taoansf, 2024-03-14:
### Describe the bug
The coordinates of track outputs are not good.

### To Reproduce
Run default configurations of Humminbird helix 7 record data.

### Expected behavior 
I expected to get high quality track coordinates as [SonarTRX](https://www.sonartrx.com/) software can do.

### Screenshots
B001_ds_highfreq_meta is the yellow track
Trackline_Smth_ss_port is the red track

[image]

[B001_ds_highfreq_meta.zip](https://github.com/CameronBodine/PINGMapper/files/14601473/B001_ds_highfreq_meta.zip)

[Trackline_Smth_ss_port.zip](https://github.com/CameronBodine/PINGMapper/files/14601476/Trackline_Smth_ss_port.zip)

[log_2024-03-05_1848.txt](https://github.com/CameronBodine/PINGMapper/files/14601487/log_2024-03-05_1848.txt)

### Desktop (please complete the following information):
 - OS: WINDOWS 11

### Additional context
Is there same configuration to improve the numerical precision of track points coordinates?

With best regards,

Taoan

CameronBodine, 2024-03-14:
Hi,
Sorry you are experiencing issues other than expected. PING-Mapper is designed to produce the highest quality imagery possible, which requires smoothing of the track line. 

If you are only interested in the depth data, please try creating a point shape file using the raw coordinates in the csv files in the meta folder.

Let me know if you have any other questions.

taoansf, 2024-03-14:
Hello,

Yes, I am interested in the depth data. The attached picture was generated using shapefiles created from the attached raw and smoothed coordinates from meta folder outputs.

I would like to gain some insights into the ping-mapper smoothing method to improve my track data output.

Could you give me some technical reference on this?

Thank you for your help.

Taoan

CameronBodine, 2024-06-10:
Hi Taoan, did this address your question? Can I close this issue?

closed 2024-06-10 by taoansf