CameronBodine/PINGMapper #115: Error when exporting GeoTiffs "array memory error"

AlexandreAvz, 2024-04-24:
This is the log the appears in the output folder:
[log_2024-04-24_1027.txt](https://github.com/CameronBodine/PINGMapper/files/15095132/log_2024-04-24_1027.txt)

I noticed it was missing the last part of the text that appeared in the prompt, so I have pasted it here:
[error_rec8.txt](https://github.com/CameronBodine/PINGMapper/files/15095193/error_rec8.txt)

Here is the link for the recorded file (annexed and WeTransfer):
[rec_8_dando_errado.zip](https://github.com/CameronBodine/PINGMapper/files/15095252/rec_8_dando_errado.zip)

https://wetransfer.com/downloads/72c78a63b770fd56268bc85616286af220240424134635/92e36433313b19c2bb7a1dc44edd361320240424134653/b5689a?trk=TRN_TDL_01&utm_campaign=TRN_TDL_01&utm_medium=email&utm_source=sendgrid

This is recording number 8, the previous 7 were processed without any issue. I have tried changing the number of chunks per mosaic setting but it does not seem to make a difference. The process goes ok until it arrives at the "Exporting GeoTiffs" part. I apologize in advance in case this is a dumb issue, I have zero previous experience with python and etc.

CameronBodine, 2024-04-24:
Thanks for the feedback! Not a dumb issue at all. It is a memory issue. Can you try re-running but specify a small thread count, like 4? Your machine just doesn't have enough resources to process 10 chunks at once. Let me know how that goes.

AlexandreAvz, 2024-04-29:
> Thanks for the feedback! Not a dumb issue at all. It is a memory issue. Can you try re-running but specify a small thread count, like 4? Your machine just doesn't have enough resources to process 10 chunks at once. Let me know how that goes.

Thanks for the feedback Dr. Bodine, 4 threads crashed my machine for some reason but 2 threads worked! This was my longest recording yet, I'll be sure to lower the thread count for similar ones in the future.

closed 2024-04-29 by AlexandreAvz