CameronBodine/PINGMapper #116: Converting images in rect_wcr folder into corresponding images in 'for_label' folder

bestplanetarian, 2024-05-26:
Currently I am working Side-scan-sonar project.

Because my teammates are labeling on the big sonar mosaic rather than small images in 'for_label' folder, the big mosaic are merged from small images in rect_wcr folder, I need to convert images in rect_wcr folder into corresponding images in 'for_label' folder.

How can I do it? which part of PINGMapper code should I use?

CameronBodine, 2024-06-10:
Thanks for your interest and very sorry for the delay. PM does not currently have a workflow for converting rectified imagery back into a waterfall type of image. This would certainly be a beneficial enhancement. I don't currently have time to investigate this. However, if you do, you can look at `rectSonParallel` [function ](https://github.com/CameronBodine/PINGMapper/blob/beee44361340d55c3e56f2e9dd37ff4d0cd49b99/src/class_rectObj.py#L1086). This function calculates the mapping from the non-rectified (waterfall) pixels to the geometric coordinates. I hope this helps. If you do generate a workflow to do this, I would be happy to implement your approach!

CameronBodine, 2024-11-25:
Closing due to inactivity.

closed 2024-11-25 by CameronBodine