CameronBodine/PINGMapper #118: IndexError: list index out of range

Jdubs6, 2024-06-07:
Hey Cam! So sorry to bother you, I've just been running into a couple index issues when trying to process the Sonar data from the DAT files on GUI. While processing, the error looks like this: 

Locating missing pings and adding NoData...
Traceback (most recent call last):
  File "C:\Users\27imb\PythonPackages\PINGMapper\gui_main.py", line 286, in <module>
    read_master_func(**params)
  File "C:\Users\27imb\PythonPackages\PINGMapper\src\main_readFiles.py", line 751, in read_master_func
    rowsToProc.append((rowsToProc[-1][-1], rowCnt))
                       ~~~~~~~~~~^^^^
IndexError: list index out of range

It then kicks me back to the standard command line without outputting the jpg. The software does create a folder with "logs" and "meta" data, with excel spreadsheets and txt files. I am using a DAT file called "Rec00008" with a folder also named "Rec00008" that includes the SON and IDX files. 

Additionally, while testing the PINGMapper software, this warning shows up: 

WARNING:tensorflow:From C:\Users\27imb\miniconda3\envs\ping\Lib\site-packages\tf_keras\src\losses.py:2976: The name tf.losses.sparse_softmax_cross_entropy is deprecated. Please use tf.compat.v1.losses.sparse_softmax_cross_entropy instead.

This could be attributing to the index error shown earlier. 

Hope this is enough information, let me know if there's anything else I need to provide! I've attached a few screenshots of the issues as well. Thanks for the help, have a great day! 

JW

CameronBodine, 2024-06-07:
Hi! Sorry for the issue. Can you attach the two log files? And can you send a link where I can download the problematic recording?

CameronBodine, 2024-06-07:
Also, the `tf.losses.sparse_softmax_cross_entropy` is just a warning and won't be causing the issue. I think it is something I need to report for another package.

Jdubs6, 2024-06-07:
[Anaconda Prompt (miniconda3).txt](https://github.com/user-attachments/files/15741921/Anaconda.Prompt.miniconda3.txt)
[image]
Here's the link to the problematic files: [https://drive.google.com/drive/folders/18Q5P4bpvKaWBLiZ2H2uMvJLHzfBblGNw?usp=sharing](url)

I've also attached a txt file for the chat log and my settings for the GUI I used. let me know if you need anything else!

CameronBodine, 2024-06-07:
The locate and flag missing pings workflow has been encountering more and more issues. Unchecking that option will allow you to process the recording. Let me know if this fixes the problem and I will close this issue.

Jdubs6, 2024-06-07:
It looks like the errors are gone!! I wanted to ask, it seems that GUI exported the sonar files as "meta" files. Is there a recommended way to view those files?

CameronBodine, 2024-06-07:
Awesome! I will close.

>  I wanted to ask, it seems that GUI exported the sonar files as "meta" files. Is there a recommended way to view those files?
The `.meta` files are for PING-Mapper to properly run. You should have other folders in the project folder with rectified imagery, etc. You can set up a Zoom with me if you want to step through any of the outputs.

closed 2024-06-07 by CameronBodine

Jdubs6, 2024-06-07:
Ah, I gotcha. I figured out how to get the data in jpg form! Thanks so much for all the help man, I really appreciate it!!