CameronBodine/PINGMapper #124: Install Update Issues

Dabbler23, 2024-10-19:
### Describe the bug
Hi. Probably user error more than a bug. I was able to install version 1 fine and the small test data ran fine. However, I pushed my luck and tried to follow the Update Installation instructions and now cannot run the test data.

### To Reproduce
Steps to reproduce the behavior: Please see screen grabs
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior 
Small test data set to be processed

### Screenshots
If applicable, add screenshots to help explain your problem.
[image]
[image]

### Desktop (please complete the following information):
Dell Laptop, 64-bit, Windows 11 Home 

### Additional context
Unfortunately I'm a complete newbie with respect to Miniconda and command prompts. As I mentioned, my initial install seemed to work fine with the test data, so I'd be quite happy just to uninstall everything and start again if that is an easier option?? But I don't know how to uninstall! Thanks very much for your help, and great tool by the way, can't wait to run some of my own sss imagery through it. Cheers

CameronBodine, 2024-10-19:
That's a strange one! Why don't you try re-installing the conda environment, since it seems to be a tensorflow issue. Try the steps below and report back:

1. Make sure `ping` environment is deactivated by running `conda deactivate`.
2. Remove the environment with `conda remove -n ping --all`
3. Clean up unused packages with `conda clean --all`
4. Then reinstall the environment starting at [Step 8](https://cameronbodine.github.io/PINGMapper/docs/gettingstarted/Installation.html#step-8)
Make sure the terminal is in the `PINGMapper` directory (step 7)

Dabbler23, 2024-10-20:
Hey Cameron
Thanks so much for the swift reply.
I tried what you suggested and it worked a treat!
Have just processed my first transect data, collected yesterday, and super
happy. Image attached.
I did get an error message (log recorded) showing the substrate pick had
not worked. But I'll have a bit of a play first as I may have got some of
the inputs wrong (this is literally my first attempt at processing).
For your interest, we're looking at associating fish telemetry data with
the sonagarms. I guess we'll see the tagged fish hanging around areas like
timber and rock bars etc. Time will tell.
Again, thank you very much for your help, and for making this program
available.
Best regards
Sean

Dabbler23, 2025-01-20:
Hi Cameron
I just installed your latest version of PingMapper with the new Pinginstaller/Ping Wizard.
For someone like myself with very little experience in Conda your latest version was great to install.
Unfortunately I'm getting the error below when trying to test the supplied data sets, or my own.
I'd be grateful if you had any pointers on what I could try? I did look on the Tensorflow site and plenty of people have received the same error but to be honest I wasn't really sure what the fix is!
Thanks very much.
Regards
Sean

C:\Users\Blake\Desktop>set conda_base=C:\Users\Blake\miniconda3_2\

C:\Users\Blake\Desktop>set conda_env="ping"

C:\Users\Blake\Desktop>call C:\Users\Blake\miniconda3_2\\Scripts\activate "ping"
C:\Users\Blake\miniconda3_2\Lib\site-packages\conda\base\context.py:201: FutureWarning: Adding 'defaults' to channel list implicitly is deprecated and will be removed in 25.3.

To remove this warning, please choose a default channel explicitly with conda's regular configuration system, e.g. by adding 'defaults' to the list of channels:

  conda config --add channels defaults

For more information see https://docs.conda.io/projects/conda/en/stable/user-guide/configuration/use-condarc.html

  deprecated.topic(

# conda environments:
#
base                   C:\Users\Blake\miniconda3_2
ping                 * C:\Users\Blake\miniconda3_2\envs\ping
                       C:\Users\Blake\miniforge3

Launching PINGWizard
Welcome to the PING Wizard!
Testing PINGMapper...

PINGMapper v4.0.9

Testing PINGMapper on small dataset...

Traceback (most recent call last):
  File "C:\Users\Blake\miniconda3_2\envs\ping\Lib\site-packages\tensorflow\python\pywrap_tensorflow.py", line 70, in <module>
    from tensorflow.python._pywrap_tensorflow_internal import *
ImportError: DLL load failed while importing _pywrap_tensorflow_internal: The specified module could not be found.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\Blake\miniconda3_2\envs\ping\Lib\site-packages\pingwizard\__main__.py", line 18, in <module>
    main()
  File "C:\Users\Blake\miniconda3_2\envs\ping\Lib\site-packages\pingwizard\__main__.py", line 13, in main
    wizard()
  File "C:\Users\Blake\miniconda3_2\envs\ping\Lib\site-packages\pingwizard\gui_wizard.py", line 142, in wizard
    runpy.run_module(module_name, run_name="__main__")
  File "<frozen runpy>", line 229, in run_module
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\Blake\miniconda3_2\envs\ping\Lib\site-packages\pingmapper\__main__.py", line 47, in <module>
    main(to_do)
  File "C:\Users\Blake\miniconda3_2\envs\ping\Lib\site-packages\pingmapper\__main__.py", line 37, in main
    from pingmapper.test_PINGMapper import test
  File "C:\Users\Blake\miniconda3_2\envs\ping\Lib\site-packages\pingmapper\test_PINGMapper.py", line 39, in <module>
    from pingmapper.main_readFiles import read_master_func
  File "C:\Users\Blake\miniconda3_2\envs\ping\Lib\site-packages\pingmapper\main_readFiles.py", line 38, in <module>
    from pingmapper.class_portstarObj import portstarObj
  File "C:\Users\Blake\miniconda3_2\envs\ping\Lib\site-packages\pingmapper\class_portstarObj.py", line 38, in <module>
    from pingmapper.funcs_model import *
  File "C:\Users\Blake\miniconda3_2\envs\ping\Lib\site-packages\pingmapper\funcs_model.py", line 42, in <module>
    import tensorflow as tf
  File "C:\Users\Blake\miniconda3_2\envs\ping\Lib\site-packages\tensorflow\__init__.py", line 40, in <module>
    from tensorflow.python import pywrap_tensorflow as _pywrap_tensorflow  # pylint: disable=unused-import
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Blake\miniconda3_2\envs\ping\Lib\site-packages\tensorflow\python\pywrap_tensorflow.py", line 85, in <module>
    raise ImportError(
ImportError: Traceback (most recent call last):
  File "C:\Users\Blake\miniconda3_2\envs\ping\Lib\site-packages\tensorflow\python\pywrap_tensorflow.py", line 70, in <module>
    from tensorflow.python._pywrap_tensorflow_internal import *
ImportError: DLL load failed while importing _pywrap_tensorflow_internal: The specified module could not be found.

Failed to load the native TensorFlow runtime.
See https://www.tensorflow.org/install/errors for some common causes and solutions.
If you need help, create an issue at https://github.com/tensorflow/tensorflow/issues and include the entire stack trace above this error message.
Press any key to continue . . .

CameronBodine, 2025-01-21:
Thanks for the feedback Sean. I am unable to reproduce your issue on my end.

Could you open your miniconda prompt, run the following, and copy and paste the output here?

```
conda run -n ping conda list
```

For my reference, this is my environment:

```
(base) PS C:\Users\cbodine> conda run -n ping conda list
# packages in environment at C:\Users\cbodine\miniconda3\envs\ping:
#
# Name                    Version                   Build  Channel
_libavif_api              1.1.1                h57928b3_2    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
absl-py                   2.1.0                    pypi_0    pypi
affine                    2.4.0              pyhd8ed1ab_1    conda-forge
aom                       3.9.1                he0c23c2_0    conda-forge
astunparse                1.6.3                    pypi_0    pypi
attrs                     24.3.0             pyh71513ae_0    conda-forge
blosc                     1.21.6               hfd34d9b_1    conda-forge
branca                    0.8.1              pyhd8ed1ab_0    conda-forge
brotli                    1.1.0                h2466b09_2    conda-forge
brotli-bin                1.1.0                h2466b09_2    conda-forge
brotli-python             1.1.0           py312h275cf98_2    conda-forge
bzip2                     1.0.8                h2466b09_7    conda-forge
c-blosc2                  2.15.2               h9e57077_1    conda-forge
ca-certificates           2024.12.14           h56e8100_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.18.2               h5782bbf_1    conda-forge
certifi                   2024.12.14         pyhd8ed1ab_0    conda-forge
cffi                      1.17.1          py312h4389bb4_0    conda-forge
charls                    2.4.2                h1537add_0    conda-forge
charset-normalizer        3.4.1              pyhd8ed1ab_0    conda-forge
click                     8.1.8              pyh7428d3b_0    conda-forge
click-plugins             1.1.1              pyhd8ed1ab_1    conda-forge
cligj                     0.7.2              pyhd8ed1ab_2    conda-forge
colorama                  0.4.6              pyhd8ed1ab_1    conda-forge
contourpy                 1.3.1           py312hd5eb7cc_0    conda-forge
cycler                    0.12.1             pyhd8ed1ab_1    conda-forge
dav1d                     1.2.1                hcfcfb64_0    conda-forge
doodleverse-utils         0.0.39                   pypi_0    pypi
double-conversion         3.3.0                h63175ca_0    conda-forge
filelock                  3.16.1                   pypi_0    pypi
flatbuffers               24.12.23                 pypi_0    pypi
folium                    0.19.4             pyhd8ed1ab_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 h77eed37_3    conda-forge
fontconfig                2.15.0               h765892d_1    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.55.3          py312h31fea79_1    conda-forge
freetype                  2.12.1               hdaf720e_2    conda-forge
freexl                    2.0.0                hf297d47_2    conda-forge
fsspec                    2024.12.0                pypi_0    pypi
gast                      0.6.0                    pypi_0    pypi
gdal                      3.10.1          py312hc39d689_1    conda-forge
geopandas                 1.0.1              pyhd8ed1ab_3    conda-forge
geopandas-base            1.0.1              pyha770c72_3    conda-forge
geos                      3.13.0               h5a68840_0    conda-forge
geotiff                   1.7.3                h496ac4d_3    conda-forge
giflib                    5.2.2                h64bf75a_0    conda-forge
git                       2.47.1               h57928b3_0    conda-forge
google-pasta              0.2.0                    pypi_0    pypi
graphite2                 1.3.13            h63175ca_1003    conda-forge
[... 150 lines cut ...]
tzdata                    2024b                hc8b5060_0    conda-forge
ucrt                      10.0.22621.0         h57928b3_1    conda-forge
unicodedata2              15.1.0          py312h4389bb4_1    conda-forge
uriparser                 0.9.8                h5a68840_0    conda-forge
urllib3                   2.3.0              pyhd8ed1ab_0    conda-forge
vc                        14.3                ha32ba9b_23    conda-forge
vc14_runtime              14.42.34433         he29a5d6_23    conda-forge
versioneer                0.29                     pypi_0    pypi
vs2015_runtime            14.42.34433         hdffcdeb_23    conda-forge
werkzeug                  3.1.3                    pypi_0    pypi
wheel                     0.45.1             pyhd8ed1ab_1    conda-forge
win_inet_pton             1.1.0              pyh7428d3b_8    conda-forge
wrapt                     1.17.0                   pypi_0    pypi
x265                      3.5                  h2d74725_3    conda-forge
[... cut ...]
xyzservices               2024.9.0           pyhd8ed1ab_1    conda-forge
zfp                       1.0.1                he0c23c2_2    conda-forge
zipp                      3.21.0             pyhd8ed1ab_1    conda-forge
zlib                      1.3.1                h2466b09_2    conda-forge
zlib-ng                   2.2.3                hbb528cf_0    conda-forge
zstandard                 0.23.0          py312h7606c53_1    conda-forge
zstd                      1.5.6                h0ea2cb4_0    conda-forge
```

Dabbler23, 2025-01-22:
Hi Cameron
Thanks very much for the swift reply.
My output is below.
Thanks again!

(base) PS C:\Users\Blake> conda run -n ping conda list
# packages in environment at C:\Users\Blake\miniconda3_2\envs\ping:
#
# Name                    Version                   Build  Channel
_libavif_api              1.1.1                h57928b3_2    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
absl-py                   2.1.0                    pypi_0    pypi
affine                    2.4.0              pyhd8ed1ab_0    conda-forge
aom                       3.9.1                he0c23c2_0    conda-forge
astunparse                1.6.3                    pypi_0    pypi
attrs                     24.2.0             pyh71513ae_0    conda-forge
blosc                     1.21.6               hfd34d9b_1    conda-forge
branca                    0.7.2              pyhd8ed1ab_0    conda-forge
brotli                    1.1.0                h2466b09_2    conda-forge
brotli-bin                1.1.0                h2466b09_2    conda-forge
brotli-python             1.1.0           py312h275cf98_2    conda-forge
bzip2                     1.0.8                h2466b09_7    conda-forge
c-blosc2                  2.15.2               h9e57077_1    conda-forge
ca-certificates           2024.12.14           h56e8100_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.18.0               h32b962e_3    conda-forge
certifi                   2024.12.14         pyhd8ed1ab_0    conda-forge
cffi                      1.17.1          py312h4389bb4_0    conda-forge
charls                    2.4.2                h1537add_0    conda-forge
charset-normalizer        3.4.0              pyhd8ed1ab_0    conda-forge
click                     8.1.7           win_pyh7428d3b_0    conda-forge
click-plugins             1.1.1                      py_0    conda-forge
cligj                     0.7.2              pyhd8ed1ab_1    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
contourpy                 1.3.1           py312hd5eb7cc_0    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
dav1d                     1.2.1                hcfcfb64_0    conda-forge
doodleverse-utils         0.0.39                   pypi_0    pypi
double-conversion         3.3.0                h63175ca_0    conda-forge
filelock                  3.16.1                   pypi_0    pypi
flatbuffers               24.3.25                  pypi_0    pypi
folium                    0.18.0             pyhd8ed1ab_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 h77eed37_3    conda-forge
fontconfig                2.15.0               h765892d_1    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.55.3          py312h31fea79_1    conda-forge
freetype                  2.12.1               hdaf720e_2    conda-forge
freexl                    2.0.0                h8276f4a_0    conda-forge
fsspec                    2024.10.0                pypi_0    pypi
gast                      0.6.0                    pypi_0    pypi
gdal                      3.10.1          py312hc39d689_1    conda-forge
geopandas                 1.0.1              pyhd8ed1ab_3    conda-forge
geopandas-base            1.0.1              pyha770c72_3    conda-forge
geos                      3.13.0               h5a68840_0    conda-forge
geotiff                   1.7.3                h496ac4d_3    conda-forge
giflib                    5.2.2                h64bf75a_0    conda-forge
git                       2.47.1               h57928b3_0    conda-forge
google-pasta              0.2.0                    pypi_0    pypi
graphite2                 1.3.13            h63175ca_1003    conda-forge
grpcio                    1.67.1                   pypi_0    pypi
h2                        4.1.0              pyhd8ed1ab_0    conda-forge
h5py                      3.12.1          nompi_py312h0db4ba1_103    conda-forge
harfbuzz                  9.0.0                h2bedf89_1    conda-forge
hdf5                      1.14.4          nompi_hd5d9e70_105    conda-forge
hpack                     4.0.0              pyh9f0ad1d_0    conda-forge
[... 146 lines cut ...]
unicodedata2              16.0.0          py312h4389bb4_0    conda-forge
uriparser                 0.9.8                h5a68840_0    conda-forge
urllib3                   2.2.3              pyhd8ed1ab_0    conda-forge
vc                        14.3                h8a93ad2_22    conda-forge
vc14_runtime              14.40.33810         hcc2c482_22    conda-forge
versioneer                0.29                     pypi_0    pypi
vs2015_runtime            14.40.33810         h3bf8584_22    conda-forge
werkzeug                  3.1.2                    pypi_0    pypi
wheel                     0.44.0             pyhd8ed1ab_0    conda-forge
win_inet_pton             1.1.0              pyh7428d3b_7    conda-forge
wrapt                     1.16.0                   pypi_0    pypi
[... cut ...]
xz                        5.2.6                h8d14728_0    conda-forge
zfp                       1.0.1                he0c23c2_2    conda-forge
zipp                      3.20.2             pyhd8ed1ab_0    conda-forge
zlib                      1.3.1                h2466b09_2    conda-forge
zlib-ng                   2.2.3                hbb528cf_0    conda-forge
zstandard                 0.23.0          py312h7606c53_1    conda-forge
zstd                      1.5.6                h0ea2cb4_0    conda-forge

(base) PS C:\Users\Blake>

Dabbler23, 2025-01-22:
Sorry Cameron, the formatting in the last message was horrible so I've just attached the .txt file in this one - hopefully a bit easier to read.
Thanks again
Sean

[PingMapper v4.0.9 Install Issue_22.01.2025.txt](https://github.com/user-attachments/files/18499461/PingMapper.v4.0.9.Install.Issue_22.01.2025.txt)

CameronBodine, 2025-01-22:
Ok thanks. Doesn't seem like a tensorflow version from what I can tell. 

I noticed your miniconda3 is in a folder called miniconda3_2. Did you install a second miniconda? There may be an issue if you have both minconda3 and miniconda3_2 folders. Something to look into. 

Why don't you try updating the installation from the wizard, and try the test again. Provide all error output as before. 

Also, does the problem persist if you run pingmapper on your own sonar logs?

Dabbler23, 2025-01-25:
Hey Cameron
Sorry for slow reply, was in the field for a couple of days.
Ok, so I tried uninstalling everything and started again. I installed
Anaconda instead of Miniconda this time to see if made any difference
(didn't)..
Things seemed to install ok and I can start up the wizard
(PingMapper_Install_24012025.txt). There were no updates available.
I received an error message when I tried to process your small dataset (log
= PingMapper_Small_data_Error_24012025.txt).
I then tried processing one of my own sonargrams. It ran for about 30
minutes and then crashed (log = PingMapper_User_Data_Error_24012025.txt,
error log = error_2025-01-24_0808.txt).
Thanks very much for your patience, I appreciate this is taking up your
time!
Regards
Sean

(base) PS C:\Users\Blake> pip install pinginstaller
Collecting pinginstaller
  Downloading pinginstaller-1.0.13-py3-none-any.whl.metadata (3.8 kB)
Downloading pinginstaller-1.0.13-py3-none-any.whl (7.0 kB)
Installing collected packages: pinginstaller
Successfully installed pinginstaller-1.0.13
(base) PS C:\Users\Blake> python -m pinginstaller
Env yml: https://github.com/CameronBodine/PINGMapper/blob/main/pingmapper/conda/PINGMapper.yml
# conda environments:
#
base                  *  C:\Users\Blake\anaconda3

conda_key: C:\Users\Blake\anaconda3\Scripts\conda.exe
Channels:
 - defaults
Platform: win-64
Collecting package metadata (repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: C:\Users\Blake\anaconda3

  added / updated specs:
    - conda

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    ca-certificates-2024.12.31 |       haa95532_0         129 KB
    certifi-2024.12.14         |  py312haa95532_0         162 KB
    conda-24.11.3              |  py312haa95532_0         1.2 MB
    ------------------------------------------------------------
                                           Total:         1.5 MB

The following packages will be UPDATED:

  ca-certificates                      2024.9.24-haa95532_0 --> 2024.12.31-haa95532_0
  certifi                         2024.8.30-py312haa95532_0 --> 2024.12.14-py312haa95532_0
  conda                              24.9.2-py312haa95532_0 --> 24.11.3-py312haa95532_0

Downloading and Extracting Packages:

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
Will remove 489 (924.3 MB) tarball(s).
Will remove 1 index cache(s).
Will remove 14 (6.3 MB) package(s).
There are no tempfile(s) to remove.
There are no logfile(s) to remove.
Requirement already satisfied: pip in c:\users\blake\anaconda3\lib\site-packages (24.2)
[... 738 lines cut ...]
________________________
CPU % | RAM % | RAM [GB]
________________________
8.9   | 78.8  | 6.6
________________________

===========================================
===========================================
***** RECTIFYING *****

!!!!!!!!!!!!!!!!!
An Error Occured.

Please consult the logfile at:
 C:\Users\Blake\Documents\Work\PINGMapper\PingMapper_Outputs\Coldstream_Test\logs\error_2025-01-24_0808.txt

Could not process: C:/Users/Blake/Documents/Work/PINGMapper/Test_Coldstream_19.10.2024/Rec00003.DAT

Total Processing Time:  0:28:48

Error Thrown:
Traceback (most recent call last):
  File "C:\Users\Blake\anaconda3\envs\ping\Lib\site-packages\pingmapper\gui_main.py", line 706, in gui
    rectify_master_func(**params)
TypeError: rectify_master_func() got an unexpected keyword argument 'wcm'

CameronBodine, 2025-01-26:
I just added an update that should fix the test script. b5e387850f9a2dde1cb4700ae8ec36d0444c654a

Dabbler23, 2025-01-27:
Hi Cameron
Great news, your update has fixed my issues!
I used the PingInstaller (so much easier for the likes of me) to update
your edits. I then ran the small test data successfully, followed by my own
test data.
And finally, I processed my main data. This is the first time I've seen it
since collecting, and I'm super happy with it.
Once again, thanks so much for your patience and help. Really looking
forward to seeing how the PingMapper outputs will mesh with the telemetry
data.
It's so cool that we now have the ability to process sonargrams from
recreational class sounders. Great job!!
Cheers
Sean

CameronBodine, 2025-01-27:
Hi Sean,
I am so glad to hear this! Keep me posted on your developments. Let me know if you have any issues or questions. 

Closing as complete.

closed 2025-01-27 by CameronBodine

Dabbler23, 2026-01-15:
Hi Cameron
I last spoke to you back in Jan 25, when you managed to get me up and
running with PINGMapper.
Fast forward a year and I haven't done anything further with my data!
This morning I dusted off PINGMapper and it recommended an update, which I
did.
On trying to run the supplied small test data though I got an error - error
log attached (error_2026-01-15_1006). It appears to have processed the SSS
mosaic but not produced a habitat map?
I then also tried to process one of my own .DAT files, the one I
successfully processed a year ago. It also cashed - log attached
(error_2026-01-15_1138).
I'd be most grateful if you could once again point me in the right
direction!
Thank you once again for your support.
Best Regards
Sean

Error Thrown:
joblib.externals.loky.process_executor._RemoteTraceback: 
"""
Traceback (most recent call last):
  File "C:\Users\Blake\anaconda3\envs\ping\Lib\site-packages\urllib3\connectionpool.py", line 464, in _make_request
    self._validate_conn(conn)
  File "C:\Users\Blake\anaconda3\envs\ping\Lib\site-packages\urllib3\connectionpool.py", line 1093, in _validate_conn
    conn.connect()
  File "C:\Users\Blake\anaconda3\envs\ping\Lib\site-packages\urllib3\connection.py", line 796, in connect
    sock_and_verified = _ssl_wrap_socket_and_match_hostname(
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Blake\anaconda3\envs\ping\Lib\site-packages\urllib3\connection.py", line 975, in _ssl_wrap_socket_and_match_hostname
    ssl_sock = ssl_wrap_socket(
               ^^^^^^^^^^^^^^^^
  File "C:\Users\Blake\anaconda3\envs\ping\Lib\site-packages\urllib3\util\ssl_.py", line 483, in ssl_wrap_socket
    ssl_sock = _ssl_wrap_socket_impl(sock, context, tls_in_tls, server_hostname)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Blake\anaconda3\envs\ping\Lib\site-packages\urllib3\util\ssl_.py", line 527, in _ssl_wrap_socket_impl
    return ssl_context.wrap_socket(sock, server_hostname=server_hostname)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Blake\anaconda3\envs\ping\Lib\ssl.py", line 455, in wrap_socket
    return self.sslsocket_class._create(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Blake\anaconda3\envs\ping\Lib\ssl.py", line 1041, in _create
    self.do_handshake()
  File "C:\Users\Blake\anaconda3\envs\ping\Lib\ssl.py", line 1319, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self-signed certificate in certificate chain (_ssl.c:1000)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Blake\anaconda3\envs\ping\Lib\site-packages\urllib3\connectionpool.py", line 787, in urlopen
    response = self._make_request(
               ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Blake\anaconda3\envs\ping\Lib\site-packages\urllib3\connectionpool.py", line 488, in _make_request
    raise new_e
urllib3.exceptions.SSLError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self-signed certificate in certificate chain (_ssl.c:1000)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\Blake\anaconda3\envs\ping\Lib\site-packages\requests\adapters.py", line 644, in send
    resp = conn.urlopen(
           ^^^^^^^^^^^^^
  File "C:\Users\Blake\anaconda3\envs\ping\Lib\site-packages\urllib3\connectionpool.py", line 841, in urlopen
    retries = retries.increment(
              ^^^^^^^^^^^^^^^^^^
  File "C:\Users\Blake\anaconda3\envs\ping\Lib\site-packages\urllib3\util\retry.py", line 535, in increment
    raise MaxRetryError(_pool, url, reason) from reason  # type: ignore[arg-type]
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='huggingface.co', port=443): Max retries exceeded with url: /nvidia/mit-b0/resolve/main/config.json (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self-signed certificate in certificate chain (_ssl.c:1000)')))
[... 124 lines cut ...]
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\Blake\anaconda3\envs\ping\Lib\site-packages\pingmapper\gui_main.py", line 746, in gui
    read_master_func(**params)
  File "C:\Users\Blake\anaconda3\envs\ping\Lib\site-packages\pingmapper\main_readFiles.py", line 1494, in read_master_func
    r = Parallel(n_jobs=np.min([len(chunks), threadCnt]))(delayed(psObj._detectShadow)(remShadow, int(chunk), USE_GPU, False, tileFile) for chunk in tqdm(range(len(chunks))))
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Blake\anaconda3\envs\ping\Lib\site-packages\joblib\parallel.py", line 2072, in __call__
    return output if self.return_generator else list(output)
                                                ^^^^^^^^^^^^
  File "C:\Users\Blake\anaconda3\envs\ping\Lib\site-packages\joblib\parallel.py", line 1682, in _get_outputs
    yield from self._retrieve()
  File "C:\Users\Blake\anaconda3\envs\ping\Lib\site-packages\joblib\parallel.py", line 1784, in _retrieve
    self._raise_error_fast()
[... cut ...] n _raise_error_fast
    error_job.get_result(self.timeout)
  File "C:\Users\Blake\anaconda3\envs\ping\Lib\site-packages\joblib\parallel.py", line 758, in get_result
    return self._return_or_raise()
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Blake\anaconda3\envs\ping\Lib\site-packages\joblib\parallel.py", line 773, in _return_or_raise
    raise self._result
numpy._core._exceptions._ArrayMemoryError: Unable to allocate 2.00 MiB for an array with shape (512, 512, 2) and data type float32

CameronBodine, 2026-01-15:
Hi! Please try to update again. It may not have processed correctly. If
that does not fix the issue, please make a new issue and attach all logs
including errors. This issue has been marked as complete.

- Cam