CameronBodine/PINGMapper #128: Error during Small Dataset Test

kbartelt1, 2024-12-03:
I followed the installation instructions without any issues, but then encountered a few warnings and errors when running the Small Dataset Test. 

First, I get the following warning after the files are downloaded and read. 

```
*******************************
****WARNING: INVALID VALUES****
*******************************
_______________________________
Sonar Channel   | Attribute
_______________________________
B000            | transect
B001            | transect
B002            | transect
B003            | transect

*******************************
****WARNING: INVALID VALUES****
*******************************

PING-Mapper detected issues with
the values stored in the above
sonar channels and attributes.
```

The program then successfully exports the bedpick plots to .jpg, but then I get several ValueErrors during the shadow detecting step.

```
Automatically detecting shadows for 7 chunks:
MODE: 1 | Remove all shadows...
[Parallel(n_jobs=7)]: Using backend LokyBackend with 7 concurrent workers.
joblib.externals.loky.process_executor._RemoteTraceback:
"""
Traceback (most recent call last):
  File "C:\Users\kbartelt\AppData\Local\miniconda3\envs\ping\Lib\site-packages\joblib\externals\loky\process_executor.py", line 463, in _process_worker
    r = call_item()
        ^^^^^^^^^^^
  File "C:\Users\kbartelt\AppData\Local\miniconda3\envs\ping\Lib\site-packages\joblib\externals\loky\process_executor.py", line 291, in __call__
    return self.fn(*self.args, **self.kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\kbartelt\AppData\Local\miniconda3\envs\ping\Lib\site-packages\joblib\parallel.py", line 598, in __call__
    return [func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\kbartelt\AppData\Local\miniconda3\envs\ping\Lib\site-packages\joblib\parallel.py", line 598, in <listcomp>
    return [func(*args, **kwargs)
            ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\kbartelt\MyPythonRepos\PINGMapper\src\class_portstarObj.py", line 1827, in _detectShadow
    port_label, port_prob = doPredict(model, MODEL, self.port.sonDat, N_DATA_BANDS, NCLASSES, TARGET_SIZE, OTSU_THRESHOLD)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\kbartelt\MyPythonRepos\PINGMapper\src\funcs_model.py", line 157, in doPredict
    E0, E1 = est_label_binary(image, model, MODEL, False, NCLASSES, TARGET_SIZE, w, h)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\kbartelt\AppData\Local\miniconda3\envs\ping\Lib\site-packages\doodleverse_utils\prediction_imports.py", line 237, in est_label_binary
    est_label = tf.squeeze(model.predict(tf.expand_dims(image[:,:,:3], 0), batch_size=1))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\kbartelt\AppData\Local\miniconda3\envs\ping\Lib\site-packages\tf_keras\src\utils\traceback_utils.py", line 70, in error_handler
    raise e.with_traceback(filtered_tb) from None
  File "C:\Users\kbartelt\AppData\Local\Temp\1\__autograph_generated_file0_u59u53.py", line 18, in tf__predict_function
    raise
ValueError: in user code:

    File "C:\Users\kbartelt\AppData\Local\miniconda3\envs\ping\Lib\site-packages\tf_keras\src\engine\training.py", line 2436, in predict_function  *
        return step_function(self, iterator)
    File "C:\Users\kbartelt\AppData\Local\miniconda3\envs\ping\Lib\site-packages\tf_keras\src\engine\training.py", line 2421, in step_function  **
        outputs = model.distribute_strategy.run(run_step, args=(data,))
    File "C:\Users\kbartelt\AppData\Local\miniconda3\envs\ping\Lib\site-packages\tf_keras\src\engine\training.py", line 2409, in run_step  **
        outputs = model.predict_step(data)
    File "C:\Users\kbartelt\AppData\Local\miniconda3\envs\ping\Lib\site-packages\tf_keras\src\engine\training.py", line 2377, in predict_step
        return self(x, training=False)
    File "C:\Users\kbartelt\AppData\Local\miniconda3\envs\ping\Lib\site-packages\tf_keras\src\utils\traceback_utils.py", line 70, in error_handler
        raise e.with_traceback(filtered_tb) from None
    File "C:\Users\kbartelt\AppData\Local\miniconda3\envs\ping\Lib\site-packages\tf_keras\src\engine\input_spec.py", line 280, in assert_input_compatibility
        raise ValueError(

    ValueError: Exception encountered when calling layer 'model' (type Functional).

    Input 0 of layer "conv2d" is incompatible with the layer: expected axis -1 of input shape to have value 1, but received input with shape (1, 512, 512, 3)

    Call arguments received by layer 'model' (type Functional):
      • inputs=tf.Tensor(shape=(1, 512, 512, 3), dtype=float64)
      • training=False
      • mask=None

"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\kbartelt\MyPythonRepos\PINGMapper\test_PINGMapper.py", line 297, in <module>
    read_master_func(**params)
  File "C:\Users\kbartelt\MyPythonRepos\PINGMapper\src\main_readFiles.py", line 1258, in read_master_func
    r = Parallel(n_jobs=np.min([len(chunks), threadCnt]), verbose=10)(delayed(psObj._detectShadow)(remShadow, int(chunk), USE_GPU, False, tileFile) for chunk in chunks)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\kbartelt\AppData\Local\miniconda3\envs\ping\Lib\site-packages\joblib\parallel.py", line 2007, in __call__
    return output if self.return_generator else list(output)
                                                ^^^^^^^^^^^^
  File "C:\Users\kbartelt\AppData\Local\miniconda3\envs\ping\Lib\site-packages\joblib\parallel.py", line 1650, in _get_outputs
    yield from self._retrieve()
  File "C:\Users\kbartelt\AppData\Local\miniconda3\envs\ping\Lib\site-packages\joblib\parallel.py", line 1754, in _retrieve
    self._raise_error_fast()
  File "C:\Users\kbartelt\AppData\Local\miniconda3\envs\ping\Lib\site-packages\joblib\parallel.py", line 1789, in _raise_error_fast
    error_job.get_result(self.timeout)
  File "C:\Users\kbartelt\AppData\Local\miniconda3\envs\ping\Lib\site-packages\joblib\parallel.py", line 745, in get_result
    return self._return_or_raise()
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\kbartelt\AppData\Local\miniconda3\envs\ping\Lib\site-packages\joblib\parallel.py", line 763, in _return_or_raise
    raise self._result
ValueError: in user code:

    File "C:\Users\kbartelt\AppData\Local\miniconda3\envs\ping\Lib\site-packages\tf_keras\src\engine\training.py", line 2436, in predict_function  *
        return step_function(self, iterator)
    File "C:\Users\kbartelt\AppData\Local\miniconda3\envs\ping\Lib\site-packages\tf_keras\src\engine\training.py", line 2421, in step_function  **
        outputs = model.distribute_strategy.run(run_step, args=(data,))
    File "C:\Users\kbartelt\AppData\Local\miniconda3\envs\ping\Lib\site-packages\tf_keras\src\engine\training.py", line 2409, in run_step  **
        outputs = model.predict_step(data)
    File "C:\Users\kbartelt\AppData\Local\miniconda3\envs\ping\Lib\site-packages\tf_keras\src\engine\training.py", line 2377, in predict_step
        return self(x, training=False)
    File "C:\Users\kbartelt\AppData\Local\miniconda3\envs\ping\Lib\site-packages\tf_keras\src\utils\traceback_utils.py", line 70, in error_handler
        raise e.with_traceback(filtered_tb) from None
    File "C:\Users\kbartelt\AppData\Local\miniconda3\envs\ping\Lib\site-packages\tf_keras\src\engine\input_spec.py", line 280, in assert_input_compatibility
        raise ValueError(

    ValueError: Exception encountered when calling layer 'model' (type Functional).

    Input 0 of layer "conv2d" is incompatible with the layer: expected axis -1 of input shape to have value 1, but received input with shape (1, 512, 512, 3)

    Call arguments received by layer 'model' (type Functional):
      • inputs=tf.Tensor(shape=(1, 512, 512, 3), dtype=float64)
      • training=False
      • mask=None
```

### Desktop (please complete the following information):
 - OS: Windows 11 Enterprise

### Log file

[log_2024-12-03_1116.txt](https://github.com/user-attachments/files/17997923/log_2024-12-03_1116.txt)

Thank you for your help!

CameronBodine, 2024-12-03:
Thanks for the information! I will take a look tomorrow. You could try running the test a second time. Sometimes it works the second time through.

kbartelt1, 2024-12-03:
Thank you! I did try running this again, but I am getting the same error.

CameronBodine, 2024-12-04:
# For documentation purposes
I was able to reproduce the error after reinstalling the `ping` environment.

**My `ping` environment before:**
```
(base) PS C:\Users\cbodine> conda activate ping
(ping) PS C:\Users\cbodine> conda list
# packages in environment at C:\Users\cbodine\miniconda3\envs\ping:
#
# Name                    Version                   Build  Channel
_libavif_api              1.1.1                h57928b3_2    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
absl-py                   2.1.0                    pypi_0    pypi
affine                    2.4.0              pyhd8ed1ab_0    conda-forge
aom                       3.9.1                he0c23c2_0    conda-forge
astunparse                1.6.3                    pypi_0    pypi
attrs                     24.2.0             pyh71513ae_0    conda-forge
blosc                     1.21.6               h85f69ea_0    conda-forge
branca                    0.7.2              pyhd8ed1ab_0    conda-forge
brotli                    1.1.0                h2466b09_2    conda-forge
brotli-bin                1.1.0                h2466b09_2    conda-forge
brotli-python             1.1.0           py311hda3d55a_2    conda-forge
bzip2                     1.0.8                h2466b09_7    conda-forge
c-blosc2                  2.15.1               hb461149_0    conda-forge
ca-certificates           2024.8.30            h56e8100_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.18.0               h32b962e_3    conda-forge
certifi                   2024.8.30          pyhd8ed1ab_0    conda-forge
cffi                      1.17.1          py311he736701_0    conda-forge
charls                    2.4.2                h1537add_0    conda-forge
charset-normalizer        3.4.0              pyhd8ed1ab_0    conda-forge
click                     8.1.7           win_pyh7428d3b_0    conda-forge
click-plugins             1.1.1                      py_0    conda-forge
cligj                     0.7.2              pyhd8ed1ab_1    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
contourpy                 1.3.0           py311h3257749_2    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
dav1d                     1.2.1                hcfcfb64_0    conda-forge
doodleverse-utils         0.0.35                   pypi_0    pypi
double-conversion         3.3.0                h63175ca_0    conda-forge
filelock                  3.16.1                   pypi_0    pypi
flatbuffers               24.3.25                  pypi_0    pypi
folium                    0.18.0             pyhd8ed1ab_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 h77eed37_3    conda-forge
fontconfig                2.15.0               h765892d_1    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.54.1          py311h5082efb_1    conda-forge
freetype                  2.12.1               hdaf720e_2    conda-forge
freexl                    2.0.0                h8276f4a_0    conda-forge
fsspec                    2024.10.0                pypi_0    pypi
gast                      0.6.0                    pypi_0    pypi
gdal                      3.9.3           py311h689aae6_2    conda-forge
geopandas                 1.0.1              pyhd8ed1ab_1    conda-forge
geopandas-base            1.0.1              pyha770c72_1    conda-forge
geos                      3.13.0               h5a68840_0    conda-forge
geotiff                   1.7.3                h496ac4d_3    conda-forge
giflib                    5.2.2                h64bf75a_0    conda-forge
git                       2.47.0               h57928b3_0    conda-forge
google-pasta              0.2.0                    pypi_0    pypi
graphite2                 1.3.13            h63175ca_1003    conda-forge
grpcio                    1.67.1                   pypi_0    pypi
h2                        4.1.0              pyhd8ed1ab_0    conda-forge
h5py                      3.12.1          nompi_py311h67016bb_102    conda-forge
harfbuzz                  9.0.0                h2bedf89_1    conda-forge
hdf5                      1.14.3          nompi_h2b43c12_105    conda-forge
[... 394 lines cut ...]
zlib                      1.3.1                h2466b09_2    conda-forge
zlib-ng                   2.2.2                he0c23c2_0    conda-forge
zstandard                 0.23.0          py311h53056dc_1    conda-forge
zstd                      1.5.6                h0ea2cb4_0    conda-forge
```

The only difference in tensorflow related packages is:
```
doodleverse-utils    0.0.35 --> 0.0.39
huggingface-hub:	0.26.2 --> 0.26.3
```

Looking at `doodleverse-utils`, there was a change in `def est_label_multiclass()`                 

```
est_label = model(tf.expand_dims(image[:,:,0], 0)).logits
```
changed to:
```
est_label = model(tf.expand_dims(image[:,:,:3], 0)).logits
```

Looking at the model configuration file for the shadow model, I trained with `NCLASSES=2` when it should have been `NCLASSES=1`. This error does not appear during the depth predictions because I correctly specified `NCLASSES=1` during model training.

I added a kludge to my workflow to get the `tf.shape` in the right size and not cause further errors.

CameronBodine, 2024-12-04:
@kbartelt1 I think I have fixed the error. Please run `git pull` in `C:\\Users\\kbartelt\\MyPythonRepos\\PINGMapper` and try running the test again. Please let me know if this fixes the issue.

CameronBodine, 2024-12-04:
Note you may need to run `git stash` prior to `git pull`.

CameronBodine, 2024-12-04:
> First, I get the following warning after the files are downloaded and read.
```
*******************************
****WARNING: INVALID VALUES****
*******************************
_______________________________
Sonar Channel   | Attribute
_______________________________
B000            | transect
B001            | transect
B002            | transect
B003            | transect

*******************************
****WARNING: INVALID VALUES****
*******************************

PING-Mapper detected issues with
the values stored in the above
sonar channels and attributes.
```

Thanks for reporting. This is a bug in the code but poses no issues during processing.

kbartelt1, 2024-12-04:
Thank you for the quick help! That worked.

CameronBodine, 2024-12-04:
Excellent! Let me know if you have any other issues.

closed 2024-12-04 by CameronBodine