CameronBodine/PINGMapper #133: Stray Lines Generated at Corner of Mosaic Images

Hesslertm, 2024-12-13:
### Describe the bug
A clear and concise description of what the bug is.

Hey Cameron, v3.0 has been great! Thanks for some QOL updates. I am still having issues from time to time however with stray lines that appear at a corner of a rectified chunk (see attached image). They will appear at the bottom of the image if the chunk is tilted to the left and not the right like in the example. I thought the max crop feature might fix the issue (which it appears to have for one of my other surveys), but the problem is present on almost every sonar chunk in this survey. Thanks again for the help. 

### To Reproduce
Steps to reproduce the behavior:

[log_2024-12-13_1029.txt](https://github.com/user-attachments/files/18129597/log_2024-12-13_1029.txt)

### Expected behavior 

A clear and concise description of what you expected to happen.

Lines shouldn't be appearing here.

### Screenshots
If applicable, add screenshots to help explain your problem.
[image]

### Desktop (please complete the following information):
 - OS: Windows 11

### Additional context
Add any other context about the problem here.

CameronBodine, 2024-12-14:
Hi Tyler, glad you found the update. I remember this being a problem for you. One thought is that the fix missing data option may be the culprit. Can you try running with that set to False? If that doesn't work, go ahead and send me the problem recording and I will take a look.

Hesslertm, 2024-12-16:
Hey Cameron, even after setting that option to false I am still getting the lines. Can I use the same email I have before to send you the recording? Also attached is another problem image where the line generated is not even close to touching the chunk but is just appearing at the top of the image. 

[image]

CameronBodine, 2024-12-16:
Yes I have access to that email. This is bizarre. I have not seen this from any other recording. I will take a look.

CameronBodine, 2024-12-16:
Can you also share the output after running conda list?

Hesslertm, 2024-12-16:
> WARNING: This message has originated from an External Source. This may be a phishing expedition that can result in unauthorized access to our IT System. Please use proper judgment and caution when opening attachments, clicking links, or responding to this email.

Attached is the recording and most recent log file. Conda list is also attached. Thanks again for the help!

Tyler Hessler

['C:/Users/tmh8nf/OneDrive - University of Missouri/Documents/MRHM/Data/Lamine_Harriman/121024/Sidescan/R00079\\B001.SON', 'C:/Users/tmh8nf/OneDrive - University of Missouri/Documents/MRHM/Data/Lamine_Harriman/121024/Sidescan/R00079\\B002.SON', 'C:/Users/tmh8nf/OneDrive - University of Missouri/Documents/MRHM/Data/Lamine_Harriman/121024/Sidescan/R00079\\B003.SON', 'C:/Users/tmh8nf/OneDrive - University of Missouri/Documents/MRHM/Data/Lamine_Harriman/121024/Sidescan/R00079\\B004.SON']

 ***User Parameters***
| inFile               : C:/Users/tmh8nf/OneDrive - University of Missouri/Documents/MRHM/Data/Lamine_Harriman/121024/Sidescan/R00079.DAT |
| projDir              : C:/Users/tmh8nf/OneDrive - University of Missouri/Documents/MRHM/Data/Lamine_Harriman/121024/Sidescan\Lamine_Harriman_121024_EGN_Shadow |
| project_mode         : 0          |
| tempC                : 3.8        |
| nchunk               : 500        |
| cropRange            : 0.0        |
| exportUnknown        : False      |
| fixNoDat             : True       |
| threadCnt            : 0          |
| aoi                  : False      |
| max_heading_deviation : 0.0        |
| max_heading_distance : 0.0        |
| min_speed            : 0.0        |
| max_speed            : 0.0        |
| pix_res_son          : 0.0        |
| pix_res_map          : 0.0        |
| x_offset             : 3.0        |
| y_offset             : -0.61      |
| egn                  : True       |
| egn_stretch          : 2          |
| egn_stretch_factor   : 0.5        |
| wcp                  : False      |
| wcr                  : True       |
| tileFile             : .jpg       |
| lbl_set              : 2          |
| spdCor               : 1.0        |
| maxCrop              : True       |
| remShadow            : 2          |
| detectDep            : 1          |
| smthDep              : False      |
| adjDep               : 0.0        |
| pltBedPick           : False      |
| rect_wcp             : False      |
| rect_wcr             : True       |
| son_colorMap         : Greys_r    |
| pred_sub             : False      |
| pltSubClass          : False      |
| map_sub              : False      |
| export_poly          : False      |
| map_class_method     : max        |
| map_predict          : 0          |
| mosaic_nchunk        : 0          |
| mosaic               : 1          |
| map_mosaic           : 0          |
| banklines            : True       |
| coverage             : True       |

===========================================
===========================================
***** READING *****
working on C:/Users/tmh8nf/OneDrive - University of Missouri/Documents/MRHM/Data/Lamine_Harriman/121024/Sidescan\Lamine_Harriman_121024_EGN_Shadow

Getting DAT Metadata...
C:/Users/tmh8nf/OneDrive - University of Missouri/Documents/MRHM/Data/Lamine_Harriman/121024/Sidescan/R00079.DAT

Done!
Time (s): 0.0

Getting Header Structure...
Header Length: 72

Done!
Time (s): 2.7
[... 522 lines cut ...]
typing-extensions         4.12.2                   pypi_0    pypi
tzdata                    2024b                hc8b5060_0    conda-forge
ucrt                      10.0.22621.0         h57928b3_1    conda-forge
unicodedata2              15.1.0          py312h4389bb4_1    conda-forge
uriparser                 0.9.8                h5a68840_0    conda-forge
urllib3                   2.2.3              pyhd8ed1ab_1    conda-forge
vc                        14.3                ha32ba9b_23    conda-forge
vc14_runtime              14.42.34433         he29a5d6_23    conda-forge
versioneer                0.29                     pypi_0    pypi
vs2015_runtime            14.42.34433         hdffcdeb_23    conda-forge
werkzeug                  3.1.3                    pypi_0    pypi
wheel                     0.45.1             pyhd8ed1ab_1    conda-forge
win_inet_pton             1.1.0              pyh7428d3b_8    conda-forge
wrapt                     1.17.0                   pypi_0    pypi
x265                      3.5                  h2d74725_3    conda-forge
xerces-c                  3.2.5                he0c23c2_2    conda-forge
xorg-libxau               1.0.11               h0e40799_1    conda-forge
xorg-libxdmcp             1.1.5                h0e40799_0    conda-forge
xyzservices               2024.9.0           pyhd8ed1ab_1    conda-forge
zfp                       1.0.1                he0c23c2_2    conda-forge
zipp                      3.21.0             pyhd8ed1ab_1    conda-forge
zlib                      1.3.1                h2466b09_2    conda-forge
zlib-ng                   2.2.2                he0c23c2_0    conda-forge
zstandard                 0.23.0          py312h7606c53_1    conda-forge
zstd                      1.5.6                h0ea2cb4_0    conda-forge

CameronBodine, 2024-12-16:
Below is for my documentation. 

User's conda environment:

<details>
<summary>Click me</summary>

```
(ping) C:\Users\tmh8nf\PythonPackages\PINGMapper>conda list
# packages in environment at C:\Users\tmh8nf\AppData\Local\anaconda3\envs\ping:
#
# Name                    Version                   Build  Channel
_libavif_api              1.1.1                h57928b3_2    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
absl-py                   2.1.0                    pypi_0    pypi
affine                    2.4.0              pyhd8ed1ab_1    conda-forge
aom                       3.9.1                he0c23c2_0    conda-forge
astunparse                1.6.3                    pypi_0    pypi
attrs                     24.2.0             pyh71513ae_1    conda-forge
blosc                     1.21.6               hfd34d9b_1    conda-forge
branca                    0.7.2              pyhd8ed1ab_1    conda-forge
brotli                    1.1.0                h2466b09_2    conda-forge
brotli-bin                1.1.0                h2466b09_2    conda-forge
brotli-python             1.1.0           py312h275cf98_2    conda-forge
bzip2                     1.0.8                h2466b09_7    conda-forge
c-blosc2                  2.15.2               h9e57077_1    conda-forge
ca-certificates           2024.8.30            h56e8100_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.18.2               h5782bbf_1    conda-forge
certifi                   2024.8.30          pyhd8ed1ab_0    conda-forge
cffi                      1.17.1          py312h4389bb4_0    conda-forge
charls                    2.4.2                h1537add_0    conda-forge
charset-normalizer        3.4.0              pyhd8ed1ab_1    conda-forge
click                     8.1.7           win_pyh7428d3b_1    conda-forge
click-plugins             1.1.1              pyhd8ed1ab_1    conda-forge
cligj                     0.7.2              pyhd8ed1ab_2    conda-forge
colorama                  0.4.6              pyhd8ed1ab_1    conda-forge
contourpy                 1.3.1           py312hd5eb7cc_0    conda-forge
cycler                    0.12.1             pyhd8ed1ab_1    conda-forge
dav1d                     1.2.1                hcfcfb64_0    conda-forge
doodleverse-utils         0.0.39                   pypi_0    pypi
double-conversion         3.3.0                h63175ca_0    conda-forge
filelock                  3.16.1                   pypi_0    pypi
flatbuffers               24.3.25                  pypi_0    pypi
folium                    0.19.0             pyhd8ed1ab_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 h77eed37_3    conda-forge
fontconfig                2.15.0               h765892d_1    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.55.3          py312h31fea79_0    conda-forge
freetype                  2.12.1               hdaf720e_2    conda-forge
freexl                    2.0.0                hf297d47_2    conda-forge
fsspec                    2024.10.0                pypi_0    pypi
gast                      0.6.0                    pypi_0    pypi
gdal                      3.10.0          py312h7a712b6_7    conda-forge
geopandas                 1.0.1              pyhd8ed1ab_2    conda-forge
geopandas-base            1.0.1              pyha770c72_2    conda-forge
geos                      3.13.0               h5a68840_0    conda-forge
geotiff                   1.7.3                h496ac4d_3    conda-forge
giflib                    5.2.2                h64bf75a_0    conda-forge
git                       2.47.1               h57928b3_0    conda-forge
google-pasta              0.2.0                    pypi_0    pypi
graphite2                 1.3.13            h63175ca_1003    conda-forge
grpcio                    1.68.1                   pypi_0    pypi
h2                        4.1.0              pyhd8ed1ab_1    conda-forge
h5py                      3.12.1          nompi_py312ha036244_102    conda-forge
[... 145 lines cut ...]
ucrt                      10.0.22621.0         h57928b3_1    conda-forge
unicodedata2              15.1.0          py312h4389bb4_1    conda-forge
uriparser                 0.9.8                h5a68840_0    conda-forge
urllib3                   2.2.3              pyhd8ed1ab_1    conda-forge
vc                        14.3                ha32ba9b_23    conda-forge
vc14_runtime              14.42.34433         he29a5d6_23    conda-forge
versioneer                0.29                     pypi_0    pypi
vs2015_runtime            14.42.34433         hdffcdeb_23    conda-forge
werkzeug                  3.1.3                    pypi_0    pypi
wheel                     0.45.1             pyhd8ed1ab_1    conda-forge
win_inet_pton             1.1.0              pyh7428d3b_8    conda-forge
wrapt                     1.17.0                   pypi_0    pypi
x265                      3.5                  h2d74725_3    conda-forge
[... cut ...]
xyzservices               2024.9.0           pyhd8ed1ab_1    conda-forge
zfp                       1.0.1                he0c23c2_2    conda-forge
zipp                      3.21.0             pyhd8ed1ab_1    conda-forge
zlib                      1.3.1                h2466b09_2    conda-forge
zlib-ng                   2.2.2                he0c23c2_0    conda-forge
zstandard                 0.23.0          py312h7606c53_1    conda-forge
zstd                      1.5.6                h0ea2cb4_0    conda-forge
```
</details>

CameronBodine, 2024-12-16:
@Hesslertm I am not able to reproduce your error. Since it looks like your conda environment is up to date, I wonder if there is something wrong with the PINGMapper files on your computer. Would you be able to delete (or rename) your PINGMapper folder, then redownload with:

```
git clone https://github.com/CameronBodine/PINGMapper
```

Re-try the recording and let me know. We will get this figured out one way or another!

Hesslertm, 2024-12-17:
Hey Cameron, I just reran the recording after reinstalling PINGMapper and am still seeing the lines. I am going to wipe everything including miniconda and try again.

Hesslertm, 2024-12-17:
Just finished reinstalling everything, lines are still showing up unfortunately.

CameronBodine, 2024-12-18:
This blows me away. Can you share a few other recordings where this is happening? I need to be able to reproduce the error in order to troubleshoot. Thanks.

Hesslertm, 2024-12-18:
I uploaded another recording that has the same issue (R00078) to that google drive folder. I also attempted to rerun the one we've been messing with at 250 pings/chunk but that just made more of the lines since there were more images. See below of what the mosaic ends up looking like. Could I by chance see the log you ran that didn't give you any issues? Maybe I could go line by line to make sure the input is the same in PINGmapper. 

[image]

CameronBodine, 2024-12-18:
Good idea having a second set of eyes. Attached is my log file. I tried to specify each of the parameters as you did. 

[Replicate_log_2024-12-16_1752.txt](https://github.com/user-attachments/files/18188723/Replicate_log_2024-12-16_1752.txt)

And here is my conda environment:

<details>
<summary>Click me</summary>

```
(ping) PS C:\Users\cbodine\PythonRepos\PINGMapper> conda list
# packages in environment at C:\Users\cbodine\miniconda3\envs\ping:
#
# Name                    Version                   Build  Channel
_libavif_api              1.1.1                h57928b3_2    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
absl-py                   2.1.0                    pypi_0    pypi
affine                    2.4.0              pyhd8ed1ab_1    conda-forge
aom                       3.9.1                he0c23c2_0    conda-forge
astunparse                1.6.3                    pypi_0    pypi
attrs                     24.2.0             pyh71513ae_1    conda-forge
blosc                     1.21.6               h85f69ea_0    conda-forge
branca                    0.7.2              pyhd8ed1ab_1    conda-forge
brotli                    1.1.0                h2466b09_2    conda-forge
brotli-bin                1.1.0                h2466b09_2    conda-forge
brotli-python             1.1.0           py312h275cf98_2    conda-forge
bzip2                     1.0.8                h2466b09_7    conda-forge
c-blosc2                  2.15.2               hb461149_0    conda-forge
ca-certificates           2024.8.30            h56e8100_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.18.2               h5782bbf_1    conda-forge
certifi                   2024.8.30          pyhd8ed1ab_0    conda-forge
cffi                      1.17.1          py312h4389bb4_0    conda-forge
charls                    2.4.2                h1537add_0    conda-forge
charset-normalizer        3.4.0              pyhd8ed1ab_1    conda-forge
click                     8.1.7           win_pyh7428d3b_1    conda-forge
click-plugins             1.1.1              pyhd8ed1ab_1    conda-forge
cligj                     0.7.2              pyhd8ed1ab_2    conda-forge
colorama                  0.4.6              pyhd8ed1ab_1    conda-forge
contourpy                 1.3.1           py312hd5eb7cc_0    conda-forge
cycler                    0.12.1             pyhd8ed1ab_1    conda-forge
dav1d                     1.2.1                hcfcfb64_0    conda-forge
doodleverse-utils         0.0.39                   pypi_0    pypi
double-conversion         3.3.0                h63175ca_0    conda-forge
filelock                  3.16.1                   pypi_0    pypi
flatbuffers               24.3.25                  pypi_0    pypi
folium                    0.19.0             pyhd8ed1ab_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 h77eed37_3    conda-forge
fontconfig                2.15.0               h765892d_1    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.55.3          py312h31fea79_0    conda-forge
freetype                  2.12.1               hdaf720e_2    conda-forge
freexl                    2.0.0                h8276f4a_0    conda-forge
fsspec                    2024.10.0                pypi_0    pypi
gast                      0.6.0                    pypi_0    pypi
gdal                      3.10.0          py312h315a139_6    conda-forge
geopandas                 1.0.1              pyhd8ed1ab_2    conda-forge
geopandas-base            1.0.1              pyha770c72_2    conda-forge
geos                      3.13.0               h5a68840_0    conda-forge
geotiff                   1.7.3                h496ac4d_3    conda-forge
giflib                    5.2.2                h64bf75a_0    conda-forge
git                       2.47.1               h57928b3_0    conda-forge
google-pasta              0.2.0                    pypi_0    pypi
graphite2                 1.3.13            h63175ca_1003    conda-forge
grpcio                    1.68.1                   pypi_0    pypi
[... 154 lines cut ...]
versioneer                0.29                     pypi_0    pypi
vs2015_runtime            14.42.34433         hdffcdeb_23    conda-forge
werkzeug                  3.1.3                    pypi_0    pypi
wheel                     0.45.1             pyhd8ed1ab_1    conda-forge
win_inet_pton             1.1.0              pyh7428d3b_8    conda-forge
wrapt                     1.17.0                   pypi_0    pypi
x265                      3.5                  h2d74725_3    conda-forge
xerces-c                  3.2.5                he0c23c2_2    conda-forge
xorg-libxau               1.0.11               h0e40799_1    conda-forge
xorg-libxdmcp             1.1.5                h0e40799_0    conda-forge
xyzservices               2024.9.0           pyhd8ed1ab_1    conda-forge
xz                        5.6.3                h208afaa_1    conda-forge
xz-tools                  5.6.3                h2466b09_1    conda-forge
[... cut ...]
zipp                      3.21.0             pyhd8ed1ab_1    conda-forge
zlib                      1.3.1                h2466b09_2    conda-forge
zlib-ng                   2.2.2                he0c23c2_0    conda-forge
zstandard                 0.23.0          py312h7606c53_1    conda-forge
zstd                      1.5.6                h0ea2cb4_0    conda-forge
```
</details>

I will take a look at the new recording. Let me know if you find anything on your end. 

BTW, thanks for being patient and helping with the debug!

CameronBodine, 2024-12-18:
Tyler's log and environment file, for reference.
[log_2024-12-16_0829.txt](https://github.com/user-attachments/files/18188803/log_2024-12-16_0829.txt)

[conda list.txt](https://github.com/user-attachments/files/18188808/conda.list.txt)

Hesslertm, 2024-12-18:
Well now I know the problem is isolated to my university laptop, I was able to run it fine on my personal pc. I still have no idea what could possibly be different between the two even after a clean wipe of python and miniconda.

CameronBodine, 2024-12-18:
Ok, making some progress! I have run into install and other issues for folks that are running on a work/uni network. For these scenarios, installing on a non-work network and logged off any vpn fixed the issue. Maybe this could work for you?

CameronBodine, 2024-12-18:
After install (specifically the `ping` environment) they were able to run everything without incident.

CameronBodine, 2024-12-27:
I might be crazy, but I think the issue might be cause by exporting data to OneDrive! Can you try processing this recording again on your work computer and export to a local directory and see if that fixes it? I just had another user report the same issue, and they also exported to OneDrive.

Hesslertm, 2024-12-30:
Yes that did it! The original file we looked at is now free of those lines. I noticed the one I ran on my personal computer that I said was free of lines actually just had a couple of them, but I did export to onedrive on that machine as well. I will test a couple more files just to make sure it's reproducible and get back to you.

Hesslertm, 2024-12-30:
Processed a couple more files and they look good so looks like OneDrive was the culprit. Thank you for assisting me with this!

CameronBodine, 2024-12-30:
Glad to have that one figured out! It really perplexed me. Keep me posted.

closed 2024-12-30 by CameronBodine