CameronBodine/PINGMapper #134: "Could not process" batchDirectory

arelkins1, 2024-12-14:
### Provide Log Files
(ping) PS C:\Users\arelkins\PythonPackages3\PINGMapper> python gui_main_batchDirectory.py
WARNING:tensorflow:From C:\Users\arelkins\AppData\Local\miniconda3\envs\ping\Lib\site-packages\tf_keras\src\losses.py:2976: The name tf.losses.sparse_softmax_cross_entropy is deprecated. Please use tf.compat.v1.losses.sparse_softmax_cross_entropy instead.

Version:  2.18.0
Eager mode:  True
2024-12-14T19:08:31.030ZE [49660:NonCelloThread] registry_win.h:57:GetProtoFromRegistryValue Opening registry key Software\Google\DriveFS\Share failed with 0x2
2024-12-14T19:08:31.030ZE [37672:NonCelloThread] registry_win.h:57:GetProtoFromRegistryValue Opening registry key Software\Google\DriveFS\Share failed with 0x2
2024-12-14T19:08:38.476ZE [60068:NonCelloThread] registry_win.h:57:GetProtoFromRegistryValue Opening registry key Software\Google\DriveFS\Share failed with 0x2
inDir C:/Users/arelkins/Documents/2024data

Browse C:/Users/arelkins/Documents/2024data

0 C:/Users/arelkins/Documents/2024data/Outputs

proj C:/Users/arelkins/Documents/2024data/Outputs

prefix All

suffix

project_mode True

tempC 20.0

nchunk 500

exportUnknown False

fixNoDat False

threadCnt 0

cropRange 0.0

max_heading_deviation 0.0

max_heading_distance 0.0

min_speed 0.0

max_speed 0.0

6

aoi

x_offset 0.1

y_offset 0.1

egn False

egn_stretch Percent Clip

egn_stretch_factor 0.5

wcp False

wcr False

tileFile .jpg

lbl_set False

spdCor 0.0

maxCrop False

remShadow False
[... 811 lines cut ...]
===========================================
===========================================
***** READING *****

Getting DAT Metadata...
C:/Users/arelkins/Documents/2024data\R00018\R00018.DAT

Done!
Time (s): 0.0

!!!!!!!!!!!!!!!!!
An Error Occured.

Please consult the logfile at:
 C:/Users/arelkins/Documents/2024data/Outputs\AllR00018\logs\error_2024-12-14_1409.txt

Could not process: C:/Users/arelkins/Documents/2024data\R00018\R00018.DAT

Total Processing Time:  0:00:00

### Provide Recording Files
https://drive.google.com/drive/folders/1iPeV6BPJpxjt86vepaaxoOZvMSfI8Pbj?usp=drive_link

### Additional Information
We have gotten batch directory to run these files before, but when looking at substrate mosaics from the output, all of the output for the tracks is identical. The "could not process" error was showing in this output as well. After attempting to run the tracks again one by one and in a batch, I am receiving the same error. I hope that when resolved, it will solve the duplicate output issue.

arelkins1, 2024-12-14:
Error log: 
Error Thrown:
Traceback (most recent call last):
  File "C:\Users\arelkins\PythonPackages3\PINGMapper\gui_main.py", line 325, in <module>
    read_master_func(**params)
  File "C:\Users\arelkins\PythonPackages3\PINGMapper\src\main_readFiles.py", line 284, in read_master_func
    sonar_obj = hum2pingmapper(inFile, projDir, nchunk, tempC, exportUnknown)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\arelkins\AppData\Local\miniconda3\envs\ping\Lib\site-packages\pingverter\converter.py", line 119, in hum2pingmapper
    del s, beam
        ^
UnboundLocalError: cannot access local variable 's' where it is not associated with a value

CameronBodine, 2024-12-15:
Hi, thanks for trying pingmapper. Can you please attach the log files? They are inside of the project folders output by pingmapper. What you shared is the console output. While helpful, it does not show me what the error is. Thanks!

CameronBodine, 2024-12-15:
Ah sorry. Just saw the error. Can you share one of the recordings with me to test?

CameronBodine, 2024-12-15:
Can you also confirm that there are son files in similarly named subfolder in the same directory as the dat files?

As shown here:

https://cameronbodine.github.io/PINGMapper/docs/advanced/HumFileStructure.html#sd-card-files

arelkins1, 2024-12-15:
2024data.zip
<https://drive.google.com/file/d/1jXeLUQDDayaNwXfKl3Kky1q6qH3hRCGL/view?usp=drive_web>
Hi Cameron,

Thank you so much for your help! I am really excited to include PINGMapper
in my thesis work. I had very minimal experience with Python, and no
experience with a Humminbird, so I was so thankful for all of your
resources and help on github. Hopefully my work in the field is able to be
properly analyzed by PINGMapper.

The SD card files seem to be correct, image attached.

Here is a link to the google drive with my sonar data: Here
<https://drive.google.com/drive/folders/1iPeV6BPJpxjt86vepaaxoOZvMSfI8Pbj?usp=drive_link>.
You should have access if you use that link.
There is a zipped file attached of the same data if you'd prefer that.

Also some log files, sorry about that! They appear to all have the same
error, so I only attached two.

-- 
Arianna Elkins
Wildlife Biology / M.S in Biology
Student Research Assistant
Office of Research and Economic Development
University of Michigan- Flint

[]

 ***User Parameters***
| inFile               : C:/Users/arelkins/Documents/2024data/R00004/R00004.DAT |
| projDir              : C:/Users/arelkins/Documents/2024data/Outputs\4HBThesis |
| project_mode         : 1          |
| tempC                : 20.0       |
| nchunk               : 500        |
| cropRange            : 0.0        |
| exportUnknown        : False      |
| fixNoDat             : False      |
| threadCnt            : 0          |
| aoi                  : False      |
| max_heading_deviation : 0.0        |
| max_heading_distance : 0.0        |
| min_speed            : 0.0        |
| max_speed            : 0.0        |
| pix_res_son          : 0.05       |
| pix_res_map          : 0.25       |
| x_offset             : 0.1        |
| y_offset             : 0.1        |
| egn                  : False      |
| egn_stretch          : 2          |
| egn_stretch_factor   : 0.5        |
| wcp                  : False      |
| wcr                  : False      |
| tileFile             : .jpg       |
| lbl_set              : 0          |
| spdCor               : 0.0        |
| maxCrop              : False      |
| remShadow            : 0          |
| detectDep            : 0          |
| smthDep              : False      |
| adjDep               : 0.0        |
| pltBedPick           : False      |
| rect_wcp             : False      |
| rect_wcr             : False      |
| son_colorMap         : Greys_r    |
| pred_sub             : True       |
| pltSubClass          : True       |
| map_sub              : True       |
| export_poly          : True       |
| map_class_method     : max        |
[... 74 lines cut ...]
sonPath C:/Users/arelkins/Documents/2024data\R00005\R00005

+++++++++++++++++++++++++++++++++++++++++++
+++++++++++++++++++++++++++++++++++++++++++
***** Working On *****
C:/Users/arelkins/Documents/2024data\R00005\R00005.DAT
Start Time:  2024-12-14 14:08

===========================================
===========================================
***** READING *****

Getting DAT Metadata...
C:/Users/arelkins/Documents/2024data\R00005\R00005.DAT

Done!
Time (s): 0.0

!!!!!!!!!!!!!!!!!
An Error Occured.

Please consult the logfile at:
 C:/Users/arelkins/Documents/2024data/Outputs\AllR00005\logs\error_2024-12-14_1408.txt

Could not process: C:/Users/arelkins/Documents/2024data\R00005\R00005.DAT

CameronBodine, 2024-12-16:
I see the issue. The dat file is inside the recording folder. Pingmapper expects the file to be outside the folder, at the same level as the recording folder. Just move the dat files and everything should work.

CameronBodine, 2024-12-16:
Let me know how it goes.

CameronBodine, 2024-12-22:
@arelkins1 were you able to successfully process your recordings?

arelkins1, 2024-12-22:
Hi Cameron,

Thanks for checking in.

After moving the dat files and re-running PINGMapper, I am getting
different error log messages now. However, It appears that only R00018 is
giving the "could not process" error in the console output. I will only be
using sonar mosaics and maybe banklines. No output is showing sonar
mosaics, so I attached my PINGMapper settings just in case I am not
correctly prompting it to create sonar mosaics.

I dont see where the error log files show the corresponding track (but I
may just be naive in this sense) so just in case:
track 4: error log file ending in  0907
track 5: ending in 0914
track 6: ending in 0914
track 8: 0919
track 9: 0920
track 10:0922
track 15:0924
track 16: 0927
track 17: 0929
track 18: 0934

Thanks again,

-- 
Arianna Elkins
Wildlife Biology / M.S in Biology
Student Research Assistant
Office of Research and Economic Development
University of Michigan- Flint

Error Thrown:
Traceback (most recent call last):
  File "C:\Users\arelkins\PythonPackages3\PINGMapper\gui_main_batchDirectory.py", line 382, in <module>
    map_master_func(**params)
  File "C:\Users\arelkins\PythonPackages3\PINGMapper\src\main_mapSubstrate.py", line 434, in map_master_func
    Parallel(n_jobs=np.min([len(toMap), threadCnt]), verbose=10)(delayed(son._pltSubClass)(map_class_method, c, f, spdCor=spdCor, maxCrop=maxCrop, probs=probs) for c, f in toMap.items())
  File "C:\Users\arelkins\AppData\Local\miniconda3\envs\ping\Lib\site-packages\joblib\parallel.py", line 1918, in __call__
    return output if self.return_generator else list(output)
                                                ^^^^^^^^^^^^
  File "C:\Users\arelkins\AppData\Local\miniconda3\envs\ping\Lib\site-packages\joblib\parallel.py", line 1847, in _get_sequential_output
    res = func(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\arelkins\PythonPackages3\PINGMapper\src\class_mapSubstrateObj.py", line 758, in _pltSubClass
    label = self._classifySoftmax(chunk, softmax, map_class_method, mask_wc=True, mask_shw=True)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\arelkins\PythonPackages3\PINGMapper\src\class_mapSubstrateObj.py", line 1023, in _classifySoftmax
    label = (label*wc_mask).astype('uint8') # Zero-out water column
             ~~~~~^~~~~~~~
ValueError: operands could not be broadcast together with shapes (1223,726) (1254,501) 

Error Thrown:
Traceback (most recent call last):
  File "C:\Users\arelkins\PythonPackages3\PINGMapper\gui_main_batchDirectory.py", line 365, in <module>
    read_master_func(**params)
  File "C:\Users\arelkins\PythonPackages3\PINGMapper\src\main_readFiles.py", line 1218, in read_master_func
    depDF = psObj._saveDepth(chunks, detectDep, smthDep, adjDep)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\arelkins\PythonPackages3\PINGMapper\src\class_portstarObj.py", line 1440, in _saveDepth
    self.star._loadSonMeta()
    ^^^^^^^^^
AttributeError: 'portstarObj' object has no attribute 'star'

Error Thrown:
Traceback (most recent call last):
  File "C:\Users\arelkins\PythonPackages3\PINGMapper\gui_main_batchDirectory.py", line 382, in <module>
    map_master_func(**params)
  File "C:\Users\arelkins\PythonPackages3\PINGMapper\src\main_mapSubstrate.py", line 434, in map_master_func
[... 139 lines cut ...]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\arelkins\PythonPackages3\PINGMapper\src\class_mapSubstrateObj.py", line 1023, in _classifySoftmax
    label = (label*wc_mask).astype('uint8') # Zero-out water column
             ~~~~~^~~~~~~~
ValueError: operands could not be broadcast together with shapes (1195,343) (1168,501) 

Error Thrown:
Traceback (most recent call last):
  File "C:\Users\arelkins\PythonPackages3\PINGMapper\gui_main_batchDirectory.py", line 382, in <module>
    map_master_func(**params)
  File "C:\Users\arelkins\PythonPackages3\PINGMapper\src\main_mapSubstrate.py", line 434, in map_master_func
    Parallel(n_jobs=np.min([len(toMap), threadCnt]), verbose=10)(delayed(son._pltSubClass)(map_class_method, c, f, spdCor=spdCor, maxCrop=maxCrop, probs=probs) for c, f in toMap.items())
  File "C:\Users\arelkins\AppData\Local\miniconda3\envs\ping\Lib\site-packages\joblib\parallel.py", line 1918, in __call__
    return output if self.return_generator else list(output)
                                                ^^^^^^^^^^^^
  File "C:\Users\arelkins\AppData\Local\miniconda3\envs\ping\Lib\site-packages\joblib\parallel.py", line 1847, in _get_sequential_output
    res = func(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\arelkins\PythonPackages3\PINGMapper\src\class_mapSubstrateObj.py", line 758, in _pltSubClass
    label = self._classifySoftmax(chunk, softmax, map_class_method, mask_wc=True, mask_shw=True)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\arelkins\PythonPackages3\PINGMapper\src\class_mapSubstrateObj.py", line 1023, in _classifySoftmax
    label = (label*wc_mask).astype('uint8') # Zero-out water column
             ~~~~~^~~~~~~~
ValueError: operands could not be broadcast together with shapes (1190,546) (1223,501)

CameronBodine, 2024-12-22:
Make sure you have selected to export water column present (wcp) or removed (wcr) georectified imagery in addition to exporting the mosaic. 

There are mosaics exported to the output folder that you linked. I can take a closer at your data this week if you continue to have issues.

CameronBodine, 2024-12-22:
Also I don't see the attached pingmapper settings you referenced.

arelkins1, 2024-12-30:
Hi Cameron,

I have made some progress with your help, but I am still unable to get the
"substrate_mosaic_raster" for 3 of my 10 tracks. I am running these all as
a batch, with most tracks returning usable data. I am unable to get
substrate mosaic rasters from tracks 5, 8 and 15. Getting track 15 data is
of most concern to me, however I am missing substrate data from a large
chunk of river. I have data for most of the run, but I am missing a large
chunk (these would be tracks 10,15 and 16.) This large chunk is pictured
below, and it backs right up to a 3m high low head dam. I am wondering if
the noise from this dam prevented the sonar from mapping the
substrate, although you will see in the picture that I do have data for
the area closest to the dam, so that doesn't quite explain my noise theory.

I have re-attached the PINGMapper settings.

Please let me know how I can be of assistance further and thank you so much
for your availability and help,
Arianna

arelkins1, 2024-12-30:
Oh and here is an image of the missing sonar chunk, with the dam and flow
of water labeled

CameronBodine, 2024-12-30:
Hi Arianna, thanks for following up. Unfortunately, I still do not see the referenced files. Perhaps it is because you are replying to the email from GitHub rather than adding to the issue directly on GitHub? Can you go ahead and email them to me, or share in a cloud drive? bodine.cs@gmail.com. Thanks!

CameronBodine, 2024-12-31:
@arelkins1 thanks for providing the files in an email. Attaching them here for reference. Also adding your notes for reference. Note that you can attach files by replying to this issue directly in GitHub: [Issue 134.](https://github.com/CameronBodine/PINGMapper/issues/134)

I will take a look at this and provide further troubleshooting steps. Thanks!

> Just to recap, ive gotten 7 of 10 tracks to spit out sonar raster mosaics, leaving me with tracks 5,8, and 15 unprocessed. With 15 being of most importance. 
Additionally, I am missing some data from tracks 10 and 16, very close to a dam. It very well could be due to user error in the field, although the remaining 75% of those specific tracks provide data. This is shown in the ArcGIS image i've attached. 

[image]

[image]

[image]

CameronBodine, 2024-12-31:
Hi Arianna, see my findings below. It looks like most of your issues can be remedied by checking `Locate and flag missing pings`. Also, any missing data are due to poor scan conditions or presence of acoustic shadows, as noted below. Please let me know if this resolves your issues!

# R00005
I was unable to process this recording and received the following error:

[error_2024-12-31_1014.txt](https://github.com/user-attachments/files/18282396/error_2024-12-31_1014.txt)

There is no starboard sonar (B003.SON) channel available in the shared [Google Drive](https://drive.google.com/drive/folders/1N5RFQI4b11Np0W3os20uU8hVoWCX8__R), only the index file (B003.IDX). PINGMapper is unable to process the recording without this file.

# R00015
I am able to process R00015 by checking `Locate and flag missing pings`. Note that this should not be checked by default as it can produce some strange anomalies (see #119 for more information).

[image]

[image]

# R00008 
I am able to process R00008 by checking `Locate and flag missing pings`. Note that this should not be checked by default as it can produce some strange anomalies (see #119 for more information).

[image]

# R00010
I am able to process R00010 by checking `Locate and flag missing pings`. Note that this should not be checked by default as it can produce some strange anomalies (see #119 for more information). Areas of missing data are due to the poor data quality in the scan. Note the dark areas in the sonar mosaic near the dam. This is likely the result of the highly aerated water resulting from the dam. Highly aerated waters will degrade scan quality.

[image]

[image]

# R00016
I am unable to reproduce the issue. Any missing data is due to acoustic shadows caused by bridge pilings, vegetation, etc. If you are exporting the data to a shared drive (Google, OneNote, etc.), I would recommend trying to export directly to a local drive. This has been a recent issue for others (#133).

[image]

CameronBodine, 2025-01-21:
Hi Arianna, did this resolve your issue?

CameronBodine, 2025-03-12:
Closing as complete due to inactivity. Please reopen if your issue persists.

closed 2025-03-12 by CameronBodine