CameronBodine/PINGMapper #151: EGN implementation choices

DWhettam, 2025-02-27:
sorry I wasn't sure what sort of issue to put this under. I'm wondering about your EGN implementation. I see you've linked some descriptions of EGN in some old issues that describe it as building a 2D table storing average values given beam angle and slant range. In your implementation it looks like you've done a 1D implementation, just taking dividing by the mean intensity across a single file. Is that correct, and if so, is there any reason you didn't go for the 2D approach? 

Thanks!

CameronBodine, 2025-02-27:
I'm afraid I don't fully understand. EGN is meant to correct for attenuation of the signal based on range. This results in a 1D vector of average values where each bin in the vector corresponds with the average intensity at that range. Can you expand on your thinking with a 2D normalizing array? I linked to the paper with the formal description in #150. Maybe that will help.

CameronBodine, 2025-02-27:
I am keen to hear any alternative or improved approaches you have in mind. It would be great to improve the approach!

DWhettam, 2025-02-27:
Thanks for the reply! Sorry I should have been clearer. I'm referring to the info you linked in this issue here (https://github.com/CameronBodine/PINGMapper/issues/67#issuecomment-1598986034). Both sonarwiz, and the linked abstract build up a 2D table where x is slant range, and y is beam angle. They then store the mean values in this table. To normalise each sample the mean is then looked up in the table according to it's sample range and beam width. I hope that makes sense! I hadn't seen that paper you've just linked, that looks very useful, and it does look like their definition is as you've implemented. I may try implementing both and see how they compare

DWhettam, 2025-02-27:
ah that's helpful, thank you! I have a single beam angle value, which I believe is the max beam angle. I was planning to do a bit of triangle maths to workout the angle from the altitude and ground range, although presumably that would apply in your case as well?

CameronBodine, 2025-02-27:
If you have altitude values for each of the intensity values, then yes, it would be a simple triangulation to calculate the beam angle. That is what I would do with the sl3 data I reference in #152. 

If you only have the single depth value at nadir, I think calculating the beam angle won't provide any additional information compared with using the range alone. At least that is what I have found when I went down this rabbit hole in the past.

DWhettam, 2025-02-27:
I see, that makes sense, and has helped my understanding. OK thank you! I will dive into the .XTF data some more and see what we have

DWhettam, 2025-02-27:
Will do, thanks!

CameronBodine, 2025-03-12:
Closing as complete. Please reopen if you have more questions.

closed 2025-03-12 by CameronBodine

DWhettam, 2025-03-14:
Just updating here to say I got it working on .XTF data using pyxtf to read the data (https://github.com/oysstu/pyxtf), and then just extracting the pings and normalising as in your paper. Thanks for the help!