CameronBodine/PINGMapper #155: ValueError when processing imagery

Hesslertm, 2025-05-08:
### Describe the bug
A clear and concise description of what the bug is.

PINGMapper runs into an error when calculating EGN, the following warning appears "ValueError: need at most 63 handles, got a sequence of length 74" and PINGMapper stops. 

### To Reproduce
Steps to reproduce the behavior:
see attached log for settings

[log_2025-05-08_0911.txt](https://github.com/user-attachments/files/20104423/log_2025-05-08_0911.txt)

### Expected behavior 

### Screenshots

[image]

### Desktop (please complete the following information):
Windows 11 Pro

### Additional context
Add any other context about the problem here.

Hesslertm, 2025-05-08:
I was able to find a work around by increasing the chunk size to 750, which reduced the total number of chunks. Would it be possible to increase the chunk # limit in the workflow?

CameronBodine, 2025-05-09:
I ran into this issue also. I will keep this open and look into it. May be a while.

> Would it be possible to increase the chunk # limit in the workflow?

Are you referring to # of chunks per mosaic? There should be no limit. Increasing the chunk size to 750 will result in fewer chunks: total number of chunks = total sonar records / chunk size

I am glad you found a workaround. I will post to this issue as I make progress.

CameronBodine, 2025-05-09:
Huh, well I don't get that error with the file I had before.

What version are you running? Current is v4.2.10. If, after you update (if necessary), can you send me the problematic recording for testing?

Hesslertm, 2025-06-04:
I just updated to 4.2.11 and am getting the same error when I keep chunk size at the default 500. How would you like me to send the recording?

CameronBodine, 2025-06-05:
I will look when possible. Upload here. Thanks. 

https://drive.google.com/drive/folders/1b_6Ja1zKqr8ZOCJsns9XHXRpTId59glz

Hesslertm, 2025-06-05:
The recording has been uploaded (R00090). Thanks again.

CameronBodine, 2025-06-23:
I gave this a look. As of v4.2.13, I was able to process the files. You can see those [here](https://drive.google.com/open?id=1ouFQ8KSJ0Ni3vGSgZgcIuZniK5fU1spX&usp=drive_fs). I didn't intentionally "fix" the EGN error you are seeing, but perhaps it is fixed now. Let me know how it goes.