CameronBodine/PINGMapper #158: Test PINGMapper Issue

belostomatidae, 2025-06-20:
Hello,

I keep encountering an issue when I try to run the Test PINGMapper feature on both the small and the large dataset. For reference, I installed Miniforge and am using a device that operates on Windows 11. I don't have a ton of experience with Python, so I'm not sure what to really do to resolve this issue. I've uninstalled and reinstalled Miniforge and PINGWizard numerous times in case there was an error when they were downloading, but that has yet to resolve the issue. Any advice/insight would be greatly appreciated! 

[image]

CameronBodine, 2025-06-21:
Hello, thanks for trying PINGMapper out. 

Does this directory exist on you computer:
C:\Users\Alexi\Desktop

Are you able to process one of your own sonar files with Process Single Dataset?

belostomatidae, 2025-06-27:
Thank you for your timely response.

I just wanted to inform you that I was able to get PINGMapper to work correctly. I basically had to disable OneDrive from syncing/backing up my Desktop folder. Prior to doing that, I couldn't get PINGMapper to work on my own sonar files. After disabling it, however, it was able to process one of my sonar files. I still cannot get it to work on the test dataset (I am now receiving a different error), but considering that it processed my sonar files, I think the issue has been resolved and it is working now.

closed 2025-06-27 by CameronBodine