CameronBodine/PINGMapper #162: Failing to process Omniscan SS log - starboard transducer only edge case?

vshie, 2025-07-05:
### Provide Log Files
Attempting to process [this log file](https://drive.google.com/file/d/1wvCWVQJI4tCVs9b8tGe8tEXf5Srb63DJ/view?usp=sharing) from OmniscanSS, created fairly recently. 

Console Output Log: https://pastebin.com/wYXKd4Mt

[PINGMapper logs](https://drive.google.com/drive/folders/1iYY3M-4SFi0dxCAWjdIJyCkUQ3hi2ebh?usp=sharing) (from project folder)

### Additional thoughts
AttributeError: 'portstarObj' object has no attribute 'star' could be the problem? Maybe related to an incorrect python version ( using Python 3.9.23) . Or - that log file only has a single transducer, rather than 2? Maybe that is causing error as things seemed to be going well initially...

CameronBodine, 2025-07-06:
Ah yes, the scripts are designed for a port and starboard transducers running concurrently. I'll see what I can do to fix. I would also need an option to select the correct side. May take a bit to address this but I suspect it may be an issue for other cerulean users. I will update this issue when I address this.

CameronBodine, 2025-07-06:
> Just conducted the test on the small dataset via the gui button without issue (and quite fast? 2:41 total time)

The recording is about 100m I think, so fairly short. The time seems about right. The outputs should be in a folder on your desktop. You can double check the logs to see if it threw an error.

vshie, 2025-07-07:
Hi! 
Went to update, got an [error](https://pastebin.com/5vPyKN6i). 

My AI terminal (Warp) edited files to fix, and update was then successful. Loaded the sonar log, and it failed to process. Worse - all items in that directory (Downloads) were deleted! Luckily nothing important... d'oh!

`I can see the issue with the PINGWizard script. The problem is in how the subprocess command is being executed. The error shows that the entire command string is being treated as a single executable file path instead of being properly split into separate arguments.

Let me examine the problematic file and fix it:
/Users/tonywhite/miniconda3-py39-25.1.1-2/envs/ping/lib/python3.12/site-packages/pinginstaller/Install_Update_PINGMapper.py (1-200)
The issue in Install_Update_PINGMapper.py is that the subprocess call is not properly splitting the command string. It should pass the command as a list of arguments instead of a single string.

Let's fix the subprocess.run call in the update_pinginstaller function.

Change Needed
Instead of:
python
We should use:
python
I'll make this change for you.
Now let's also check if there's a similar issue in the update function on line 67:
Perfect! I've fixed the subprocess call issues in the PINGWizard script. The problem was that the subprocess commands were being passed as single strings instead of properly split argument lists. 

The main changes made:

1. Line 84: Changed subprocess.run('''\"{}\" run -n ping pip install pinginstaller -U'''.format(conda_key)) to subprocess.run([conda_key, 'run', '-n', 'ping', 'pip', 'install', 'pinginstaller', '-U'])
2. Line 67: Changed subprocess.run('''\"{}\" run -n ping pip install --upgrade -i https://PySimpleGUI.net/install PySimpleGUI'''.format(conda_key)) to subprocess.run([conda_key, 'run', '-n', 'ping', 'pip', 'install', '--upgrade', '-i', 'https://PySimpleGUI.net/install', 'PySimpleGUI'])

These changes ensure that the subprocess calls properly parse the command and arguments instead of treating the entire command as a single file path.

Now you can try running ./PINGWizard.sh again and it should work properly!`

vshie, 2025-07-07:
after making those changes, I was able to process a single transducer log!